Sloeber/arduino-eclipse-plugin #1355: mvn build started to fail with ClassNotFoundException: org.eclipse.core.runtime.IPlatformRunnable cannot be found by org.eclipse.tycho.surefire.osgibooter_2.0.0

jantje, 2021-06-17:
When I run
mvn verify -Pall,sloeber_release,NOSDK -Dtest=NightlyJenkins

I see following error
```
[INFO] Total time:  09:56 min
[INFO] Finished at: 2021-06-17T15:21:16+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-surefire-plugin:2.0.0:test (default-test) on project io.sloeber.tests: An unexpected error occurred while launching the test runtime (process returned error code 13 (HRESULT Code 0xD, check for example https://www.hresult.info/ for further details)). Command-line used to launch the sub-process was null -Dosgi.noShutdown=false -Dosgi.os=win32 -Dosgi.ws=win32 -Dosgi.arch=x86_64 -Dosgi.clean=true -Dorg.eclipse.cdt.core.console=org.eclipse.cdt.core.systemConsole -jar C:\Users\jan\.m2\repository\p2\osgi\bundle\org.eclipse.equinox.launcher\1.6.200.v20210416-2027\org.eclipse.equinox.launcher-1.6.200.v20210416-2027.jar -data C:\Users\jan\git\arduino-eclipse-plugin\io.sloeber.tests\target\work\data -install C:\Users\jan\git\arduino-eclipse-plugin\io.sloeber.tests\target\work -configuration C:\Users\jan\git\arduino-eclipse-plugin\io.sloeber.tests\target\work\configuration -application org.eclipse.tycho.surefire.osgibooter.headlesstest -testproperties C:\Users\jan\git\arduino-eclipse-plugin\io.sloeber.tests\target\surefire.properties in working directory C:\Users\jan\git\arduino-eclipse-plugin\io.sloeber.tests -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :io.sloeber.tests
```

```
The mentioned log contains following info
!SESSION 2021-06-17 15:21:12.603 -----------------------------------------------
eclipse.buildId=unknown
java.version=11.0.4
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_GB
Framework arguments:  -application org.eclipse.tycho.surefire.osgibooter.headlesstest -testproperties C:\Users\jan\git\arduino-eclipse-plugin\io.sloeber.tests\target\surefire.properties
Command-line arguments:  -data C:\Users\jan\git\arduino-eclipse-plugin\io.sloeber.tests\target\work\data -application org.eclipse.tycho.surefire.osgibooter.headlesstest -testproperties C:\Users\jan\git\arduino-eclipse-plugin\io.sloeber.tests\target\surefire.properties

!ENTRY org.eclipse.swt.cocoa.macosx.x86_64 2 0 2021-06-17 15:21:16.190
!MESSAGE Could not resolve module: org.eclipse.swt.cocoa.macosx.x86_64 [128]
  Unresolved requirement: Require-Capability: eclipse.platform; filter:="(& (osgi.ws=cocoa) (osgi.os=macosx) (osgi.arch=x86_64) )"
  Unresolved requirement: Require-Capability: eclipse.platform; filter:="(& (osgi.ws=cocoa) (osgi.os=macosx) (osgi.arch=x86_64) )"

!ENTRY org.eclipse.swt.gtk.linux.x86_64 2 0 2021-06-17 15:21:16.191
!MESSAGE Could not resolve module: org.eclipse.swt.gtk.linux.x86_64 [127]
  Unresolved requirement: Require-Capability: eclipse.platform; filter:="(& (osgi.ws=gtk) (osgi.os=linux) (osgi.arch=x86_64))"
  Unresolved requirement: Require-Capability: eclipse.platform; filter:="(& (osgi.ws=gtk) (osgi.os=linux) (osgi.arch=x86_64))"

!ENTRY org.eclipse.osgi 4 0 2021-06-17 15:21:16.192
!MESSAGE Application error
!STACK 1
org.eclipse.core.runtime.CoreException: Plug-in org.eclipse.tycho.surefire.osgibooter was unable to load class org.eclipse.tycho.surefire.osgibooter.HeadlessTestApplication.
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.throwException(RegistryStrategyOSGI.java:212)
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.createExecutableExtension(RegistryStrategyOSGI.java:198)
	at org.eclipse.core.internal.registry.ExtensionRegistry.createExecutableExtension(ExtensionRegistry.java:920)
	at org.eclipse.core.internal.registry.ConfigurationElement.createExecutableExtension(ConfigurationElement.java:246)
	at org.eclipse.core.internal.registry.ConfigurationElementHandle.createExecutableExtension(ConfigurationElementHandle.java:63)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:198)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:654)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:591)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1462)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1435)
Caused by: java.lang.NoClassDefFoundError: org/eclipse/core/runtime/IPlatformRunnable
	at java.base/java.lang.ClassLoader.defineClass1(Native Method)
	at java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1016)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.defineClass(ModuleClassLoader.java:293)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.defineClass(ClasspathManager.java:718)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findClassImpl(ClasspathManager.java:641)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClassImpl(ClasspathManager.java:609)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClassImpl(ClasspathManager.java:589)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:568)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:345)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:401)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:480)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:170)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.loadClass(EquinoxBundle.java:622)
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.createExecutableExtension(RegistryStrategyOSGI.java:196)
	... 16 more
Caused by: java.lang.ClassNotFoundException: org.eclipse.core.runtime.IPlatformRunnable cannot be found by org.eclipse.tycho.surefire.osgibooter_2.0.0
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:519)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:170)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	... 31 more

```

This started happening on my build server after change https://github.com/Sloeber/arduino-eclipse-plugin/commit/2a452b503fc3aa18752fd9298bdd026466ace5ed which changes a const.TRUE to a const.FALSE
I can reproduce this on my local windows and the logs above are from the windows run

jantje, 2021-06-17:
@wimjongman 
Can you shed some light on this mystery for me?
Thanks

wimjongman, 2021-06-17:
Eclipse released yesterday. I guess this is related. Let me poke around a bit.

wimjongman, 2021-06-17:
Fixed

wimjongman, 2021-06-17:
Please note that every PR is now build automatically with the new github actions framework

closed 2021-06-17 by wimjongman

jantje, 2021-06-17:
:-) thanks Wim
I guess the automatic builds via the gitrhub action framework is the similar to the travis build?
I assume this implements #1276
Is there a reason to keep the travis builds? If not we can close some issues :-)

wimjongman, 2021-06-18:
Yes, Travis can be discontinued.

jantje, 2021-06-18:
> 
> 
> Yes, Travis can be discontinued.

Great. This will allow to close some old bugs