Sloeber/arduino-eclipse-plugin #1358: ESP32 1.0.6 / 1.0,4 Partition Problem

rmerriam, 2021-06-25:
This is the same as issue #1312. 

Ubuntu 20.04
Eclipse 4.2 / CDT 10.3 (2021-06)
Sloeber 4.3.3

Started with ESP32 1.0.6 but get error:
`
o such file or directory: '/mnt/devel/ws2/eps32/Release/eps32.partitions.bin'. A `hex` file is being built, though. 
`
Don't see the command line for creating the partition version of the `bin` file. 

Saw issue 1312 so changed to ESP 1.0.4 but encounter the same error. Command line: 
`
Launching: python /home/rmerriam/eclipse/eclipse//arduinoPlugin/packages/esp32/tools/esptool_py/2.6.1/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/rmerriam/eclipse/eclipse//arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/partitions/boot_app0.bin 0x1000 /home/rmerriam/eclipse/eclipse//arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 /mnt/devel/ws2/eps32/Release/eps32.bin 0x8000 /mnt/devel/ws2/eps32/Release/eps32.partitions.bin 
`

jantje, 2021-06-26:
V4.4 is still in beta so this is not fixed unless you use the nightly.

rmerriam, 2021-06-26:
How do I get V4.4? 

I tried using the URL in Eclipse but it fails because the sub-URB for the plotter is incorrect. Couldn't see how to download the nightly build and how to install it.

jantje, 2021-06-26:
>Every night, a little gnome called Jenkins walks around our servers looking for changes occurred during the day and, when he finds any, he feels so happy he starts the build process!

>Beware though, as the nightly build represents the latest and greatest version off the code, but there is no guarantee that these versions do not contain any issue (they probably do) as the little gnome is not renowned for his patience and he doesn't test the build extensively: actually he runs away right after launching it!

>And probably worse: The nightly is the playground of the contributors. No testing has been done, no validation has been done.

https://github.com/Sloeber/arduino-eclipse-plugin/releases

rmerriam, 2021-06-26:
Will try getting the nightly build installed.

As a workaround:

1. Goto Project Properties | C/C++ Build | Settings
2. Select the "Build Steps" tab
3. Add to the "Post-build steps": 
'
${A.TOOLS.GEN_ESP32PART.CMD} -q ${JANTJE.SELECTED.PLATFORM}/tools/partitions/${A.BUILD.PARTITIONS}.csv ${A.BUILD.PATH}/${ProjName}.partitions.bin
'

closed 2021-06-27 by jantje