Sloeber/arduino-eclipse-plugin #1359: after adding to environment variable PATH, build fails

Pestimist, 2021-06-29:
HI,

I am trying to integrate automated git-hash inclusion as a pre-build step in Sloeber 4.3.3.202004230915 on Windows 10.
Therefore I had to add c:\git\bin to project->properties->C/C++ Build->Environment Variable PATH although git can be executed in a command prompt.
After that, the git command is executed but the build fails :/
This is the console output:

```"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
git log --pretty=format:'#define GIT_INFO_PRESENT%n static const char* GIT_INFO = "Version Information=[%H,%d]\r\n";' -n 1 > ../gitcommit.h
makefile:107: recipe for target 'pre-build' failed
process_begin: CreateProcess(NULL, echo " ", ...) failed.
make (e=2): The system cannot find the file specified.

make[1]: [pre-build] Error 2 (ignored)
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
makefile:76: recipe for target 'sloeber_Test' failed
process_begin: CreateProcess(NULL, echo "Building target: sloeber_Test", ...) failed.
make (e=2): The system cannot find the file specified.

make[1]: *** [sloeber_Test] Error 2
makefile:68: recipe for target 'all' failed
make: *** [all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.```

Only when I remove the added path, the build will be successful, but then the git command fails again as eclipse cannot find git executable.

I already tried adding path with ${PathDelimiter} instead of ';', or adding in front or at the end of the existing path, but no variant would lead to success.

Any ideas what I am doing wrong?

Cheers,
Bob

Pestimist, 2021-06-29:
read a bit on other issues and tried setting the variable sloeber_path_extension  to "c:\Git\bin"

no change :/

jantje, 2021-06-29:
On windows the paths is very sensitive and that is exactly why Sloeber uses FQN (fully qualified names) and reduces the path to the bare minimum.
Adding more locations at the end of the path string is the safest. But apparently in your case not safe enough.
${PathDelimiter} is only used because Linux and windows use a different character. If you are not considering cross-platform you can use the local character.
Defining sloeber_path_extension is the way to go.
Try adding the make location sloeber_path_extension =C:/Sloeber/arduinoPlugin/tools/make;c:\Git\bin

Pestimist, 2021-06-30:
Hi jantje,

I'm sorry but I didn't understand what I should exactly do to try your solution...
The only "location" I found is "make location" under project->properties->C/C++ Build
But I cannot change that...

meanwhile...
I tried specifying full path name for git in the prebuild step, which seems to work as far as executing the git command is concerned.
The problem then is that the output file is not created.
Even if I specify full path to the output file or use ${workspace_loc:/sloeber_Netzteil}

jantje, 2021-06-30:
>Try adding the make location sloeber_path_extension =C:/Sloeber/arduinoPlugin/tools/make;c:\Git\bin

should have said
Try adding the make location to the sloeber_path_extension environment variable so it becomes
sloeber_path_extension "C:/Sloeber/arduinoPlugin/tools/make;c:\Git\bin"
it is important to have the make before the git folder

Pestimist, 2021-06-30:
ok, now I understand. Thanks for the clarification.

alas, it did not change things for the better :(

```
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" -j8 all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
git log --pretty=format:'#define GIT_INFO_PRESENT%n static const char* GIT_INFO = "Version Information=[%H,%d]\r\n";' -n 1 > ../gitcommit.h
makefile:107: recipe for target 'pre-build' failed
process_begin: CreateProcess(NULL, echo " ", ...) failed.
make (e=2): The system cannot find the file specified.

make[1]: [pre-build] Error 2 (ignored)
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
makefile:76: recipe for target 'test' failed
process_begin: CreateProcess(NULL, echo "Building target: test", ...) failed.
make (e=2): The system cannot find the file specified.

make[1]: *** [test] Error 2
makefile:68: recipe for target 'all' failed
make: *** [all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make -j8 all" terminated with exit code 2. Build might be incomplete.

15:16:26 Build Failed. 6 errors, 0 warnings. (took 2s.322ms)
```

Apparently the git command now is found, but something is not right with the --pretty=format:
Just as if the quotation marks were "lost"

But still, make does not succeed either

jantje, 2021-06-30:
I'm not sure you are (only) hitting a path issue here
To me the first error seems like the echo command fails
`process_begin: CreateProcess(NULL, echo " ", ...) failed.`

Are you using the CDT prebuild functionality or are you modifying a platform.txt?

Pestimist, 2021-06-30:
I'm not sure of anything ;-)
although I have been using eclipse (although not sloeber) for quite a while, I never had to dive into configuration issues...
It just worked "out of the box" for the simple things I do and did
So my knowledge on these matters is very limited and I am not always able to understand...

I use the "product", not the plugin
I just created a new Arduino project using the "wizard" and copied my code to the workspace folder.
Did some coding and eventually came to the point where I wanted to get git-hash automagically "imported"

... and the trouble began...
since it did not work out of the box I tried a few things I read here and there
installed TM

-  terminal - no luck

- then started meddling with the PATH - still no luck

- tried your suggestion - still no luck

- did some testing with git on the command line to verify my git-command is ok
well, it is ok, but it makes a difference whether i call it from command prompt or from within eclipse-sloeber
The quotation marks use have to be " in cmd an ' in prebuil-step and in cmd it is sufficient to just give the filename of the file to be created whereas in prebuild-step i have to prefix with ../

- removed TM terminal - still no luck

jantje, 2021-06-30:
>I'm not sure of anything ;-)

I am only sure of 1 thing  "everything is way more complicated than we think it is"

>It just worked "out of the box" for the simple things I do and did

great to hear

>So my knowledge on these matters is very limited and I am not always able to understand...

That is true for everybody; though not all of us realize it. ;-)

>I use the "product", not the plugin

smart move

>....I wanted to get git-hash automagically "imported"

I have no clue what git-hash is, how it is supposed to work or what it is supposed to do, how you tried to import it.

>... and the trouble began...

You mean ... you started to learn how much effort other people have done to get things to work 😉 and how fragile it is?

>well, it is ok, but it makes a difference whether i call it from command prompt or from within eclipse-sloeber

I can write a book about this. There are a couple of things to know in this area
1) Sloeber modifies the system path in windows (because otherwise it doesn't work) not on linux or mac
2) Sloeber creates the make file
3) CDT starts the makefile
4) So the command in sloeber is interpreted by make before sending it to the command line
5) There are several makes around in windows and they all behave differently (That is why sloeber comes with a make and calls it with FQN); I have not seen similar problems on Linux
6) the make.exe sloeber (on windows) comes with behaves differently when there is a sh.exe in the path. The new Sloeber version will come with a sh.exe next to make.exe. See https://github.com/Sloeber/arduino-eclipse-plugin/issues/1198 for more details than you want
7) ... it is complicated ....

Pestimist, 2021-07-01:
Hi jantje,

thank you for your kind words and - of course - your support ! I really do appreciate

I am well aware how hard it can be to make things work. On the soft- and hardware side. And I also do know how painstakingly long it can be to track down and correct errors! And if I have learned one thing, then it is that the more complexity we bring into a system, the more fragile it becomes...

I am very grateful for all the work that has been done to bring up really great products like sloeber, and even more to every person that besides creating these wonderful tools also are like always available to help when problems show up.
By reporting bugs and trying to help resolving them, i hope that I can contribute a small part in the creation and enhancement of those tools. I am aware that this can only be a very small share of the huge workload, but I tend to believe that it somehow helps a little bit.

Just for clarification as I think it has no impact on the current issue: when I write git-hash I am referring to the hash value that git gives every commit for identification purpose. I wanted to include that into my build process to be able to exactly identify the version of software that I produce. I read a bit on this subject and thought it a very smart way to keep track of different versions.

I tired to install/upgrade to the nightly but I am not sure it worked... Installation details has a different look and sloeber is no longer listed under "installed software" but under "plugins", but still has 4.3.3 version number.
However, it changed nothing on the current issue :/

Inspired by your remark of the presence of sh.exe in PATH and reading through issue #1198 I quickly checked for the presence of sh.exe using `where sh.exe` and my suspicion was confirmed! The cygwin version I have installed for a different µC project is present in PATH and has a sh.exe!
Then, I downloaded the updated make from [http://www.baeyens.it/eclipse/download/makeV2.zip](http://www.baeyens.it/eclipse/download/makeV2.zip)  which **did the trick!**

So, in short:
### Using the new make can resolve PATH issues in current 4.3.3 stable product

As far as I am concerned, this issue is resolved. But as I am not familiar with github and how things are managed I'd like to leave it to you to decide whether to close or not.
Although I have learned quite a lot during the investigation of this problem, I can only guess how frustrating it must be to run into such "trivial" problems when having a perfectly functional product that runs on loads of different installations / systems / configurations.

Cheers,
Bob

P.S.: if there is anything I can do to place a smile on your face, let me know! I am really thankful for your help :)

Pestimist, 2021-07-01:
alas, there still is a part that is not working

now `build clean` fails with an error:
```
08:29:18 **** Clean-only build of configuration Release for project test ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" -j8 clean 
del   .\libraries\WSWireLib_rep_start-master\utility\twi.c.d  .\core\core\WInterrupts.c.d .\core\core\hooks.c.d .\core\core\wiring.c.d .\core\core\wiring_analog.c.d .\core\core\wiring_digital.c.d .\core\core\wiring_pulse.c.d .\core\core\wiring_shift.c.d  C:\Users\xXxXxXxXx\Documents\sloeber-workspace\test\Release\arduino.ar  .\core\core\CDC.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\HardwareSerial0.cpp.o .\core\core\HardwareSerial1.cpp.o .\core\core\HardwareSerial2.cpp.o .\core\core\HardwareSerial3.cpp.o .\core\core\IPAddress.cpp.o .\core\core\PluggableUSB.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\Tone.cpp.o .\core\core\USBCore.cpp.o .\core\core\WInterrupts.c.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\hooks.c.o .\core\core\main.cpp.o .\core\core\new.cpp.o .\core\core\wiring.c.o .\core\core\wiring_analog.c.o .\core\core\wiring_digital.c.o .\core\core\wiring_pulse.S.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o  .\Chiller.cpp.o .\Controller.cpp.o .\Diode.cpp.o .\I2C_Communication.cpp.o .\MyEEPROM.cpp.o .\MyEthernetClient.cpp.o .\MyEthernetServer.cpp.o .\SPI_EEPROM.cpp.o .\TelnetServer.cpp.o .\Time.cpp.o .\WebServer.cpp.o .\test.cpp.o  .\libraries\WSWireLib_rep_start-master\utility\twi.c.o  .\libraries\WSWireLib_rep_start-master\WSWire_rep_start.cpp.o  .\libraries\SPI\src\SPI.cpp.o  .\libraries\Ethernet\src\utility\w5100.cpp.o  .\libraries\Ethernet\src\Dhcp.cpp.o .\libraries\Ethernet\src\Dns.cpp.o .\libraries\Ethernet\src\Ethernet.cpp.o .\libraries\Ethernet\src\EthernetClient.cpp.o .\libraries\Ethernet\src\EthernetServer.cpp.o .\libraries\Ethernet\src\EthernetUdp.cpp.o .\libraries\Ethernet\src\socket.cpp.o  test.hex  .\core\core\wiring_pulse.S.d  test.elf  .\Chiller.cpp.d .\Controller.cpp.d .\Diode.cpp.d .\I2C_Communication.cpp.d .\MyEEPROM.cpp.d .\MyEthernetClient.cpp.d .\MyEthernetServer.cpp.d .\SPI_EEPROM.cpp.d .\TelnetServer.cpp.d .\Time.cpp.d .\WebServer.cpp.d .\test.cpp.d  .\libraries\WSWireLib_rep_start-master\WSWire_rep_start.cpp.d  .\libraries\SPI\src\SPI.cpp.d  .\libraries\Ethernet\src\utility\w5100.cpp.d  .\libraries\Ethernet\src\Dhcp.cpp.d .\libraries\Ethernet\src\Dns.cpp.d .\libraries\Ethernet\src\Ethernet.cpp.d .\libraries\Ethernet\src\EthernetClient.cpp.d .\libraries\Ethernet\src\EthernetServer.cpp.d .\libraries\Ethernet\src\EthernetUdp.cpp.d .\libraries\Ethernet\src\socket.cpp.d  .\core\core\CDC.cpp.d .\core\core\HardwareSerial.cpp.d .\core\core\HardwareSerial0.cpp.d .\core\core\HardwareSerial1.cpp.d .\core\core\HardwareSerial2.cpp.d .\core\core\HardwareSerial3.cpp.d .\core\core\IPAddress.cpp.d .\core\core\PluggableUSB.cpp.d .\core\core\Print.cpp.d .\core\core\Stream.cpp.d .\core\core\Tone.cpp.d .\core\core\USBCore.cpp.d .\core\core\WMath.cpp.d .\core\core\WString.cpp.d .\core\core\abi.cpp.d .\core\core\main.cpp.d .\core\core\new.cpp.d 
process_begin: CreateProcess(NULL, del .\libraries\WSWireLib_rep_start-master\utility\twi.c.d .\core\core\WInterrupts.c.d .\core\core\hooks.c.d .\core\core\wiring.c.d .\core\core\wiring_analog.c.d .\core\core\wiring_digital.c.d .\core\core\wiring_pulse.c.d .\core\core\wiring_shift.c.d C:\Users\xXxXxXxXx\Documents\sloeber-workspace\test\Release\arduino.ar .\core\core\CDC.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\HardwareSerial0.cpp.o .\core\core\HardwareSerial1.cpp.o .\core\core\HardwareSerial2.cpp.o .\core\core\HardwareSerial3.cpp.o .\core\core\IPAddress.cpp.o .\core\core\PluggableUSB.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\Tone.cpp.o .\core\core\USBCore.cpp.o .\core\core\WInterrupts.c.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\hooks.c.o .\core\core\main.cpp.o .\core\core\new.cpp.o .\core\core\wiring.c.o .\core\core\wiring_analog.c.o .\core\core\wiring_digital.c.o .\core\core\wiring_pulse.S.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o .\Chiller.cpp.o .\Controller.cpp.o .\Diode.cpp.o .\I2C_Communication.cpp.o .\MyEEPROM.cpp.o .\MyEthernetClient.cpp.o .\MyEthernetServer.cpp.o .\SPI_EEPROM.cpp.o .\TelnetServer.cpp.o .\Time.cpp.o .\WebServer.cpp.o .\test.cpp.o .\libraries\WSWireLib_rep_start-master\utility\twi.c.o .\libraries\WSWireLib_rep_start-master\WSWire_rep_start.cpp.o .\libraries\SPI\src\SPI.cpp.o .\libraries\Ethernet\src\utility\w5100.cpp.o .\libraries\Ethernet\src\Dhcp.cpp.o .\libraries\Ethernet\src\Dns.cpp.o .\libraries\Ethernet\src\Ethernet.cpp.o .\libraries\Ethernet\src\EthernetClient.cpp.o .\libraries\Ethernet\src\EthernetServer.cpp.o .\libraries\Ethernet\src\EthernetUdp.cpp.o .\libraries\Ethernet\src\socket.cpp.o test.hex .\core\core\wiring_pulse.S.d test.elf .\Chiller.cpp.d .\Controller.cpp.d .\Diode.cpp.d .\I2C_Communication.cpp.d .\MyEEPROM.cpp.d .\MyEthernetClient.cpp.d .\MyEthernetServer.cpp.d .\SPI_EEPROM.cpp.d .\TelnetServer.cpp.d .\Time.cpp.d .\WebServer.cpp.d .\test.cpp.d .\libraries\WSWireLib_rep_start-master\WSWire_rep_start.cpp.d .\libraries\SPI\src\SPI.cpp.d .\libraries\Ethernet\src\utility\w5100.cpp.d .\libraries\Ethernet\src\Dhcp.cpp.d .\libraries\Ethernet\src\Dns.cpp.d .\libraries\Ethernet\src\Ethernet.cpp.d .\libraries\Ethernet\src\EthernetClient.cpp.d .\libraries\Ethernet\src\EthernetServer.cpp.d .\libraries\Ethernet\sr [... cut ...] rial2.cpp.d .\core\core\HardwareSerial3.cpp.d .\core\core\IPAddress.cpp.d .\core\core\PluggableUSB.cpp.d .\core\core\Print.cpp.d .\core\core\Stream.cpp.d .\core\core\Tone.cpp.d .\core\core\USBCore.cpp.d .\core\core\WMath.cpp.d .\core\core\WString.cpp.d .\core\core\abi.cpp.d .\core\core\main.cpp.d .\core\core\new.cpp.d, ...) failed.
make (e=2): The system cannot find the file specified.

make: [makefile:103: clean] Error 2 (ignored)
 

08:29:18 Build Failed. 1 errors, 0 warnings. (took 282ms)
```

jantje, 2021-07-01:
> Just for clarification as I think it has no impact on the current issue: when I write git-hash I am referring to the hash value that git gives every commit for identification purpose. I wanted to [...]

I do not understand your reasoning here. Building and committing are 2 different steps. IMHO creating a git-hash only is useful if you also committed your changes.
I guess you do know eclipse/sloeber comes with the egit plugin. If this "git-hash" at build makes sense the providers of the e-git plugin should be able to give you some pointers

Thanks for catching the clean issue. Seems make+sh does not support a del command but does support the rm command.
This should be a easy fix.
By the way: the safest way to clean is to delete the configuration folder. Due to "lack of info in the platform.txt" Sloeber assumes you always build a hex file. When you do not build for avr the clean will not be complete.

Pestimist, 2021-07-01:
Yes, you are right...
this mechanism to include the last commit-hash in the build also depends on a bit of self-discipline, but up to now it is the best I could think of...
(Would you agree to continue discussing this topic outside this issue? I'm curious to get to know the views of someone with loads more experience in software development)

And thanks again for the bits of extra-information you give! I learned a lot

Pestimist, 2021-07-02:
> By the way: the safest way to clean is to delete the configuration folder. Due to "lack of info in the platform.txt" Sloeber assumes you always build a hex file. When you do not build for avr the clean will not be complete.

Where is this configuration folder to be found?

jantje, 2021-07-02:
>Where is this configuration folder to be found?

In the root of the project with the same name as the configuration
In that folder you should see a file called makefile and when you had a successful build arduino.ar and your binary files to upload

Pestimist, 2021-07-02:
Oh, now I understand that this refers to the build configuration (Release as a default iirc)
Thanks again for helping so much

jantje, 2021-07-04:
Can you try with the make in this zip and see if you can add the git folder to the path and build?

Pestimist, 2021-07-05:
I would, if I could find the .zip file...

jantje, 2021-07-05:
Strange I could have sworn I had added the link
https://github.com/Sloeber/arduino-eclipse-plugin/files/2695478/make.zip

Pestimist, 2021-07-06:
You did, over there in issue #1360 ;)
I can build with the modified PATH, but -clean still fails

Pestimist, 2021-07-09:
I think that the workaround in #1360 is also applicable here:
create .bat / script that includes the relevant path and resets to initial path

maybe it can be closed as well?

jantje, 2021-07-09:
indeed
Closing as known limitation with a workaround
**Limitation**
In windows adding linux based tools to the sloeber path (this is not your windows path) may cause the build to fail.

**workaround**
If you want to run a script/tool as part of the makefile that has path specific requirements that are incompatible with the tools provided by your arduino platform and/or the tools provided by Sloeber; workaround as follows:
Create a folder with a script file. In the script file set the path you need and call the tool you want
Set sloeber_path_extension  equal to the folder you created above

closed 2021-07-09 by jantje