Sloeber/arduino-eclipse-plugin #1361: Sloeber product won't install on 2021-06 after most recent update

eshicks4, 2021-07-12:
An update came through yesterday (2021-11-07) which doesn't appear compatible with the current version 2021-06.  I've tried multiple times from completely fresh installs of different "flavors" and the fewest errors I've gotten has been from the C/C++ variety:

An error occurred while collecting items to be installed
session context was:(profile=_Users_ehicks_Applications_Eclipse.app_Contents_Eclipse, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,org.eclipse.nebula.cwt,1.1.0.202011020719
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.cdatetime,1.5.0.202011020719
No repository found containing: osgi.bundle,org.hamcrest.core,1.3.0.v20180420-1519
No repository found containing: osgi.bundle,org.junit,4.13.0.v20200204-1500

I'm also attaching an export of my install sources.  It's from one of today's fresh installs but after having installed a few other packages to make sure it was just Sloeber that was having issues.

[bookmarks.xml.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/6798819/bookmarks.xml.txt)

If you need any other info please let me know.  Thanks

jantje, 2021-07-12:
I tested with downloading Sloeber version V.4.3.3 and upgrading.
Apart from having to replace the jre because of the java update (causing me to have to restart 2 times) all went well.

Can you please be more precise on what you are trying to do?
What do you men with 2021-06? There has not been a Sloeber release on 2021-06

eshicks4, 2021-07-12:
2021-06 is the latest Eclipse.  Installing Sloeber product via the Marketplace had worked until yesterday.

I haven't  been able to get the customized Eclipse version to work at all.  I'm on MacOS Big Sur with jdk1.8.0_291.jdk installed.  When I try to launch it I just get an alert window that says "Failed to create the Java Virtual Machine".  Attempting to run the binary directly I see this:

iMac:~ user$ Applications/Sloeber.app/Contents/MacOS/sloeber-ide
JavaVM: Failed to load JVM: /Library/Java/JavaVirtualMachines/jdk1.8.0_291.jdk/Contents/Home/lib/libserver.dylib
JavaVM FATAL: Failed to load the jvm library.

The binary just hangs from there until I cntl-c out of it.  It isn't lying about the file not being there but I'm not sure why it wasn't installed with the rest of the JDK or what I need to install to get it.

eshicks4, 2021-07-12:
BTW, I did see the note that the newer Sloebers need JDK11 but I'm not sure how to convince it to use it.  It is installed and a link to it is in /Libaray/Java/JavaVirtualMachines but the trick it convincing the software to use it instead of JDK8.

```
iMac:~ user$ ls -l /Library/Java/JavaVirtualMachines/
drwxr-xr-x  3 root  wheel   96 Jul 12 01:32 jdk1.8.0_291.jdk
lrwxr-xr-x  1 root  wheel   33 Jun 22 10:20 openjdk.jdk -> /usr/local/Cellar/openjdk/16.0.1/
lrwxr-xr-x  1 root  wheel   37 Jun 22 10:21 openjdk@11.jdk -> /usr/local/Cellar/openjdk@11/11.0.10/
```

jantje, 2021-07-12:
this page explains how you can set the java path
https://wiki.eclipse.org/Eclipse.ini
having a jre folder works as well if you have enough disk space
[image]

eshicks4, 2021-07-13:
The JRE subdirectory part didn't work for me as it still wanted to use the system JDK8; however, forcing the path to JDK11 in the ini file did.  Still kind of bummed that I need to split across 2 IDEs but at least I can push off learning PlatformIO a bit longer.  Feel free to do whatever you like with this issue report.  Thanks!

jantje, 2021-07-13:
The jre subfolder is a standard eclipse way to do things. This should work though.

>Still kind of bummed that I need to split across 2 IDEs 

What do you mean by that?

florianfa, 2021-08-19:
I'm facing the same issue here.

wimjongman, 2021-08-19:
https://stackoverflow.com/questions/21964709/how-to-set-or-change-the-default-java-jdk-version-on-macos

florianfa, 2021-08-19:
Hi wim,

JDK11 also won't work. I,m actually using the following environment:

Eclipse Version: 2021-06
Sloeber Version: 4.4.0
openjdk 11.0.11 2021-04-20
OpenJDK Runtime Environment (build 11.0.11+9-Ubuntu-0ubuntu2.18.04)
OpenJDK 64-Bit Server VM (build 11.0.11+9-Ubuntu-0ubuntu2.18.04, mixed mode, sharing)

Florian

wimjongman, 2021-08-19:
@florianfa what is exactly your problem? Please attach logs and stuff (attach, not paste in comment). Cheers.

Kilill, 2021-09-08:
Seems i have the same issue? 

Trying to install the plugin from the marketplace on a fresh install of eclipse:

i get the following error
Installing software has enountered a problem.

```
An error occurred while collecting items to be installed
session context was:(profile=E__eclipse_eclipse, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,jnr.x86asm,1.0.2.v201505052040
No repository found containing: osgi.bundle,org.eclipse.nebula.cwt,1.1.0.202011020719
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.cdatetime,1.5.0.202011020719
No repository found containing: osgi.bundle,org.hamcrest.core,1.3.0.v20180420-1519
No repository found containing: osgi.bundle,org.junit,4.13.0.v20200204-1500
```

OS: Windows 10

Eclipse version: 
Version: 2021-06 (4.20.0)
Build id: 20210612-2011

Attaching parts of the configuration dump from eclipse
[eclipseDump.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/7129611/eclipseDump.txt)

jantje, 2021-09-08:
From the log I would assume you do not have nebula in your update sites
http://download.eclipse.org/nebula/releases/latest/
This should resolve the 2 nebula repositories and may introduce new ones
I'm not sure where the others should be found.
Do you have http://download.eclipse.org/releases/latest/ as update site? I mean junit seems pretty common in eclipse.

Kilill, 2021-09-08:
Thanks, you was almost right, the Nebula repository was there and did contain the actual files. After a lot of trial and error i disabled the Nebula site and lo and behold it installed without complaining.

Tried to reenable the Nebula site after and sure enough, it fails.

So it seems the eclipse-21-06 install sets up some weird repositories

Thanks for the help!

jantje, 2021-09-08:
@Kilill 
The whole update site ecosystem is still unknown territory to me. I simply even can't get my head around how it is supposed to work in a multiple update site scenario as the add sloeber to standard eclipse.
Basically I think it is a miracle it works correctly so often.
Anyway that is also why we created the project that is one download with all the correct plugins in one go.

btoland, 2021-09-10:
I'm having the same problem trying to install the plugin into a fresh eclipse 2021-06 CDT install.

I'm getting exactly the same errors as @Kilill.  Tried the solution of removing and adding back in the Nebula repository.  Nebula errors go away when repository removed but others remain... and come back when Nebula re-added.  Checked the basic update site urls too - all good -as expected from a fresh install.

Any chance we could get a link to the previous (or older) release of the plugin?  I think archive versions on the website are the Sloeber Product (I could be wrong there?) but I need to create a custom bundle for teaching.

I tried fresh older versions of eclipse 10 CDT too and got same results.  Tried eclipse 9 CDT as well but plugin won't install on that - tries to upgrade to eclipse 10.  Also tried on a fresh eclipse Java 2021-06 - same errors as CDT 2021-06.

It would be great to be able to select different versions of the plugin to help hunt down incompatibility issues?

Win 10 on Virtualbox guest, macOS host.
OpenJDK 11, HotSpot JRE
Eclipse version 2021-06 CDT 10.3.2

jantje, 2021-09-10:
>I'm getting exactly the same errors as @Kilill. Tried the solution of removing and adding back in the Nebula repository. Nebula errors go away when repository removed but others remain...

And these are?

>I think archive versions on the website are the Sloeber Product (I could be wrong there?) but I need to create a custom bundle for teaching.

Why don't you take the product and start from there to build your custom bundle?
The product is eclipse +egit + cdt + sloeber + ...

Alternatively; check the master out from github go to the branch that is linked to the release you want and build with maven. That will give you the update site and the product.
The some old products install zips/archives are available but I don't want to go into the nightmare of old update sites.
Update sites are a nightmare and that is why we created the self contained product and do not support update sites

>Tried eclipse 9 CDT as well but plugin ...

https://github.com/Sloeber/arduino-eclipse-plugin/issues/1263
A similar issue exists with java versions 

>It would be great to be able to select different versions of the plugin to help hunt down incompatibility issues?

I don't see any new incompatibility issues right now.

>Win 10 on Virtualbox guest, macOS host.

Any reason why you don't run native?

.

btoland, 2021-09-10:
Hi Jantje,

Thanks for the quick reply.

I was obsessing about putting together a super "clean" bundle with the Sloeber plugin and a some others but I'm going to take your advice and try build off the Sloeber product instead.

I'm using this for teaching embedded programming and as students are mainly on Win10 so I work in a VM.  Also, then students on linux or macos can run on VM too and I only have one platform to support.

In future I might go back and build the plugin from source but just don't have time right now.

Thanks for the great work on this tool - it is excellent.

jantje, 2021-09-10:
> teaching embedded programming

Can you share the school and class? I like to boast from time to time ;-)
Don't hesitate to create specific school situation tickets or share experiences.

btoland, 2021-09-10:
TU Dublin - Ireland.  Computer Engineering Programme.

Thinking about it again now, I realise one of the reasons I'd like the plugin is because I want to introduce Eclipse as our IDE for native C/C++ development when teaching programming before adding Sloeber, AVR plugin etc and moving on to embedded.

There are arguments to keeping separate eclipse instances for different development platforms but it would be nice to have one IDE to rule them all... but maybe separate might be better... I'll keep playing with it...

Kilill, 2021-09-10:
I have tried to reproduce the original error without success, but,  thinking back to what i did i might have an idea, at some point i exterminated any trace of eclipse on the machine, and i think the important directory to get rid of /rename is the C:\Users\<yourUserNameHere>\.eclipse in addition to any install directory. (This will kill of any plugins/extensions etc that you might have previously) Then I did the following dance

1. Install eclipse normally, this will populate the "Software Sites" with the normal suspects:
[image]

2.  Try to install Sloeber Plugin from marketplace , (Help->Eclipse Marketplace,  Search for Sloeber) this will fail with:
[image]
3. The "Software List" has now been updated to:
[image]
4. Disabled all the new entries  **Except** the Sloeber site
5. Install Sloeber by Help-Install New Software, select the Sloeber site and add the Plugin.
6. It now installed without errors.

jantje, 2021-09-10:
@btoland 
Thanks
Thing is the sloeber product is eclipse + eclipse ide + CDT + sloeber
So even having sloeber installed you can use it as CDT 
Sloeber adds the sloeber toolbar and arduino menu but you can easily remove this from the CDT perspective giving you a plain CDT look.
The only direct visual hints about sloeber (once started) then are the about and the arduino in preferences 
Note that the arduino in new CDT project is from CDT and has nothing to do with Sloeber.

jantje, 2021-09-10:
@Kilill 
Thanks for the research. I'll see whether sloeber specifies these update sites. What I think about them
The CDT9 is bad 
Nebula is probably referenced and apparently no more needed
The eclipse repository should be the way to go.
The embedded C++ is "competition" that has been abandoned and the reference for sure should not be coming from sloeber
Part of GNU MCU is part of the sloeber product but not from the plugin. IMHO this has recently become part of CDT so the update site may no longer be needed for the product

Thinking about this. Are you trying to install the product and not the plugin via marketplace?
Because via marketplace adding a plugin you should add the plugin not the product.
I know it is confusing but marketplace lets you specify whether the install is a product or a plugin so I added both but to add a plugin one should use the plugin not the product.
As should be clear from my update site remarks the product installs other goodies (like the gnu mcu debugger and egit) and that is why we advice to install the product (using the downloadable archive).

Kilill, 2021-09-10:
> Thinking about this. Are you trying to install the product and not the plugin via marketplace?

No its the plugin (background for that: Already have an eclipse installation...)

wimjongman, 2021-09-10:
I am working on this via https://github.com/Sloeber/arduino-eclipse-plugin/issues/1381

jantje, 2021-09-10:
Looking at the update sites from the products
[image]
I see a new entry with http://https for which I think @wimjongman should take a look
The update site plugin holds clearly wrong values
[image]

Based on that and the info @Kilill presented it looks to me that the marketplace adds the update sites of the update plugin to the update sites of the eclipse install before trying to install the plugin.
From previous experiences with update sites and the install functionality it looked as if these update sites mentioned in the update site plugin were not used (and therefore neglected)

jantje, 2021-09-10:
pressed the wrong button

jantje, 2021-09-10:
Done some more checking
Gnu mcu definitely moved to CDT. I need to check if the package name is still valid or has changed in the transition. But the update site is no longer needed.
The nebula update site is needed

btoland, 2021-09-13:
@Kilill 
> Then I did the following dance

Thanks @Kilill.  That dance worked perfectly for me.  Didn't have to delete the .eclipse directory - just disabled the sites after the marketplace install and then installed as New Software.

btoland, 2021-09-13:
Got excited and spoke too soon!
Getting the following after that dance on eclipse restart...
Getting same when just trying to install plugin as New Software (no Marketplace attempt) on a fresh eclipse (deleted the.eclipse folder in user home folder too).

[image]

This is what my vanilla eclipse update site looks like:
[image]

This is it post Sloeber plugin install from marketplace:
[image]

When I disable all the new sites as suggested by @Kilill I get the below: - guess some dependency unresolved.
[image]

[image]

Continuing with using product as base but would still like option to walk students through from a vanilla Eclipse install.

jantje, 2021-09-13:
@btoland 
Sloeber is clearly installed as the error message "failed to parse..." is an error message from sloeber.
So indeed probably a dependency problem but to no longer a update problem.
For the dependency problem I would like to have the stack trace that should be visible in details or the log.

In the mean time  did some changes to the update site and @wimjongman made the github builds available so could you download the repository (=zipped update site) of the latest build here https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/1224353006
Unzip the repository 
point to it in update sites 
deactivate the baeyens.it update sites
And try to add sloeber to a eclipse install from the local repository

Note that the safest way for teaching is to have a specific download zip and update repository on your local network as this minimise"s external dependencies.

If this works I think I'll release 4.1.1

btoland, 2021-09-13:
Hi @jantje,

Tried the local repository install but getting similar issues - details below:

[image]

[image]

Is this the stack trace you requested?  It's the logfile from that last attempt to install the plugin from the local repository.
[.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7154567/default.log)

On teaching, yes I agree the safest/easiest/least painful is to provide a specific download which is what I'm working on.  However, part of teaching is to prep them for the real world so a bit of exposure to tool configuration will stand them well - always a balancing act!

jantje, 2021-09-13:
It is really strange the nebula stuff is not found.
When I check the nebula update site on my system I see the versions That can not be installed on you system.
[image]

Was the nebula there before you tried?
Can you try again?

wimjongman, 2021-09-13:
@btoland please press CTRL+3 -> Copy configuration data to clipboard -> paste to file -> attach

btoland, 2021-09-13:
@jantje 

Nebula site not in list before trying to install plugin:
[image]

Then after:
[image]

btoland, 2021-09-13:
@wimjongman 

Config data:
[config data.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/7155449/config.data.txt)

jantje, 2021-09-13:
I can reproduce the issue.

Download the repository (and unzipped it)
Download the eclipse java https://www.eclipse.org/downloads/download.php?file=/technology/epp/downloads/release/2021-06/R/eclipse-java-2021-06-R-win32-x86_64.zip
Start eclipse java
Add the repository to the update sites
[image]

Try to add sloeber plugin

[image]

jantje, 2021-09-13:
So I tried to install nebula
[image]
Will start from fresh to do a nebula install

jantje, 2021-09-13:
installing nebula in a fresh java ide worked
[image]

btoland, 2021-09-13:
@jantje 
I went that route too when I initially hit this.  Nebula install went fine on vanilla CDT but then ran into bunch of probs when I tried installing Sloeber afterwards.  Haven't tried with local repository version yet.

jantje, 2021-09-13:
I could install sloeber after nebula install and then ran into the "failed to parse json" problem.
Seems something is missing

jantje, 2021-09-13:
I did some more testing and I though I knew what was happening and then ... it started working :-(

Here is my reasoning (all tests have done starting from a fresh java eclipse download mentioned above)
1) If I succeed in installing nebula in eclipse I can install sloeber without problems
2) To install nebula I need to add https://download.eclipse.org/releases/latest/ to the update sites or it fails to install.
3) with only the update sites https://download.eclipse.org/releases/latest/ and https://download.eclipse.org/nebula/releases/latest/ I can install nebula and then add Sloeber from the repository

[image]
[image]
[image]
So this tells me nebula does not reference all update sites needed
Adding https://download.eclipse.org/releases/latest/ too the update sites and nebula installs
[image]
Then adding the repository mentioned above I can install Sloeber (I did not even restart eclipse)
[image]
And after restart I get the error related to java 16
[image]
And I end up with these update sites (which contains nothing alarming to me)
[image]

Deleting eclipse install and starting over
Setting the update sites to
[image]
And trying to install Sloeber
[image]
Seems to work

jantje, 2021-09-13:
I think I get it now
To install sloeber one needs to add the sloeber repository and the nebula repository and then all works fine.
However when you tried to install sloeber without having the nebula repository you need to restart for it to work.
I think the need for the nebula repository to be in the update site is because nebula does not reference the https://download.eclipse.org/releases/latest/ update site (which it should)

@wimjongman can you try to add https://download.eclipse.org/releases/latest/ to the category.xml of the update site of nebula?
Note that you have to add it to the text editor or an extra http:// will be added.

wimjongman, 2021-09-14:
The default update sites for any stock Eclipse installation are:

* Eclipse Simultaneous Release - https://download.eclipse.org/releases/xx.xx
* The Eclipse Project Updates - http://download.eclipse.org/eclipse/updates/xx.xx

These should be also part of the Sloeber product, together with any other update sites.

wimjongman, 2021-09-14:
> However when you tried to install Sloeber without having the nebula repository you need to restart eclipse for it to work.

Sometimes you need to restart...
sometimes not...
sometimes you need to restart more than once... 
or first, remove the nebula repository and then select Sloeber again...

This is not consistent behavior therefore, I think this is an Eclipse issue.

jantje, 2021-09-14:
> Sometimes you need to restart...
> sometimes not...
> sometimes you need to restart more than once...
> or first, remove the nebula repository and then select Sloeber again...
>
> This is not consistent [...]

What I mean is: once it failed with the error message during install, eclipse only recovers after restarting.

wimjongman, 2021-09-14:
> What I mean is: once it failed with the error message during install, eclipse only recovers after restarting.

 Not always.

jantje, 2021-09-14:
Well it never did in this test case during these tests

jantje, 2021-09-15:
> The default update sites for any stock Eclipse installation are:
>
>     * Eclipse Simultaneous Release - https://download.eclipse.org/releases/xx.xx
>
>     * The Eclipse Project Updates - [...]

If I understand correctly you say the sloeber update site should reference these update sites (which it does) and so should nebula (which it doesn't; I checked; tried to add it; everything became complex)

wimjongman, 2021-09-15:
The Sloeber **update site** should only add update sites that are not expected to be available. So it is not needed to add the two Eclipse Repos. It should add Nebula because that is not expected to be there.

Nebula should not add the two Eclipse repos because they are expected to be there.

**Our product** should add all repos because no one else adds them.

wimjongman, 2021-09-15:
I am currently debugging P2 but it is a big confusing log. I don't understand why it tries to add the CDateTime widget because we don't use that.

We should try to install only CDT and see if that one fails to install as well.

jantje, 2021-09-15:
First of all thanks for the update on what is expected in the update sites.
I think debugging P2 is the way to go but I also think it is a long lasting problem and as such probably hard to find.
My guess from the behaviour I see is that the install exists out of 2 main steps. Step 1 is "Validation of the installability of the selected components" (enables finish) and step 2 "actual install of the components".
It looks to me as if the list of update sites used in these 2 steps are different.
I say so because I remember testing a sloeber update site without the nebula update site and I got "Can not install nebula" error message in step 1.

>We should try to install only CDT and see if that one fails to install as well.

Isn't that what we are doing when installing nebula first and then add Sloeber? That works fine.

jantje, 2021-09-15:
> We should try to install only CDT and see if that one fails to install as well.

Thinking about this. I tried this test (which failed in the past)
Install nebula
[image]
Installs fine; restart eclipse
Remove all update sites and add sloeber update site.
This puts CDT in the situation it needs to be installed based on a update site only referenced in the sloeber update site
[image]
Step 2 fails
[image]
I restarted eclipse (just to flush all install caches)
As you can see the update sites do not contain http://download.eclipse.org/eclipse/updates/xx.xx 
[image]

remove all but sloeber and add http://download.eclipse.org/eclipse/updates/latest
[image]

and I get the same error
[image]

without editing the update stites
tried again same error
restarted and tried again 
installed

From this I conclude that installation step 2 does not take all the update sites into account gathered in step 1

jantje, 2021-09-15:
And I also have the feeling it is the caching that doesn't update as it should. 
If one can flush the caches easily; I would add a cache flush when pressing finish. That should answer the question "is this a cache issue or not"

wimjongman, 2021-09-15:
I already tried that. You can go to the preferences and "Reload" every update site but it does not work either.

jantje, 2021-09-15:
Good now we know it is not only a caching issue 🥇 (or there are more caches than we know of )
Next thing I would try is dumping/validating the update sites searched for at the time of failing to find.

Mental note to myself: As this should never happen -due to step 1 validation- this may be an ignored exception

btoland, 2021-09-21:
Finally got the plugin to install on vanilla eclipse 2021-06/2021-09 CDT.  Windows 10.  No JRE installed.

The workarounds listed at the top of this thread did not work for me but gave me a hunch so tried below which worked for me:

1. Started with eclipse CDT 2021-06 download as per CDT download site.  All previous eclipse folders, workspaces etc deleted so fresh install.
2. Before running the "Check for Updates" to update CDT - manually added to "Available Software Sites":
    https://download.eclipse.org/nebula/releases/latest/
    http://eclipse.baeyens.it/update/V4/stable
3. Ran Check for Updates to update CDT/Eclipse and did restart requested. (perhaps this sorted out the various caching issues mentioned above??).
4. After update installed Sloeber by using Help->Install New Software and selecting Sloeber update site.
5. Sloeber installed and eclipse restarted.
6. Then ran into the JRE16 issue with "Cannot parse..." as using the bundled JRE that now comes with eclipse.
7. Added the " --illegal-access=permit" (#1339, #1384 workaround) line to the END of my eclipse.ini (didn't work after -vm line) and restarted.
8. All good. Happy Days.

wimjongman, 2022-01-02:
I think this is fixed with: https://github.com/Sloeber/arduino-eclipse-plugin/pull/1431

rtek1000, 2022-06-05:
I believe I had this problem too, with the current version IDE: Version: 2022-03 (4.23.0)

jantje, 2022-06-05:
@rtek1000 I think upgrading will always have it's difficulties.

CelimoC, 2022-11-11:
to install Arduino sloeber in Eclipse IDE.

1) there is an error when installing from the market place

An error occurred while collecting items to be installed
session context was:(profile=C__Users_dcela_cpp-2022-092_eclipse, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope,1.4.0.202208171909
No repository found containing: org.eclipse.update.feature,org.eclipse.nebula.widgets.oscilloscope.feature,1.4.0.202208171909
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope.source,1.4.0.202208171909
No repository found containing: osgi.bundle,org.hamcrest.core,1.3.0.v20180420-1519
No repository found containing: osgi.bundle,org.junit,4.13.2.v20211018-1956

2) solution
Help->Install new update-> Add
copy paste in Location: https://download.eclipse.org/nebula/releases/latest/
in work with : io.sloeber.updatesite - http://eclipse.baeyens.it/update/V4/stable

Select all and install.
then Sloeber should be installed. it is not needed market place

got this error, but not paying attention:

Due to issues with long pathnames on Windows, the Sloeber installation path must be less than 40 characters. 
Your current path: C:/Users/dcela/cpp-2022-092/eclipse/arduinoPlugin is too long and the plugin will no longer function correctly for all boards.
Please visit issue #705 for details. https://github.com/Sloeber/arduino-eclipse-plugin/issues/705
Alternatively use the environment var SLOEBER_HOME.

Sloeber might still function but if you get strange results you know where to look.
Do not reate an issue if you see this!!!

cheers
/Celimo

[2 more replies]

closed 2024-02-20 by jantje