Sloeber/arduino-eclipse-plugin #1362: avrdude port and baud rate missing

sarrala, 2021-07-14:
**Describe the bug**
```
Starting upload
Uploading project "Sketch" with "avrdude"
no reset because we are using a programmer

Launching: X:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CX:\Sloeber\arduinoPlugin\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf -pattiny84 -cstk500v1 -P{serial.port} -b{program.speed} -Uflash:w:X:\Workspace\Sketch\Release/Sketch.hex:i 
Output:
avrdude: invalid baud rate specified '{program.speed}'
The execution of command "6.3.0-arduino18/bin/avrdude" is done.
```

Serial port is set to COM4 in project configuration. I assume baud rate should come right from boards configuration (did not look there). Arduino has following with same board: `-pattiny84 -cstk500v1 -PCOM4 -b19200`.

Exact same configuration and board **does work with Arduino IDE**.

**To Reproduce**
Steps to reproduce the behavior:
1. Install fresh Sloeber V4 (product package).
2. Add `http://drazzy.com/package_drazzy.com_index.json` to boards manager.
3. Enable ATTinyCore boards (used 1.5.2 which is latest).
4. Try to upload sketch to ATtiny84 using Arduino as ISP and see error.

Sloeber product V4.4.0_win64.2021-07-04_06-48-31 on Windows 10 Pro.

[image]

**Expected behavior**
avrdude gets correct options and uploads program to chip.

jantje, 2021-07-14:
>Exact same configuration and board does work with Arduino IDE.

I assume this is not true.
Are you using the default upload in Arduino IDE or the upload using ....
In other words I think you want to use default in upload protocol

sarrala, 2021-07-14:
> > Exact same configuration and board does work with Arduino IDE.
> 
> I assume this is not true.
> Are you using the default upload in Arduino IDE or the upload using ....
> In other words I think you want to use default in upload protocol

`programmer=ATTinyCore:arduinoasisp` in Arduino IDE menu it reads "Arduino as ISP"

jantje, 2021-07-14:
There are no tests for upload apart from default. So if you really use  Arduino as ISP it is likely there is a issue
Check the "Arduino as ISP" command and look for the variable name that is being used
Probably creating a global environment variable in Sloeber equalling it to ${serial.port} should do the trick

sarrala, 2021-07-14:
Seems like this is simply issue with replacing variables, I've changed
```ini
arduinoasisp.program.extra_params=-P{serial.port} -b{program.speed}
```
to
```ini
arduinoasisp.program.extra_params=-P${serial.port} -b${program.speed}
```
in programmers.txt, Arduino IDE probably just handles variables bit different way.

Also verified that using programmers.txt shipped with Sloeber has same issue.
**Maybe change / recreate this issue as feature request to fix variable usage with Sloeber?**

I'm just assuming that Sloeber is eventually supposed to support configurations that do work with Arduino IDE and this one works in Arduino IDE but not with Sloeber. My assumption is simply based on this note found from website:
> Realize that only issues that do not exist in the arduino IDE and that can be reproduced in the latest product version are valid Sloeber issues.

jantje, 2021-07-14:
>Arduino IDE probably just handles variables bit different way.

That is a nice way to put it ;-)

Seems like the  programmers.txt is not processed properly in Sloeber

jantje, 2021-07-14:
I looked into this issue a bit deeper and I found that #1167 overlooked the programmers.txt
So basically right now in V4.4 Sloeber does not work with commands from programmers.txt

jantje, 2021-07-14:
I have very little test possibilities for this problem. But I fixed the core issue that is that there is no programmers.sloeber.txt generated that converts {xx} to ${xx} and deals with OS specific commands.
Please test with the nightly tomorrow

closed 2021-09-10 by jantje