Sloeber/arduino-eclipse-plugin #1363: "bash" in the path conflicting with the external builder

elgui, 2021-07-23:
No successful build (blink included).

At launch :
"eclipse.buildId=unknown
java.fullversion=11.0.10+9
JRE 11 Windows 8 amd64-64-Bit Compressed References 20210120_899 (JIT enabled, AOT enabled)
OpenJ9   - 345e1b09e
OMR      - 741e94ea8
JCL      - 0a86953833 based on jdk-11.0.10+9
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective
it.baeyens.arduino.core

we have found programs in the path that might conflict with our external builder.
The conflicting programs are bash.
The program might still function but if you get strange build errors you know where to look

jantje, 2021-07-23:
Given I see "it.baeyens.arduino.core' in the dump.
Which version of sloeber are you using?

elgui, 2021-07-25:
I just downloaded 4.4.0

elgui, 2021-08-02:
reinstalled, all good...

closed 2021-08-02 by elgui