Sloeber/arduino-eclipse-plugin #1366: Many unexisting errors marked in source after correct build

ecobdoor, 2021-08-01:
**Environment**   

- Ubuntu 20.04
- Eclipse 2021-06 
- Sloeber 4.4.0.202107110422
- https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json vs 2.0.0-rc1
- [image]

**Describe the bug**   
As stated in the title, take a look at the screenshot below
[image]

This screenshot comes from a large program targeted to an AI Thinker ESP32-CAM (customization of espressif demo). Build & Upload work fine, program run's. Serial.printf, int16_t, class properties, single functions, #define... all these fake errors are checked & also appear in the Problems tab.

But how to use Eclipse CDT editor facilities? I know that Eclipse sometimes also leaves some unjustified errors but this is a lot ... I think it's a problem with C/C++ indexer (practically only local scope variables are ok). The main inconvenience is that the editor popup menu items (Open Declaration & others...) don't work.

**Additional context**   
boards.sloeber.txt has been customized to offer a partition choice

jantje, 2021-08-01:
The icons show that it are errors provided by the code analyser that is dependent on the indexer.
The indexer is known to somethimes go wonky. Sloeber somethimes helps the indexer go wonky ;-)
Do you see errors in the error log?

Have you tried to 
1)Delete the errors in the problems tab?
2)rebuild the index? project properties->index->rebuild
3) open and close the project

If it is a multiconfiguration project check the preferences
[image]

ecobdoor, 2021-08-08:
Tank you, yes it's ok...

Some rare fake errors remain for fairly deep items like in WiFi /src/ETH.cpp, but no problem...
[image]

Tank you

closed 2021-08-08 by ecobdoor