Sloeber/arduino-eclipse-plugin #1368: Problem to link project with references to another

ecobdoor, 2021-08-09:
**Environment:**
> Ubuntu 20.04
> Eclipse 2021-06
> Sloeber 4.4.0.202107110422
> https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json vs 2.0.0-rc1
> [image]](https://user-images.githubusercontent.com/83874439/128693141-999be3bc-8e5b-475e-914e-056e34f0cfa8.png)](https://user-images.githubusercontent.com/83874439/128691322-b0e1cbfd-864e-4cba-b9d0-e3b5e7458603.png)

**Bug description:**
In fact make compiler runs but not linker

To avoid having to define many include paths of libraries in each project, I reference a child project (Esp32_000testREF) compared to a parent project (Esp32_000root) which exports its include via "paths & symbols" as below. I specify that these are personal libraries not registered in the Arduino sense. Arduino IDE builds these sketches correctly.

[image]

These parent exported paths are well propagated in the child project and also appear in C / C ++ Settings Arduino ... Include Folders tab.

[image]

- The parent project (Esp32_000root) builds without errors
- The child project (Esp32_000testREF)  compiles well (ie compiler finds sources from parent). 
- But the linkage does not find the matching xxxx.cpp.o files probably in the parent arduino.ar (maybe before combiner, the child does not archive correctly references to parent). 

Yet the child makfile contains Esp32_000-root dependents & invocations.
[image]
However there is nothing about parent linkage that in the child attached build.log archiver & combiner parts

- Of course I clean the 2 projects and destroy Release folders. 
- Building the child makes the parent is automatically built before
- Not easy to understand pattern 1 to 6 and makefile generation...

**Expected behavior**   
Something as start-group /media/sf_media-host/DEV/ROB_000/GIT_EclipseWS/**Esp32_000-testREF**/Release/arduino.ar -l...
with the parent before /media/sf_media-host/DEV/ROB_000/GIT_EclipseWS/**Esp32_000-root**/Release/arduino.ar -l...

Greetings to you

[Esp32_000-root.build.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/6955152/Esp32_000-root.build.log)

[Esp32_000-testREF.build.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/6955104/Esp32_000-testREF.build.log)

jantje, 2021-08-09:
This a not a supported way of working and I would not know how rtto get this supported.
Basically you want a library project and a exe project and now you have 2 exe projects
Sloeber only supports exe projects

ecobdoor, 2021-08-10:
Ok I understand the complexity. It was to keep my project management habits including hierarchical units tests levels.
I did not know your plugin which has been criticized a lot in the past. Personally I find this version 4 quite functional.

Retired image processing and robotics engineer, tank you & best regards.

closed 2021-08-10 by ecobdoor

jonbarril, 2022-10-16:
The Eclipse-Sloeber platform is a great tool and I greatly appreciate the work you have put into it. However, i consider the lack of support for "library" projects to be a major flaw. To this end I have put together a set of procedures allowing crude support for library projects. I share this here for the benefit of others.

Perhaps in the future you can improve and/or automate some of these procures. Ideally, perhaps someone from the Eclipse CDT developer community can help you figure out how to get the Eclipse "project references" settings to work.

Thanks
--jon

============================

Background:

Using the latest Eclipse and Sloeber Arduino plugin as of 2022-10-15, the plugin
provides Eclipse with an Arduino IDE. It has several serious limitations as
regards project development involving reusable "libaries":

1) It does NOT support project references (i.e. project executable A requires
library project B). In Eclipse, normally, a project can reference another by
right-clicking on a project and using the "Referenced Projects" tab to specify
dependent projects. This does NOT work with the Sloeber plugin.

2) Sloeber supports one and only one executable sketch per project. Upon a clean
build the plugin generates a "sloeber.ino.cpp" file, which includes a reference
to the project .ino executable file. Without the sloeber file the compile and 
upload will fail. Note that even if a project is intended as a "library",
without an .ino executable, the plugin automatically generates a sloeber file
anyway.

Details:

The following procedures have been developed through trial and error. These are
not foolproof,  especially since the Eclipse-Sloeber platform is brittle, easily
succumbing to missteps and synchronization problems.

1) For a given "parent" project, for each of its "dependent" projects...
1.1) Right click on the parent project.
1.2) Select Import->Arduino->"Import a folder containing source...."
1.3) Browse to and select the dependent project folder.
 
2) For a "library" project (i.e. with no executable .ino file)...
2.1) Rename the "sloeber.ino.cpp" file as "sloeber.ino.cpp.xxx" (for later
possible use, as in 4).
2.2) Delete the "Release" directory.
2.3) Verify the project by clicking on the "verify" button in the toolbar. This
builds the "Release" directory, including the projects linked via import (as
in 1). The project will not fully compile since it is missing the sloeber file.
This prevents the project from being executed, but does not affect its use as a 
library.

3) For an "executable" project (i.e. with an executable .ino file)...
3.1) Assure there is a "sloeber.ino.cpp" in the project, with the "#includes"
line referencing the executable .ino file. All but the arduino.h include file
references are not needed.
3.2) Delete the "Release" directory.
3.3) Verify the project by clicking on the "verify" button in the toolbar. This
builds the "Release" directory, including any projects linked via import (as
in 1).
3.4) Upload the project by clicking on the "upload" button in the toolbar.
3.5) If the code is good but the upload fails the most likely problem is that
extra sloeber file .o and .d reference have crept into the imported library
projects. Rebuild all library projects as in 2).

4) For a library project that also contains an executable (i.e. used as a library
but also contains a .ino file, such as for testing)...
4.2) Treat it as in 3), an "executable" project.
4.3) The resulting "Release" directory will subsequently contain sloeber
file .o and .d references. This will spoil its use as a library.
4.4) Before using the project as a library treat it as in 2), a "library" 
project. Failure to do so will cause sloeber file references to be included via
the "Release" directory, which will be pulled into the parent projects.

jantje, 2022-10-16:
I'm quite happy to discuss this but please do not post in old threads with only partially related content.
I disagree with your statements on support for libraries as that is an explicit goal of sloeber.
So have you read the documentation on how to deal with personal libraries in Sloeber? http://eclipse.baeyens.it/how_to.shtml#/j
If you want to discuss please create a discussion