Sloeber/arduino-eclipse-plugin #1369: 4.4 nightly Windows - Arduino incompatibility in expansion of double quotes from platform.txt

ArminJo, 2021-08-10:
The compiler flags from the platform.txt for the 2.0.0 version from https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json
[image]

are:
`build.info.flags=-D{build.series} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DBOARD_NAME="{build.board}" -DVARIANT_H="{build.variant_h}"`
and are converted in platform.sloeber.txt to:
`build.info.flags=-D${build.series} -DARDUINO=${runtime.ide.version} -DARDUINO_${build.board} -DARDUINO_ARCH_${build.arch} "-DBOARD_NAME=\"${build.board}\"" -DVARIANT_H="${build.variant_h}"`

This results in the commandline: `... -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103C8\"" -DVARIANT_H="variant_PILL_F103Cx.h" ...`

But in Arduino the comandline is different: `... -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103C6\"" "-DVARIANT_H=\"variant_BLUEPILL_F103C6.h\"" ...`

The **-DVARIANT_H="variant_PILL_F103Cx.h"** does not work! It requires at least a  **-DVARIANT_H=\"variant_BLUEPILL_F103C6.h\"** at the commandline.
Workaround is to add a compile option `-DVARIANT_H=\"variant_PILL_F103Cx.h\"` to override the defect value.

jantje, 2021-08-10:
I'm confused here.
From the title of the issue it looks like there is a quoting issue where the example and workaround also provide a different value

ArminJo, 2021-08-10:
You are right, I missed to select the correct BluePill variant in Arduino. But this does not matter, so I corrected the above issue description.

jantje, 2021-08-10:
Thanks
I see that -DVARIANT_H is not handled when doing the double quoting workaround processing
So I confirm the issue and the workaround looks great 👍 
It may be even better to set the compile command as -DVARIANT_H=\"${build.variant_h}\" to avoid problems when changing variant

jantje, 2021-08-22:
It should be fixed in the nightly (after next build).
Can you verify?

ArminJo, 2021-08-22:
I just updated and the bug is **not** fixed.

jantje, 2021-08-23:
Thanks for testing. 🎈 
IMHO You were to early. Last successful nightly build is 12 hours ago and your comment is marked as 14 hours ago.
You can easily check whether you have the version with the fix by checking the first line in the sloeber.txt files. It should say
"#Sloeber created workaound file V1.02.test 20 "
https://github.com/Sloeber/arduino-eclipse-plugin/commit/6f0eaa4966c95fd480ed72fa38ff545a546d923c

So IMHO There should be a new nightly that contains the fix.

ArminJo, 2021-08-24:
Ficed now 👍

closed 2021-08-24 by ArminJo

jantje, 2021-08-24:
thanks for testing :-)