Sloeber/arduino-eclipse-plugin #1370: Sloeber 4.4 Windows - Deadlock when opening a closed project from 4.3

ArminJo, 2021-08-11:
Hello, I have 17 sloeber 4.3 projects (for Arduino Nano) which I cannot open any more. Sloeber just hangs with 0% CPU.
The project `1_1_Ampel` now opens sucessfully after I removed and (re)imported it.
But when opening another like e.g. `1_2_ButtonInputToLedAndSerial` Sloeber hangs 😢

The log file is like this:
```
!ENTRY io.sloeber.core 8 0 2021-08-11 10:15:47.235
!MESSAGE libraries added for project IRremote_ATtiny167

!ENTRY io.sloeber.core 8 0 2021-08-11 10:18:52.688
!MESSAGE Do not index 1_1_Ampel

!ENTRY io.sloeber.core 8 0 2021-08-11 10:18:52.699
!MESSAGE index 1_1_Ampel

!ENTRY io.sloeber.core 8 0 2021-08-11 10:19:00.424
!MESSAGE Do not index 1_1_Ampel

!ENTRY io.sloeber.core 8 0 2021-08-11 10:19:00.674
!MESSAGE index 1_1_Ampel

!ENTRY io.sloeber.core 8 0 2021-08-11 10:19:01.992
!MESSAGE do not pospone index 1_1_Ampel

!ENTRY io.sloeber.core 8 0 2021-08-11 10:19:05.510
!MESSAGE Do not index 1_2_ButtonInputToLedAndSerial

!ENTRY org.eclipse.core.jobs 4 2 2021-08-11 10:19:05.564
!MESSAGE Deadlock detected. All locks owned by thread main will be suspended.
!STACK 0
java.lang.IllegalStateException
	at org.eclipse.core.internal.jobs.DeadlockDetector.reportDeadlock(DeadlockDetector.java:604)
	at org.eclipse.core.internal.jobs.DeadlockDetector.lockWaitStart(DeadlockDetector.java:408)
	at org.eclipse.core.internal.jobs.LockManager.addLockWaitThread(LockManager.java:147)
	at org.eclipse.core.internal.jobs.ThreadJob.waitForRun(ThreadJob.java:307)
	at org.eclipse.core.internal.jobs.ThreadJob.joinRun(ThreadJob.java:205)
	at org.eclipse.core.internal.jobs.ImplicitJobs.begin(ImplicitJobs.java:95)
	at org.eclipse.core.internal.jobs.JobManager.beginRule(JobManager.java:299)
	at org.eclipse.core.internal.resources.WorkManager.checkIn(WorkManager.java:124)
	at org.eclipse.core.internal.resources.Workspace.prepareOperation(Workspace.java:2263)
	at org.eclipse.core.internal.resources.File.create(File.java:106)
	at org.eclipse.core.internal.resources.File.create(File.java:179)
	at io.sloeber.core.api.SloeberProject.storeConfigurationFile(SloeberProject.java:770)
	at io.sloeber.core.api.SloeberProject.createSloeberConfigFiles(SloeberProject.java:737)
	at io.sloeber.core.api.SloeberProject.configure(SloeberProject.java:430)
	at io.sloeber.core.api.SloeberProject.configure(SloeberProject.java:387)
	at io.sloeber.core.api.SloeberProject.getEnvironmentVariables(SloeberProject.java:1070)
	at io.sloeber.core.toolchain.SloeberConfigurationVariableSupplier.getVariable(SloeberConfigurationVariableSupplier.java:36)
	at org.eclipse.cdt.managedbuilder.internal.envvar.ExternalExtensionEnvironmentSupplier.getVariable(ExternalExtensionEnvironmentSupplier.java:68)
	at org.eclipse.cdt.managedbuilder.internal.dataprovider.BuildEnvironmentContributor.getVariable(BuildEnvironmentContributor.java:56)
	at org.eclipse.cdt.internal.core.envvar.BuildSystemEnvironmentSupplier.getVariable(BuildSystemEnvironmentSupplier.java:164)
	at org.eclipse.cdt.internal.core.envvar.EnvironmentVariableManager.getVariable(EnvironmentVariableManager.java:162)
	at org.eclipse.cdt.internal.core.envvar.EnvironmentVariableManager.getVariable(EnvironmentVariableManager.java:193)
	at org.eclipse.cdt.internal.core.cdtvariables.EnvironmentVariableSupplier.getMacro(EnvironmentVariableSupplier.java:135)
	at org.eclipse.cdt.internal.core.cdtvariables.CoreMacroSupplierBase.getVariable(CoreMacroSupplierBase.java:26)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableManager.getVariable(SupplierBasedCdtVariableManager.java:35)
	at org.eclipse.cdt.internal.core.cdtvariables.CdtVariableManager.getVariable(CdtVariableManager.java:67)
	at org.eclipse.cdt.managedbuilder.internal.macros.CoreMacrosSupplier.getVariable(CoreMacrosSupplier.java:43)
	at org.eclipse.cdt.managedbuilder.internal.macros.CoreMacrosSupplier.getMacro(CoreMacrosSupplier.java:33)
	at org.eclipse.cdt.managedbuilder.internal.macros.BuildCdtVariablesSupplierBase.getVariable(BuildCdtVariablesSupplierBase.java:43)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableManager.getVariable(SupplierBasedCdtVariableManager.java:35)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.resolveMacro(SupplierBasedCdtVariableSubstitutor.java:274)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.resolveMacro(SupplierBasedCdtVariableSubstitutor.java:265)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.getResolvedMacro(SupplierBasedCdtVariableSubstitutor.java:250)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.resolveToString(SupplierBasedCdtVariableSubstitutor.java:218)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.resolveToString(SupplierBasedCdtVariableSubstitutor.java:234)
	at org.eclipse.cdt.utils.cdtvariables.CdtVariableResolver.resolveToString(CdtVariableResolver.java:127)
	at org.eclipse.cdt.managedbuilder.internal.macros.BuildMacroProvider.resolveValue(BuildMacroProvider.java:225)
	at org.eclipse.cdt.managedbuilder.internal.core.Builder.getBuildLocation(Builder.java:1867)
	at org.eclipse.cdt.managedbuilder.core.ManagedBuildManager.getBuildFullPath(ManagedBuildManager.java:3791)
[... 40 lines cut ...]
	at org.eclipse.ui.internal.Workbench$$Lambda$256/0x0000000000000000.run(Unknown Source)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:654)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:591)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1462)
!SUBENTRY 1 org.eclipse.core.jobs 4 2 2021-08-11 10:19:05.565
!MESSAGE Thread main has locks: OrderedLock (26) and is waiting for lock P/1_2_ButtonInputToLedAndSerial
!SUBENTRY 1 org.eclipse.core.jobs 4 2 2021-08-11 10:19:05.565
!MESSAGE Thread Worker-15: Sloeber.cfg modification processor has locks: R/, P/1_2_ButtonInputToLedAndSerial and is waiting for lock OrderedLock (26)
```

Do you have any idea?
Thanks in advance!

ArminJo, 2021-08-11:
A lot of other projects, even for Arduino Nano can be opened without any problems.
But I have another project for another core than the arduino one, that has the same problem.

```
!ENTRY io.sloeber.core 8 0 2021-08-11 11:05:41.617
!MESSAGE Do not index IRremote_TinyCore32

!ENTRY org.eclipse.core.jobs 4 2 2021-08-11 11:05:41.618
!MESSAGE Deadlock detected. All locks owned by thread main will be suspended.
!STACK 0
java.lang.IllegalStateException
	at org.eclipse.core.internal.jobs.DeadlockDetector.reportDeadlock(DeadlockDetector.java:604)
	at org.eclipse.core.internal.jobs.DeadlockDetector.lockWaitStart(DeadlockDetector.java:408)
	at org.eclipse.core.internal.jobs.LockManager.addLockWaitThread(LockManager.java:147)
	at org.eclipse.core.internal.jobs.OrderedLock.doAcquire(OrderedLock.java:171)
	at org.eclipse.core.internal.jobs.OrderedLock.acquire(OrderedLock.java:109)
	at org.eclipse.core.internal.jobs.OrderedLock.acquire(OrderedLock.java:85)
	at org.eclipse.cdt.internal.core.settings.model.xml.XmlProjectDescriptionStorage.getProjectDescription(XmlProjectDescriptionStorage.java:233)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescriptionInternal(CProjectDescriptionManager.java:427)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:409)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:403)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:396)
	at org.eclipse.cdt.core.resources.RefreshScopeManager.loadSettings(RefreshScopeManager.java:474)
	at org.eclipse.cdt.core.resources.RefreshScopeManager$1$1.visit(RefreshScopeManager.java:154)
	at org.eclipse.core.internal.events.ResourceDelta.accept(ResourceDelta.java:67)
	at org.eclipse.core.internal.events.ResourceDelta.accept(ResourceDelta.java:77)
	at org.eclipse.core.internal.events.ResourceDelta.accept(ResourceDelta.java:51)
	at org.eclipse.cdt.core.resources.RefreshScopeManager$1.resourceChanged(RefreshScopeManager.java:142)
	at org.eclipse.core.internal.events.NotificationManager$1.run(NotificationManager.java:305)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.core.internal.events.NotificationManager.notify(NotificationManager.java:295)
	at org.eclipse.core.internal.events.NotificationManager.broadcastChanges(NotificationManager.java:158)
	at org.eclipse.core.internal.resources.Workspace.broadcastPostChange(Workspace.java:381)
	at org.eclipse.core.internal.resources.Workspace.endOperation(Workspace.java:1503)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:49)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
!SUBENTRY 1 org.eclipse.core.jobs 4 2 2021-08-11 11:05:41.618
!MESSAGE Thread Worker-6: Open Project has locks: R/, OrderedLock (0), P/IRremote_TinyCore32 and is waiting for lock OrderedLock (43)
!SUBENTRY 1 org.eclipse.core.jobs 4 2 2021-08-11 11:05:41.618
!MESSAGE Thread main has locks: OrderedLock (43) and is waiting for lock P/IRremote_TinyCore32
```

This are the projects which I cannot open:
[image]

jantje, 2021-08-11:
From the trace it is clear that the upgrade to 4.4 causes a lock situation after which basically everything is halted.
I have not seen this on my system and the log does not show any decoration but it is also not clear what is the other part of the lock. And decoration caused many issues for the upgrade.

So I would try to disable the sloeber decorator; close the workspace; reopen the workspace and retry opening the projects.
If this has worked you can reenable the sloeber decorator if you want.
To disable the Sloeber decorator goto windows->preferences and disable it as below (Dutch version)
[image]
I hope this helps

ArminJo, 2021-08-11:
Thanks, but this does not help 😞 
I get:
```
!ENTRY io.sloeber.core 8 0 2021-08-11 21:29:38.754
!MESSAGE Do not index IRMP_TinyCore32

!ENTRY org.eclipse.core.jobs 4 2 2021-08-11 21:29:38.829
!MESSAGE Deadlock detected. All locks owned by thread main will be suspended.
!STACK 0
java.lang.IllegalStateException
	at org.eclipse.core.internal.jobs.DeadlockDetector.reportDeadlock(DeadlockDetector.java:604)
	at org.eclipse.core.internal.jobs.DeadlockDetector.lockWaitStart(DeadlockDetector.java:408)
	at org.eclipse.core.internal.jobs.LockManager.addLockWaitThread(LockManager.java:147)
	at org.eclipse.core.internal.jobs.ThreadJob.waitForRun(ThreadJob.java:307)
	at org.eclipse.core.internal.jobs.ThreadJob.joinRun(ThreadJob.java:205)
	at org.eclipse.core.internal.jobs.ImplicitJobs.begin(ImplicitJobs.java:95)
	at org.eclipse.core.internal.jobs.JobManager.beginRule(JobManager.java:299)
	at org.eclipse.core.internal.resources.WorkManager.checkIn(WorkManager.java:124)
	at org.eclipse.core.internal.resources.Workspace.prepareOperation(Workspace.java:2263)
	at org.eclipse.core.internal.resources.File.create(File.java:106)
	at org.eclipse.core.internal.resources.File.create(File.java:179)
	at io.sloeber.core.api.SloeberProject.storeConfigurationFile(SloeberProject.java:770)
	at io.sloeber.core.api.SloeberProject.createSloeberConfigFiles(SloeberProject.java:737)
	at io.sloeber.core.api.SloeberProject.configure(SloeberProject.java:430)
	at io.sloeber.core.api.SloeberProject.configure(SloeberProject.java:387)
	at io.sloeber.core.api.SloeberProject.getEnvironmentVariables(SloeberProject.java:1070)
	at io.sloeber.core.toolchain.SloeberConfigurationVariableSupplier.getVariable(SloeberConfigurationVariableSupplier.java:36)
	at org.eclipse.cdt.managedbuilder.internal.envvar.ExternalExtensionEnvironmentSupplier.getVariable(ExternalExtensionEnvironmentSupplier.java:68)
	at org.eclipse.cdt.managedbuilder.internal.dataprovider.BuildEnvironmentContributor.getVariable(BuildEnvironmentContributor.java:56)
	at org.eclipse.cdt.internal.core.envvar.BuildSystemEnvironmentSupplier.getVariable(BuildSystemEnvironmentSupplier.java:164)
	at org.eclipse.cdt.internal.core.envvar.EnvironmentVariableManager.getVariable(EnvironmentVariableManager.java:162)
	at org.eclipse.cdt.internal.core.envvar.EnvironmentVariableManager.getVariable(EnvironmentVariableManager.java:193)
	at org.eclipse.cdt.internal.core.cdtvariables.EnvironmentVariableSupplier.getMacro(EnvironmentVariableSupplier.java:135)
	at org.eclipse.cdt.internal.core.cdtvariables.CoreMacroSupplierBase.getVariable(CoreMacroSupplierBase.java:26)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableManager.getVariable(SupplierBasedCdtVariableManager.java:35)
	at org.eclipse.cdt.internal.core.cdtvariables.CdtVariableManager.getVariable(CdtVariableManager.java:67)
	at org.eclipse.cdt.managedbuilder.internal.macros.CoreMacrosSupplier.getVariable(CoreMacrosSupplier.java:43)
	at org.eclipse.cdt.managedbuilder.internal.macros.CoreMacrosSupplier.getMacro(CoreMacrosSupplier.java:33)
	at org.eclipse.cdt.managedbuilder.internal.macros.BuildCdtVariablesSupplierBase.getVariable(BuildCdtVariablesSupplierBase.java:43)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableManager.getVariable(SupplierBasedCdtVariableManager.java:35)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.resolveMacro(SupplierBasedCdtVariableSubstitutor.java:274)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.resolveMacro(SupplierBasedCdtVariableSubstitutor.java:265)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.getResolvedMacro(SupplierBasedCdtVariableSubstitutor.java:250)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.resolveToString(SupplierBasedCdtVariableSubstitutor.java:218)
	at org.eclipse.cdt.utils.cdtvariables.SupplierBasedCdtVariableSubstitutor.resolveToString(SupplierBasedCdtVariableSubstitutor.java:234)
	at org.eclipse.cdt.utils.cdtvariables.CdtVariableResolver.resolveToString(CdtVariableResolver.java:127)
	at org.eclipse.cdt.managedbuilder.internal.macros.BuildMacroProvider.resolveValue(BuildMacroProvider.java:225)
	at org.eclipse.cdt.managedbuilder.internal.core.Builder.getBuildLocation(Builder.java:1867)
	at org.eclipse.cdt.managedbuilder.core.ManagedBuildManager.getBuildFullPath(ManagedBuildManager.java:3791)
	at org.eclipse.cdt.managedbuilder.internal.core.Builder.getDefaultOutputSettings(Builder.java:2767)
	at org.eclipse.cdt.managedbuilder.internal.core.Builder.getOutputEntries(Builder.java:2751)
	at org.eclipse.cdt.managedbuilder.internal.dataprovider.BuildBuildData.getOutputDirectories(BuildBuildData.java:58)
	at org.eclipse.cdt.core.settings.model.extension.impl.CDefaultBuildData.copySettingsFrom(CDefaultBuildData.java:58)
	at org.eclipse.cdt.internal.core.settings.model.CBuildSettingCache.<init>(CBuildSettingCache.java:45)
	at org.eclipse.cdt.internal.core.settings.model.CConfigurationDescriptionCache.copyBuildData(CConfigurationDescriptionCache.java:247)
	at org.eclipse.cdt.core.settings.model.extension.impl.CDefaultConfigurationData.copySettingsFrom(CDefaultConfigurationData.java:110)
	at org.eclipse.cdt.internal.core.settings.model.CConfigurationDescriptionCache.loadData(CConfigurationDescriptionCache.java:154)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescription.loadDatas(CProjectDescription.java:199)
[... cut ...]
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:654)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:591)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1462)
!SUBENTRY 1 org.eclipse.core.jobs 4 2 2021-08-11 21:29:38.829
!MESSAGE Thread main has locks: OrderedLock (13) and is waiting for lock P/IRMP_TinyCore32
!SUBENTRY 1 org.eclipse.core.jobs 4 2 2021-08-11 21:29:38.829
!MESSAGE Thread Worker-8: Sloeber.cfg modification processor has locks: R/, P/IRMP_TinyCore32 and is waiting for lock OrderedLock (13)
```

jantje, 2021-08-11:
Is there no other stacktrace of the other locking thread?
From what I can see Sloeber detected the project open and started the thread "Sloeber.cfg modification processor" (no stack trace) because the project needs to upgrade which means writing files and Sloeber is not allowed to do file modifications at this point
While this thread is working another thread (stack trace provided) tries to get environment variables and sloeber again sees it needs to write files which causes a deadlock

ArminJo, 2021-08-11:
This seems right.
If I remove all 4.3 entries in org.eclipse.cdt.core.prefs, then opening the project does not hang!
A race condition? Since for other projects and for importing these problematic projects it works.

What is the difference between opening (which does not work) and importing, which works?

jantje, 2021-08-21:
Can this issue be closed?

ArminJo, 2021-08-22:
Only workaround found is to reimport the project.

closed 2021-08-22 by ArminJo

ArminJo, 2021-08-24:
Another workaround is to open the projects right after startup of sloeber. This seems to avoid the race condition.

jantje, 2021-08-24:
That is very well possible as I never tested the scenario open workspace with closed sloeber projects; open a sloeber project.