Sloeber/arduino-eclipse-plugin #1375: Cannot update form 4.3.3 to 4.4

rlogiacco, 2021-08-29:
Looks like I cannot update from 4.3.3 due to a Sloeber 4.4 dependency which seems no more available, specifically `ECF Filetransfer Feature 3.14.1703`: I have `ECF Filetransfer Feature 3.14.1400` and only `5.0` and `5.1` seem available for download...

```
Cannot complete the install because of a conflicting dependency.
  Software being installed: Sloeber 4.4.0.202107110422 (io.sloeber.product 4.4.0.202107110422)
  Software currently installed: ECF Filetransfer Feature 3.14.1400.v20200812-2314 (org.eclipse.ecf.filetransfer.feature.feature.group 3.14.1400.v20200812-2314)
  Only one of the following can be installed at once: 
    ECF Filetransfer API 5.1.102.v20210409-2301 (org.eclipse.ecf.filetransfer 5.1.102.v20210409-2301)
    ECF Filetransfer API 5.1.100.v20200611-2221 (org.eclipse.ecf.filetransfer 5.1.100.v20200611-2221)
    ECF Filetransfer API 5.0.100.v20180301-0132 (org.eclipse.ecf.filetransfer 5.0.100.v20180301-0132)
  Cannot satisfy dependency:
    From: Sloeber 4.4.0.202107110422 (io.sloeber.product 4.4.0.202107110422)
    To: org.eclipse.equinox.p2.iu; org.eclipse.ecf.filetransfer.feature.feature.group [3.14.1703.v20210409-2301,3.14.1703.v20210409-2301]
  Cannot satisfy dependency:
    From: ECF Filetransfer Feature 3.14.1400.v20200812-2314 (org.eclipse.ecf.filetransfer.feature.feature.group 3.14.1400.v20200812-2314)
    To: org.eclipse.equinox.p2.iu; org.eclipse.ecf.filetransfer [5.1.100.v20200611-2221,5.1.100.v20200611-2221]
  Cannot satisfy dependency:
    From: ECF Filetransfer Feature 3.14.1703.v20210409-2301 (org.eclipse.ecf.filetransfer.feature.feature.group 3.14.1703.v20210409-2301)
    To: org.eclipse.equinox.p2.iu; org.eclipse.ecf.filetransfer [5.1.102.v20210409-2301,5.1.102.v20210409-2301]
```

jantje, 2021-08-29:
Seems this is defined in the product.
On my system I see there is no version info given but when I double click on 
[image]

I get 

[image]

I am updating my development environment and I hope that version increases
Don't know what else I can do

jantje, 2021-08-29:
After the upgrade I get a newer version
[image]

but no git files have been changed so IMHO the server build should be exactly the same
So I do not know how this works :-(
But given it is in the product .... I can remove it :-) 

@wimjongman 
I see org.eclipse.ecf.filetransfer is referenced in the first version of  arduino.product https://github.com/Sloeber/arduino-eclipse-plugin/blob/7edc7e9d383b0322c0ff30d3847781c552e653c7/io.sloeber.product/arduino.product#L312
Do you know if this is needed?
Can you explain the problem @rlogiacco is experiencing? 
Do you know how to fix it?

wimjongman, 2021-08-30:
ECF (Eclipse Communication Framework) is the backbone of all communications in Eclipse. It is needed, e.g., for updating software.

Roberto, you must have another item in your installation that requires this older version of ECF. Did you install Sloeber into an existing Eclipse or are you running the 4.3.3 product?

wimjongman, 2021-08-30:
I downloaded Sloeber 4.3.3 and started it. Then I  did a "check for updates" and I could install 4.4. without issues. After restarting, I ran into some other issues [1] but not this specific one.

[1]  Old Sloeber product cannot be updated #1376

wimjongman, 2021-09-12:
Closing. Please reopen if you can reproduce the issue on a fresh 4.3.3.

closed 2021-09-12 by wimjongman