Sloeber/arduino-eclipse-plugin #1376: Old Sloeber product cannot be updated

wimjongman, 2021-08-30:
When people are using 4.3.3 product, the jre is not updated when "Check for updates" is performed. The product is updated and restarted, but it still runs the old JRE which is version 8.   

After closing and starting Sloeber, a message is shown that the jre is outdated. The jre directory must be renamed (e.g. to jrexxx) or deleted. Then Sloeber will pickup the installed jre.

This is not a pleasant experience and is even more problematic when people are not into "installing java" and just want to update Sloeber and run.

 One solution is to include the justj java environment. This is an embedded jvm and Eclipse also ships with it.

jantje, 2021-08-30:
This would be a great thing as I have been struggling in getting jre's into the product and I am only confident enough to do it for windows.
If this would add a jre to all supported osses that would be great :-)

wimjongman, 2021-08-30:
I assume that you are adding the jre library to the build in a post step, correct?

jantje, 2021-08-30:
>I assume that you are adding the jre library to the build in a post step, correct?

yes
On the jenkings build

I tried the new code here; Did a maven build and unzipped "sloeber-ide-sloeber_release-win32.win32.x86_64"
Started it and saw it was using my system installed java
This was not what I expected :-(

wimjongman, 2021-08-30:
> Started it and saw it was using my system installed java

Are you on the correct branch?

jantje, 2021-08-30:
I'm on master (all the time ;-) )

wimjongman, 2021-08-30:
> 
> 
> I'm on master (all the time ;-) )

Yeah, that is the wrong branch.

jantje, 2021-08-31:
>Yeah, that is the wrong branch.

I'm proving my incompetence again 🥇

jantje, 2021-08-31:
Woot !!! 🎆 
Seems this works (on windows) 
Need to test this on linux

jantje, 2021-09-08:
I tested this on windows.
Is there any reason not to merge this into master?

wimjongman, 2021-09-08:
Please merge if you are happy with it.

jantje, 2021-09-08:
I'm happy :-)

closed 2021-09-08 by jantje