Sloeber/arduino-eclipse-plugin #1378: problems using stm32duino in sloeber 4.4

Pestimist, 2021-09-02:
Hi,

I am using a fresh install of sloeber 4.4 and I even cleared the workspace and created a new project in which I copied my old source files because of issues during update - but that is not the concern.

I am evaluating whether I can switch my project over to STM32 and therefore downloaded the platform from https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json

When I build (after clean or incremental) I get several times the same error in the console:

`C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino/Arduino.h:33:12: fatal error: SrcWrapper.h: No such file or directory
   33 |   #include "SrcWrapper.h"
      |            ^~~~~~~~~~~~~~`

which is present in the folder
`C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src`

I guess that I somehow have to tell eclipse / sloeber / gcc where to find SrcWrapper.h but have no clue on how to do that...
Recalling the last issue I had with paths and sloeber I recall that this can be a mess...

Cheers,
Bob

jantje, 2021-09-02:
SrcWrapper.h is a library that should be included in the libraries folder.
It happened to me that the library was added after the first build.
Try to rebuild.
If that does not happen check wether you have add libraries to project active
If that fails add the library manually (adruino->add library to project)

Pestimist, 2021-09-02:
Hi jantje,

oh my dear, I some how used ambiguous wording in my initial post.
I by no means think sloeber is a mess, what I tried to say is that from a former issue I had (and with which you helped me a lot! ) I remember that on windows, playing with PATH is a very fragile thing.

I very much appreciate sloeber, your work and help here, and think it is a very good tool.

back to the initial problem:
only the last step - adding the library manually - helped resolving this
maybe my project got messed up because I switched the platform in midst of developement...

Again, many thanks for helping :)

jantje, 2021-09-02:
Thanks for the clarification.
This issue should not be related to path issue you had.
If you look in the error log (windows->show view->others look for error) you should see whether Sloeber tried to add the library or not. 
Something the indexer misses the "unresolved includes" see more details abouty the include briowser here https://www.youtube.com/watch?v=5Yi5xeeqlsk&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF&index=3

Pestimist, 2021-09-17:
Since this is quite a generic title, I hope it is OK to reopen this issue as I ran into further problems trying to switch my project from AVR to STM32.

It seems to me that the variant.h is not included properly, which leads to many many consecutive errors

here is part of the console output:

```
In file included from C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/STM32L4xx/stm32l4xx_hal_conf.h:4,
                 from C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal.h:30,
                 from C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/stm32l4xx.h:235,
                 from C:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/stm32_def.h:48,
                 from C:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/clock.h:19,
                 from C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino/wiring_time.h:23,
                 from C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino/wiring.h:38,
                 from C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino/Arduino.h:36,
                 from ..\Controller.cpp:1:
C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino/variant.h:15:10: error: #include expects "FILENAME" or <FILENAME>
   15 | #include VARIANT_H
      |          ^~~~~~~~~
```
this repeats several times, and then follows some errors about undefined symbols, which I think relate to this initial error

jantje, 2021-09-17:
Plerase provide the command line that leads to this error and a screenshot of project properties->arduino

Pestimist, 2021-09-20:
Hi jantje,

I'm not sure what you mean with "command line" (well, this clearly shows my noob status, I guess...)
I think it might be this:

```08:11:40 **** Build of configuration Release for project test ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" -j8 all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\STM32Tools\2.0.0/win/busybox.exe" sh "C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/extras/prebuild.sh" "C:\Users\xXxXx\Documents\_fs-Laser\sloeber_fs-Laser\test\Release" "C:\Users\xXxXx\Documents\_fs-Laser\sloeber_fs-Laser\test"
' '
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\Chiller.cpp'
'Building file: ..\Controller.cpp'
'Building file: ..\Diode.cpp'
'Starting C++ compile'
"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\9.3.1-1.3/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb "@C:\Users\xXxXx\Documents\_fs-Laser\sloeber_fs-Laser\test\Release/sketch/build_opt.h" -c -O3 -flto -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IC:\Users\xXxXx\Documents\_fs-Laser\sloeber_fs-Laser\test" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/avr" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/LL" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/usb" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/OpenAMP" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/usb/hid" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/usb/cdc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/STM32L4xx_HAL_Driver/Inc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/STM32L4xx_HAL_Driver/Src" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/STM32L4xx" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/OpenAMP" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/OpenAMP/virtual_driver" -DSTM32L4xx -DARDUINO=10812 -DARDUINO_GENERIC_L496ZGTX -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"GENERIC_L496ZGTX\"" -DVARIANT_H="variant_generic.h"  -DSTM32L496xx -DUSBCON  -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/Core/Include/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/Include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/PrivateInclude"   -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\variants\STM32L4xx\L496Z(E-G)T_L4A6ZGT" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino" -I"C:\Users\xXxXx\Documents\Arduino\libraries\WSWireLib_rep_start-master\utility" -I"C:\Sloeber\arduinoPlugin\libraries\Ethernet\2.0.0\src" -I"C:\Users\xXxXx\Documents\Arduino\libraries\WSWireLib_rep_start-master" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SPI\src" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src" -MMD -MP -MF"Chiller.cpp.d" -MT"Chiller.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\Chiller.cpp"   -o "Chiller.cpp.o"
'Starting C++ compile'
"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\9.3.1-1.3/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb "@C:\Users\xXxXx\Documents\_fs-Laser\sloeber_fs-Laser\test\Release/sketch/build_opt.h" -c -O3 -flto -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IC:\Users\xXxXx\Documents\_fs-Laser\sloeber_fs-Laser\test" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/avr" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32" "-IC:/Sloeber/ar [... cut ...]
   15 | #include VARIANT_H
      |          ^~~~~~~~~```

I could also post the entire console output, but it is quite a lot...

Properties screenshots:

![properites_1](https://user-images.githubusercontent.com/85557472/133962229-20efa67c-885b-44f0-9a18-9b1be1cec67b.png)
![properties_2](https://user-images.githubusercontent.com/85557472/133962226-610367fb-10a7-4996-b6a2-c26f6cfd15f5.png)
![properties_3](https://user-images.githubusercontent.com/85557472/133962228-78bc9e1d-a270-4b32-9624-6a4bca72a377.png)

Pestimist, 2021-09-20:
Don't know if its related, but there seems to be a general issue when switching platforms....
I now changed properties for the project to build for Atmega2560 and for some files it keeps looking for stm32 includes...

```"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR     -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\variants\mega" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\cores\arduino" -I"C:\Sloeber\arduinoPlugin\libraries\Ethernet\2.0.0\src" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\Wire\src" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\SPI\src" -MMD -MP -MF"libraries\SrcWrapper\src\stm32\PortNames.c.d" -MT"libraries\SrcWrapper\src\stm32\PortNames.c.o" -D__IN_ECLIPSE__=1 "C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\PortNames.c"   -o "libraries\SrcWrapper\src\stm32\PortNames.c.o"
'Finished building: C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\Wire\src\Wire.cpp'
"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR     -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\variants\mega" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\cores\arduino" -I"C:\Sloeber\arduinoPlugin\libraries\Ethernet\2.0.0\src" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\Wire\src" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\SPI\src" -MMD -MP -MF"libraries\SrcWrapper\src\stm32\analog.cpp.d" -MT"libraries\SrcWrapper\src\stm32\analog.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\analog.cpp"   -o "libraries\SrcWrapper\src\stm32\analog.cpp.o"
' '
"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR     -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\variants\mega" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\cores\arduino" -I"C:\Sloeber\arduinoPlugin\libraries\Ethernet\2.0.0\src" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\Wire\src" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\SPI\src" -MMD -MP -MF"libraries\SrcWrapper\src\stm32\bootloader.c.d" -MT"libraries\SrcWrapper\src\stm32\bootloader.c.o" -D__IN_ECLIPSE__=1 "C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\bootloader.c"   -o "libraries\SrcWrapper\src\stm32\bootloader.c.o"
C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\PortNames.c:30:10: fatal error: PortNames.h: No such file or directory
 #include "PortNames.h"
          ^~~~~~~~~~~~~
compilation terminated.
make: *** [libraries\SrcWrapper\src\stm32\subdir.mk:63: libraries\SrcWrapper\src\stm32\PortNames.c.o] Error 1
make: *** Waiting for unfinished jobs....
' '
C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\analog.cpp:13:10: fatal error: analog.h: No such file or directory
 #include "analog.h"
          ^~~~~~~~~~
compilation terminated.
make: *** [libraries\SrcWrapper\src\stm32\subdir.mk:70: libraries\SrcWrapper\src\stm32\analog.cpp.o] Error 1
'Building file: C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\clock.c'
C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\bootloader.c:1:10: fatal error: bootloader.h: No such file or directory
 #include "bootloader.h"
          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [libraries\SrcWrapper\src\stm32\subdir.mk:77: libraries\SrcWrapper\src\stm32\bootloader.c.o] Error 1
'Starting C compile'
"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR     -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\variants\mega" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\cores\arduino" -I"C:\Sloeber\arduinoPlugin\libraries\Ethernet\2.0.0\src" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\Wire\src" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.3\libraries\SPI\src" -MMD -MP -MF"libraries\SrcWrapper\src\stm32\clock.c.d" -MT"libraries\SrcWrapper\src\stm32\clock.c.o" -D__IN_ECLIPSE__=1 "C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\clock.c"   -o "libraries\SrcWrapper\src\stm3 [... cut ...]
C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src\stm32\clock.c:13:10: fatal error: backup.h: No such file or directory
 #include "backup.h"
          ^~~~~~~~~~
compilation terminated.
make: *** [libraries\SrcWrapper\src\stm32\subdir.mk:84: libraries\SrcWrapper\src\stm32\clock.c.o] Error 1
"C:/Sloeber/arduinoPlugin/tools/make/make -j8 all" terminated with exit code 2. Build might be incomplete.

08:53:03 Build Failed. 8 errors, 0 warnings. (took 6s.474ms)
```

jantje, 2021-09-20:
>Don't know if its related, but there seems to be a general issue when switching platforms....

Sloeber does not delete all libraries when swapping platform. In your case sloeber does not find an alternative for SrcWrapper and keeps using the stm32 provided library.
If you want to support multiple platforms it is better to use the multiple configurations (or multiple project pointing to the same code) and exclude resources from configuration if needed

jantje, 2021-09-20:
I see that the command
`"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\9.3.1-1.3/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb "@C:\Users\xXxXx\Documents\_fs-Laser\sloeber_fs-Laser\test\Release/sketch/build_opt.h" -c -O3 -flto -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IC:\Users\xXxXx\Documents\_fs-Laser\sloeber_fs-Laser\test" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/avr" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/LL" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/usb" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/OpenAMP" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/usb/hid" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/usb/cdc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/STM32L4xx_HAL_Driver/Inc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/STM32L4xx_HAL_Driver/Src" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/STM32L4xx" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/OpenAMP" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Middlewares/OpenAMP/virtual_driver" -DSTM32L4xx -DARDUINO=10812 -DARDUINO_GENERIC_L496ZGTX -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"GENERIC_L496ZGTX\"" -DVARIANT_H="variant_generic.h"  -DSTM32L496xx -DUSBCON  -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/Core/Include/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/Include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/PrivateInclude"   -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\variants\STM32L4xx\L496Z(E-G)T_L4A6ZGT" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\cores\arduino" -I"C:\Users\xXxXx\Documents\Arduino\libraries\WSWireLib_rep_start-master\utility" -I"C:\Sloeber\arduinoPlugin\libraries\Ethernet\2.0.0\src" -I"C:\Users\xXxXx\Documents\Arduino\libraries\WSWireLib_rep_start-master" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SPI\src" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.0.0\libraries\SrcWrapper\src" -MMD -MP -MF"MyEthernetClient.cpp.d" -MT"MyEthernetClient.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\MyEthernetClient.cpp"   -o "MyEthernetClient.cpp.o"
`
states
`-DVARIANT_H="variant_generic.h"`
and that should be 
`"-DVARIANT_H=\"variant_generic.h\""`
can you check the version of the  platform.sloeber.txt. It should state 
`#Sloeber created workaound file V1.03.test 01 `
If not please upgrade to the nightly as this is probably a duplicate issue

Pestimist, 2021-09-21:
Hi jantje,

thank you for the advice to use multiple configurations (will have to figure out how this works) or different projects with shard code.
Both are new concepts to me and I am always amazed how much I didn't know. :)

I have two platform.sloeber.txt files.
One in a subdirectory of 
> Sloeber\arduinoPlugin\packages\arduino

and one in a subdirectory of 
> Sloeber\arduinoPlugin\packages\STMicroelectronic

both have v1.01.test 20 version

Upgrade to the nightly build resulted in a java/registry error:

```!SESSION 2021-09-21 06:29:24.829 -----------------------------------------------
eclipse.buildId=unknown
java.fullversion=11.0.10+9
JRE 11 Windows 8 amd64-64-Bit Compressed References 20210120_899 (JIT enabled, AOT enabled)
OpenJ9   - 345e1b09e
OMR      - 741e94ea8
JCL      - 0a86953833 based on jdk-11.0.10+9
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=de_DE
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective

!ENTRY org.eclipse.osgi 4 0 2021-09-21 06:29:26.070
!MESSAGE Application error
!STACK 1
java.lang.RuntimeException: Application "org.eclipse.ui.ide.workbench" could not be found in the registry. The applications available are: org.eclipse.cdt.codan.core.application, org.eclipse.cdt.managedbuilder.core.headlessbuild, org.eclipse.equinox.p2.garbagecollector.application, org.eclipse.equinox.p2.reconciler.application, io.sloeber.application.application, org.eclipse.cdt.core.GeneratePDOM, org.eclipse.cdt.core.CodeFormatter, org.eclipse.e4.ui.workbench.swt.E4Application, org.eclipse.e4.ui.workbench.swt.GenTopic, org.eclipse.equinox.app.error, org.eclipse.equinox.p2.publisher.InstallPublisher, org.eclipse.equinox.p2.publisher.EclipseGenerator, org.eclipse.equinox.p2.publisher.ProductPublisher, org.eclipse.equinox.p2.publisher.FeaturesAndBundlesPublisher.
	at org.eclipse.equinox.internal.app.EclipseAppContainer.startDefaultApp(EclipseAppContainer.java:252)
	at org.eclipse.equinox.internal.app.MainApplicationLauncher.run(MainApplicationLauncher.java:33)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)
```

is there a way to downgrade again?
Or can I manually fix this?

jantje, 2021-09-21:
http://eclipse.baeyens.it/how_to.shtml#/n
Always take a backup before upgrading.

You can also download the nightly from the actions option at the top of this page.
The latest version right now is here https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/1251379721

>is there a way to downgrade again?
>Or can I manually fix this?

I don't know

Pestimist, 2021-09-22:
Latest nightly did the trick!
**Thanks!** 

And thanks as well for the backup reminder, I sometimes still forget it ;-)

[maybe unrelated]
Does the nightly use a different gcc version? Because now I run into an error using initializers in struct type declarations that wasn't there before.

jantje, 2021-09-22:
Cool
I'm closing this as a duplicate of #1369 (just a guess)

>Does the nightly use a different gcc version?

Sloeber does not come with any gcc/platform. So the time you start using sloeber decides on the default arduino avr platform that gets installed (the latest ). Other platforms can be installed with json files and other hardware and are completely up to you.

One thing to note is that if you install with json it can be the download only partially succeeded and the you have a consistent incomplete gcc/core/library.
Sloeber caches the downloaded files in the arduinoPlugin/downloads folder. If the problem is consistent over uninstal/install (also delete the tools/provider folder) a rough workaround is uninstall delete the provider folder (in you case E:\platform\arduinoPlugin\packages\STMicroelectronics) and the downloads folder
Then reinstall.

closed 2021-09-22 by jantje