Sloeber/arduino-eclipse-plugin #1379: Build issue with Arduino Nano 33 IOT

pbd9608, 2021-09-05:
I have been working with the Arduino IDE and decided to branch out to a new tool.  I found Sloeber, installed it and tried it first on the Blink project with an Arduino Uno board.  It worked well with that board.  I am working a project with the Arduino Nano 33 IOT board and tried the example SerialCallResponse example and I am getting a Build Error:

C:\Users\...\AppData\Local\Temp\ccDaToI4.s:1446: Error: selected processor does not support `mrc p15,0,r2,c12,c1,0' in Thumb mode
make: *** [core\core\USB\subdir.mk:30: core\core\USB\CDC.cpp.o] Error 1

I am not sure where to even go look for this problem.

jantje, 2021-09-05:
Works fine here
[image]

When I see

>C:\Users...\AppData\Local\Temp\ccDaToI4.s

And as I know Sloeber normally does not do anything outside of the folder it is installed and the workspace folder selected at the start (unless you use version control which uses liks to you version control folders)
So I think there is a problem with you setup.
A couple of screenshots would get us a long way.
Here are my project properties
[image]
Here my arduino preferences
[image]
And my boardsmanager
[image]

If you can upload the screenshots of these settings on your system we can take a deeper look

jantje, 2021-09-05:
FYI the upload protocol on my setting in the project properties is empty because I tested on the nightly which works differently for upload.
Normal in latest release V4.4.0 is to have default

pbd9608, 2021-09-06:
[image]
[image]
[image]

pbd9608, 2021-09-06:
thank you  for the quick response.  Would it be good to remove and reinstall?

jantje, 2021-09-06:
This all looks good. 
I see no reason why reinstall of sloeber is needed. Maybe reinstall of the samd platform could fix the issue but I doubt that for the problem at hand.
Can you provide an expanded view from the project like below?
[image]
note that examples is under libraries which should not be the case. I'll make a issue for that problem.

And the build log in a file?

pbd9608, 2021-09-06:
[image]

jantje, 2021-09-06:
This project view seems to be fine
The malformed url's in the error log should not be there
Can you provide 
[image]
And please the content of the console for the build log in a file

pbd9608, 2021-09-06:
[image]

pbd9608, 2021-09-06:
[Console content.docx](https://github.com/Sloeber/arduino-eclipse-plugin/files/7117928/Console.content.docx)

jantje, 2021-09-07:
I don't know why you added the last line to the third party index url's
That may cause problems, at least it should remove the malformed format messages in the error log

jantje, 2021-09-07:
I have no clue whatsoever is going on on your system.
You say the default sketch works fine but this example not.
This seems contradictory with the fact that the file that fails compiling must be compiled for both the project and with the same compile instructions. As this file is linked it is the exact same file and the command should be the exact same for both projects and is basically the same as on my system (I only see a swap of 2 include paths which I think is acceptable)

It is also strange that there is no red square with a cross on the 2 core folders here as the failing file is in those folders https://github.com/Sloeber/arduino-eclipse-plugin/issues/1379#issuecomment-913854473

Also the error message does not relate back to the cpp file and is marked as an assembler error in ccDaToI4.s

In my setup I failed to find any references to ccDaToI4.s

Somewhere somehow something on your system is wierd and causing problems. I would propose to start from a new sloeber install and document all the steps taken.

pbd9608, 2021-09-07:
> I don't know why you added the last line to the third party index url's
> That may cause problems, at least it should remove the malformed format messages in the error log
The area to choose the microcontroller had boxes that I assumed were check boxes and they would not respond, so I tried to add the Arduino .json file.  I then figured out that the text was a folder that would open when selected.

pbd9608, 2021-09-07:
> I have no clue whatsoever is going on on your system.
> You say the default sketch works fine but this example not.
> This seems contradictory with the fact that the file that fails compiling must be compiled for both the project and with the same compile instructions. As this file is linked it is the exact same file and the command should be the exact same for both projects and is basically the same as on my system (I only see a swap of 2 include paths which I think is acceptable)
> 
> It is also strange that there is no red square with a cross on the 2 core folders here as the failing file is in those folders [#1379 (comment)](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1379#issuecomment-913854473)
> 
> Also the error message does not relate back to the cpp file and is marked as an assembler error in ccDaToI4.s
> 
> In my setup I failed to find any references to ccDaToI4.s
> 
> Somewhere somehow something on your system is wierd and causing problems. I would propose to start from a new sloeber install and document all the steps taken.

I was beginning to think a re-install would be the best way to attack this. 

From your comments below, the Blink project worked on an UNO board.  When I went to the Nano board, I changed to a project that would give me a serial output so that I could verify the serial ports were working.

I will delete and reinstall Sloeber, but it may be next week before I can get to it.  I will let you know how it goes.

Thank you for the help and insight.

Paul

jantje, 2021-12-08:
It has been a while now.
Do you have a update?

closed 2024-02-20 by jantje