Sloeber/arduino-eclipse-plugin #1380: When creating a project based on an example the example folder is in the libraries folder

jantje, 2021-09-06:
Sloeber 4.4 nightly
When working for issue #1379 I noticed that a new project created based on an example has the example folder in libraries.
[image]
In this case this will not cause issues but it should not be there
Simply deleting should do the trick

pbd9608, 2021-09-06:
I don't understand,  What are you asking me to delete?

I thought that is where you put projects or the includes did not work correctly.

jantje, 2021-09-06:
The libraries/examples folder should not be there. You can safely delete it.
In this case it is not needed as the project does not contain troublesome files (ino is only processed if it is in the project root)

closed 2021-12-08 by jantje