Sloeber/arduino-eclipse-plugin #1381: Revisit the update site

wimjongman, 2021-09-10:
The current update site is not self-contained. It means that Sloeber depends on other update sites to be available as well. Since we cannot depend on them, I suggest that we make the update site self-contained meaning that it contains everything that Sloeber needs.

I have been testing with the self-contained repository and it basically does not require anything else but the CDT and the Oscilloscope widget to plot.

wimjongman, 2021-09-11:
@jantje +1 for removing the product entry from the marketplace?

jantje, 2021-09-11:
Seems like a plan.

jantje, 2021-09-11:
There were already references to the product download pages on the plugin marketplace page so I simply deleted the product market page.

jantje, 2021-09-11:
I'm not sure whether the self-contained update site is a good thing.
I think we had that in the past and it caused lots of issues to me.
Now that gnu mcu is part of cdt we are only dependent on the following update sites

- eclipse release 
- nebula
- optionally babel

This only adds nebula to what a standard internationalized eclipse install should have

I personally think that the problems were amongst others cause by product installs that used the old gnu mcu requiring old CDT which is incompatible with sloeber 4.4 requiring CDT 10 or higher
And outdated update site references in sloeber

wimjongman, 2021-09-12:
Yes, this is a nice fix. However, this will not solve the plugins issue. The update sites in the product do not work for the plug-in.

Any problems we had in the past are probably solved with later Tycho and P2 versions.

jantje, 2021-09-12:
>The update sites in the product do not work for the plug-in.

I do not understand this

wimjongman, 2021-09-12:
People report that the plugin update fails. Not the product update. You have been changing the repositories in the .product file. This has no effect on issue https://github.com/Sloeber/arduino-eclipse-plugin/issues/1361

jantje, 2021-09-12:
I also updated the update site
https://github.com/Sloeber/arduino-eclipse-plugin/commit/23c8bf165fc4841de46f81b3fd746f1c6d568e2d
Which also contained gnu mcu outdated update site
https://github.com/Sloeber/arduino-eclipse-plugin/commit/23c8bf165fc4841de46f81b3fd746f1c6d568e2d#diff-1bd341ca9a93db8adedda7fe851eda08d24b7a51cb554aeb4a63ac8b7a94a198L21

wimjongman, 2021-09-13:
I see. That makes sense.

wimjongman, 2021-09-13:
Update site stays at it is then.

I have changed the Github build. All artifacts of a github build are now saved and can be downloaded. 

All products and the update site.

e.g. https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/1224353006

closed 2021-09-13 by wimjongman

jantje, 2021-09-13:
I'm considering releasing 4.4.1 
This because some serious fixes/improvements have been made and your java inclusion may make transition to 4.4.X lots easier

wimjongman, 2021-09-13:
Yes but we have to make sure that it works first.