Sloeber/arduino-eclipse-plugin #1383: Can't install latest stable release on Mac

ricozinn, 2021-09-11:
**Describe your environment**   
I downloaded Sloeber from your website (latest stable version 4.4.0).  The zip file was: V4.4.0_mac64.2021-07-04_06-48-31.zip

❯ java -version
openjdk version "11.0.12" 2021-07-20
OpenJDK Runtime Environment Temurin-11.0.12+7 (build 11.0.12+7)
OpenJDK 64-Bit Server VM Temurin-11.0.12+7 (build 11.0.12+7, mixed mode)

❯ echo $JAVA_HOME
/Library/Java/JavaVirtualMachines/temurin-11.jdk/Contents/Home

OS: MacOS 11.4

**Describe the bug**   
When I try to install it, following your instructions I get this error:
[image]

See to the left of the error the command line steps (wasn't in the ~/downloads folder, used the archive utility, etc).

**To Reproduce**   
1. Go to the website and download the latest. 
2. Move the zip file to my ~/ directory
3. Un-archive the zip file using the archive utility
3. run `sudo xattr -r -d com.apple.quarantine Sloeber.app`
4. Open Sloeber and I get the error

**Expected behavior**   
Hopefully we can get it to install.  Thanks!

**Data to back up your claim**   
[image]

jantje, 2021-09-12:
As far as I know apple does not allow you to run in the quarantine folder

ricozinn, 2021-09-12:
Oh, thanks. What folder should I put it in?

jantje, 2021-09-12:
I don't own an apple computer but as far as I know the "quarantine " folder has limited functionality
Any folder with full functionality should do.

ricozinn, 2021-09-13:
What fixes it is installing the 11.0.11+9 JDK (instead of the latest release 11.0.12+7) and then editing the Info.plist to point Sloeber at that jvm.  You can download that JDK here: https://adoptopenjdk.net/archive.html

Thanks

closed 2021-09-13 by ricozinn