Sloeber/arduino-eclipse-plugin #1384: The included jrexxx java is version 16 which is incompatible with Sloeber

jantje, 2021-09-13:
Issue #1376 includes java as part of the sloeber plugin.
During investigation of #1361 we asked to install the latest version and then the json parsing failed.
I successfully reproduced this issue on my system.
In the log I can see that java 16 is being used. 
As we know from issue  #1339 sloeber does not work with java16 or higher due to the new security rules
As such we should limit the java version that is included to below version 16
I assume this issue comes up because jrexxx released a new version recently

```
!ENTRY io.sloeber.core 4 0 2021-09-13 17:46:05.049
!MESSAGE Failed to parse json file "C:\test\eclipse-java-2021-06-R-win32-x86_64\eclipse\arduinoPlugin\package_index.json"
!STACK 0
java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.lang.String java.io.File.path accessible: module java.base does not "opens java.io" to unnamed module @2693b58e
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:357)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
	at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:177)
	at java.base/java.lang.reflect.Field.setAccessible(Field.java:171)
	at com.google.gson.internal.reflect.UnsafeReflectionAccessor.makeAccessible(UnsafeReflectionAccessor.java:44)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.getBoundFields(ReflectiveTypeAdapterFactory.java:159)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.create(ReflectiveTypeAdapterFactory.java:102)
	at com.google.gson.Gson.getAdapter(Gson.java:458)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.createBoundField(ReflectiveTypeAdapterFactory.java:117)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.getBoundFields(ReflectiveTypeAdapterFactory.java:166)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory.create(ReflectiveTypeAdapterFactory.java:102)
	at com.google.gson.Gson.getAdapter(Gson.java:458)
	at com.google.gson.Gson.fromJson(Gson.java:931)
	at com.google.gson.Gson.fromJson(Gson.java:870)
	at io.sloeber.core.api.PackageManager.loadPackage(PackageManager.java:699)
	at io.sloeber.core.api.PackageManager.loadJson(PackageManager.java:687)
	at io.sloeber.core.api.PackageManager.loadJsons(PackageManager.java:658)
	at io.sloeber.core.managers.InternalPackageManager.startup_Pluging(InternalPackageManager.java:75)
	at io.sloeber.core.Activator$2.run(Activator.java:255)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
```

I think this is related to https://github.com/google/gson/issues/1540 where a solution is proposed which I fail to understand.

jantje, 2021-09-13:
Here is the full log
[the.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7156175/the.log)

jantje, 2021-09-13:
Also wondering ... should this  not be part of the product only.
I mean, now this is part of sloeber plugin and as such forcing all sloeber configs to use this java, unless it can be overridden.
But still, it seems more logic to me this is part of the product.

wimjongman, 2021-09-14:
Java is not part of the plugin, it is only part of the product.

wimjongman, 2021-09-14:
We can try a couple of things:

1. Upgrade to gson 2.8.8
2. Move to another json parser
3. add the --illegal-access variable to sloeber.ini/eclipse.ini : [See this link](https://jaxenter.com/jdk-9-replace-permit-illegal-access-134180.html)

wimjongman, 2021-09-14:
Option 1 is tried first. Please check from the result after Action https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/1233509983 has been built.

wimjongman, 2021-09-14:
I installed a fresh Eclipse with the Eclipse installer and selected java16. Then I installed Sloeber from the update site ^^ After restart Sloeber plugins worked as expected.

However, this was not solved by gson 2.8.8 but by the fact that Eclipse now come by default with 

--illegal-access=permit

When above is added to the Sloeber.ini/Eclipse.ini then Sloeber can continue as usual.

jantje, 2021-09-14:
> 
> 
> Java is not part of the plugin, it is only part of the product.

I was mistaken due to the fact that the eclipse install already used jrexxx. I assumed the eclipse install used my system 11  jre and after sloeber install switched to jrexx. But it was running on jrexx all the time.

@wimjongman thanks for the support.

jantje, 2021-12-05:
This is fixed but as this was a nightly only issue I'll simply close the issue

closed 2021-12-05 by jantje