Sloeber/arduino-eclipse-plugin #1390: Eclipse includes are not more resolved for ESP32 v2.0.0

ArminJo, 2021-09-29:
Setting ESP Platform version to 1.0.6 

[image]
is OK with a lot of include paths:

[image]
Setting ESP Platform version to 2.0.0 

[image]

results in missing all (internal) includes, which lead to tons of errors of unresolved inclusions.

[image]
:

In platform.sloeber.txt in 1.0.6 there is:
`compiler.cpp.flags=-std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib ${compiler.warning_flags} -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c`

in 2.0.0
`compiler.cpp.flags=${compiler.cpp.flags.${build.mcu}}}`
with:
`compiler.cpp.flags.esp32=-mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -O2 -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fexceptions -fno-rtti  -MMD -c`

changing it for 2.0.0 to
`compiler.cpp.flags=-mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -O2 -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fexceptions -fno-rtti  -MMD -c`

works for me 😀 but maybe you can keep an eye on it.

Thanks for caring!

jantje, 2021-09-29:
`compiler.cpp.flags=${compiler.cpp.flags.${build.mcu}}}`

Seems to me there is a closing } to much

ArminJo, 2021-09-29:
This is a copy paste modify error made by me :-(

jantje, 2021-09-29:
I seem to fail to reproduce the issue :-(

'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"e:\platform\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4-dev-2313-gc69f0ec32" -DESP_PLATFORM   "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/config" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/newlib/platform_include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/freertos/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/freertos/port/xtensa/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_hw_support/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_hw_support/include/soc" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/heap/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/log/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/lwip/include/apps" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/lwip/include/apps/sntp" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/lwip/lwip/src/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/lwip/port/esp32/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/soc/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/soc/esp32" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/soc/esp32/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/hal/esp32/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/hal/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/hal/platform_port/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_rom/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_rom/esp32" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_rom/include/esp32" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_common/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_system/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_system/port/soc" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_system/port/public_compat" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp32/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/xtensa/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/xtensa/esp32/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/driver/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/driver/esp32/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_pm/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_ringbuf/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/efuse/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/efuse/esp32/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/vfs/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_wifi/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_event/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_netif/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_eth/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/tcpip_adapter/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_phy/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_phy/esp32/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/app_trace/include" "-Ie:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/sdk/esp32/include/esp_ti [... cut ...] DUINO_VARIANT="doitESP32devkitV1"  -DESP32 -DCORE_DEBUG_LEVEL=1    -DARDUINO_USB_CDC_ON_BOOT=0 "@L:\test\runtime-EclipseApplication_eng\sss\Release/build_opt.h"   -I"E:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0\variants\doitESP32devkitV1" -I"E:\platform\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0\cores\esp32" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
'Finished building: ..\sloeber.ino.cpp'

ArminJo, 2021-09-29:
Sorry, building is not the problem, but having tons of errors in the editor and therefore not finding declarations is the problem.

jantje, 2021-09-29:
Have you tried to rebuild the index?

ArminJo, 2021-09-29:
1000 times :-(

ArminJo, 2021-09-29:
Switch between the two versions and you see the effect in the includes in the project explorer wiew.
The indexer has no chance without include paths.

jantje, 2021-09-30:
Did you try opening and closing the project?

jantje, 2021-09-30:
I see no includes folder when in version 2.0.0 but I do when in 1.0.6
I'm very confused

ArminJo, 2021-09-30:
I found no place in .projects or similar files responsible for this. So I think you set the include path programmatical e.g. during change of version.

jantje, 2021-09-30:
Things are way more complicated than Sloeber just setting include paths.
1) Sloeber sets and removes user maintained include paths when it adds/removes libraries
2) Sloeber provides CDT with a "discovery command" that CDT uses to discover "compiler specs" defines, include path,....(I don't know much about this because it is part of CDT not sloeber)

The problem you are experiencing is in the second case.
You can see the discovery command in project properties->C/C++ general->preprocessor include paths macros etc->providers->arduino compiler settings
[image]
If you check the "allocate a console in the console view" and change the command (adding a space is enough) you will see the command executed after selecting apply and close.
Note that $COMMAND} is tool dependent (c, c++, assembly, linker, archiver ) and resolves to recipes.XX.YY in the platform.txt file of the selected platform.

When you change the board Sloeber tells CDT that all is invalid now; resulting in CDT triggering a rediscover and then reindex.

jantje, 2021-09-30:
I simply fail to get the situation you are referring to (with the latests nightly and esp version 2.0.0
[image]

ArminJo, 2021-10-01:
Ok I found the problem. 😀 
**It is the missing build_opt.h file!** leading to the error
`xtensa-esp32-elf-gcc: error: E:\WORKSPACE_SLOEBER\Talkie_ESP32\_10Bit_Koefficients/build_opt.h: No such file or directory` and aborting the includes discovery.
When you switch to 2.0.0 the bild directory is empty and the prebuild.X.pattern did not run, which create an empty build_opt.h file.  -> discovery fails.
If you create this file, e.g. by performing a build and then run rediscovery manually (is there an easier way than inserting spaces?), then the missing includes appear (it took a few 10 seconds before the appeareance in my project Explorer).
What a nasty hack of the ESP32 software 😭 
Any ideas?
Thanks so far for your effort.
Armin

jantje, 2021-10-05:
I've been thinking about this problem and I propose as workaround (and as final solution) to delete the
"@{build.opt.path}" 
from the platform.txt
This will disable the functionality to modify the command line; but in Sloeber you have other options to modify the command line.
If you really want this feature you can add "@{build.opt.path}" in the sloeber command line options to re-enable the default esp32 V2.0.0 (and probably later) behaviour.
@ArminJo @me-no-dev what do you think?

ArminJo, 2021-10-06:
I hope for #1391 to be fixed 😀 . 
Meanwhile this seems to be the only solution, at least it works for me 😀 .

jantje, 2021-10-06:
I expect the fix for #1391 to take time.
I do not have lots of time to spend on Sloeber right now and this workaround/fix is simple and easy.

jantje, 2021-12-08:
As #1391 is now fixed the ESP32 problems should all be fixed. So I'm thinking of making a new release.
Some testing would be appreciated