Sloeber/arduino-eclipse-plugin #1392: Sloeber 4.4 won't update to nightly build 4.4.1

stokni, 2021-10-01:
1. Installed Fresh installation of Sloeber (V4.4.0_win64.2021-07-04_06-48-31) - Working
2. Update via Stable release to new version of 4.4.0 - Working
3. Selected Nightly builds repository in update sites 
4. Check for software update: 4.4.1.2021092300221 available and able to install - Working
5. After installation, error message java.lang.RuntimeException: Application "org.eclipse.ui.ide.workbench" could not be found in the registry. 

[1633120519548.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7270039/1633120519548.log)

Tried this on a clean W10 VM as well - same result.

ospilos, 2022-01-23:
The same problem persists.
To reproduce:
1. Install lateste Sloeber product.
2. Enable Nightly builds repo
3. Update io.sloeber.product
4. Restart Sloeber to apply updates
5. Dialog window opened: „An error occured. See the log file“ + path to the log file.
6. After pressing OK, Sloeber closes.
[1642967943395.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7921289/1642967943395.log)
[1642967987525.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7921290/1642967987525.log)

The problem manifests itself both on Win7x64 and Win10x64.

Before September 2021 it worked fine.

jantje, 2022-01-23:
>Updating io.sloeber.plugin only works fine.

interesting
Did you try the workaround of #1361

ospilos, 2022-02-02:
Yeah, I tried, but it does nothing - the plugin updates fine, but as soon as I try to update the "product", the Eclipse no longer starts.