Sloeber/arduino-eclipse-plugin #1393: OTA port and password wrong for ESP32

ArminJo, 2021-10-01:
I try to upload with OTA to an ESP32.

With the Arduino IDE all went fine, the resulting command was:
`E:\Elektro\arduino\portable\packages\esp32\hardware\esp32\2.0.0/tools/espota.exe -i 192.168.42.36 -p 3232 --auth= -f C:\Users\Armin\AppData\Local\Temp\arduino_build_37431/esp32-cam-webserver.ino.bin `

in Sloeber the resulting command was:
`Launching: E:\Elektro\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.0/tools/espota.exe -i 192.168.42.36 -p 8266 --auth=otaPassword -f E:\WORKSPACE_SLOEBER\ESP32_Cam\Release/ESP32_Cam.bin `

I added entries in Properties > C/C++ Build > Environment 
[image]

for the recipe:
[image]

but nothing changed, I see the same random? password and port even after a Sloeber restart or removing the password with:

[image]

Uploading via serial connection is sucessful.

jantje, 2021-10-02:
Did you see #1387?

ArminJo, 2021-10-02:
Yes, I have read it before, but I did not quite understand it.

jantje, 2021-10-03:
In short
You need the latest nightly (you can download the product via actions above)

If you use literally "ArduinoOTA.setPassword" in setup() in active code (no comments nor defined out).
That password in the code will be used to upload via OTA 
Else you need to provide a password in project properties->arduino or no password will be used causing the command to fail.

To enable passwords changes (OTA upload security will not allow you to upload a sketch with a new password as Sloeber will use the new password and OTA requires the old password) or to have more control over the upload command you can create a target as explained here https://github.com/Sloeber/arduino-eclipse-plugin/issues/1387#issuecomment-922553968

ArminJo, 2021-10-03:
Hi Jaintje,
thanks for the explanation!

I want to compile ESP32-cam example and the approriate lines are:
```
char otaPassword[] = "";
...
        if (strlen(otaPassword) != 0) {
            ArduinoOTA.setPassword(otaPassword);
            Serial.printf("OTA Password: %s\n\r", otaPassword);
        } else {
            Serial.printf("\n\rNo OTA password has been set! (insecure)\n\r\n\r");
        }
```
Maybe you should check at least for double quotes?

> Else you need to provide a password in project properties->arduino or no password will be used causing the command to fail.

In arduino it does not fail with an empty password.

The question left is, where to set the port.
192.168.42.36:**3232** does not wok for me
Why not use network.port (see above) for it?

jantje, 2021-10-03:
You need to have
```
setup(){
....
ArduinoOTA.setPassword("");
....
}
```

The port can be set in the same way

ArminJo, 2021-10-06:
Is this a right conclusion for ESP32?

- On Sloeber you must eventally change you code to use OTA.
- On Arduino it works, at least without an OTO password.
- Change the` tools.esptool_py.upload.network_pattern` in platform.txt to: `tools.esptool_py.upload.network_pattern=
{network_cmd} -i "{serial.port}" -p "3232" "--auth=" -f "{build.path}/{build.project_name}.bin"`

closed 2021-10-06 by ArminJo

jantje, 2021-10-06:
>On Sloeber you must eventally change you code to use OTA.

Maybe, but that is not intentionally. If so that needs fixing.