Sloeber/arduino-eclipse-plugin #1396: Sloeber/eclipse is re-indexing every project of a workspace at startup

hreintke, 2021-10-16:
Hi Jantje,

Back to arduino development after a pause.

Moving from 4.3 to 4.4 and/or git-master.
In git-master there is a fix to the stm32 VARIANT_H issue. 
But for this I see similar but not 100% identical behavior on startup of sloeber.

Main issue is that a startup, all projects in the workspace are re-indexed by eclipse.
In my case, with a workspace with around 65 projects, it takes 5-10 minutes of high cpu usage before all gets quiet.
The first 30-60 seconds sloeber/eclipse is not even responsive. 

What I notice in the eclipse .log file, that it probably is triggered by an sloeber update of the index.

These are the messages from the .log at startup (4.4 version). 
I use a small test workspace with two projects.

```
!ENTRY org.eclipse.egit.ui 2 0 2021-10-16 13:15:08.685
!MESSAGE Warning: The environment variable HOME is not set. The following directory will be used to store the Git
user global configuration and to define the default location to store repositories: 'C:\Users\Herman'. If this is
not correct please set the HOME environment variable and restart Eclipse. Otherwise Git for Windows and
EGit might behave differently since they see different configuration options.
This warning can be switched off on the Team > Git > Confirmations and Warnings preference page.

!ENTRY io.sloeber.core 8 0 2021-10-16 13:15:29.841
!MESSAGE Index of project changed :ArdISP

!ENTRY io.sloeber.core 8 0 2021-10-16 13:15:33.201
!MESSAGE Index of project changed :BluetoothTest
```

What could be the reason for the index change at startup ?

hreintke, 2021-10-17:
Did not know about open/close project, thx.

On startup.
After migrating the workspace, I removed the 
```
                // This is not install job but to migrate 4.3 to 4.4
                // A refresh seems to trigger the code at a convenient time
                final IWorkspaceRoot workspaceRoot = ResourcesPlugin.getWorkspace().getRoot();
                for (IProject curProject : workspaceRoot.getProjects()) {
                    if (curProject.isOpen()) {
                        try {
                            curProject.refreshLocal(DEPTH_INFINITE, monitor);
                        } catch (@SuppressWarnings("unused") CoreException e) {
                            // ignore
                        }
                    }
                }
```
from `runInstallJob()`

As once the upgrade is done this is not necessary anymore. Correct ?

Sloeber starts without indexing all open projects, does not show platform/port information in the project explorer view.

-> can you point me to the location in the code where that is added ?

Selecting a project (pushing the > in front of the name) triggers the indexing of that project.
It does not remove the "release directory" from the project.

But : Doing a build of that project triggers the re-indexing of all other open projects in the workspace.
After that, only part of the projects have their platform/port info displayed. 

-> Any idea why / where the trigger happens ?

In the logfile there are notifications like this : 
```
!ENTRY org.eclipse.cdt.core 4 0 2021-10-17 14:50:15.153
!MESSAGE Error: Cannot run program "-D__IN_ECLIPSE__=1": Launching failed
!STACK 0
java.io.IOException: Cannot run program "-D__IN_ECLIPSE__=1": Launching failed
	at org.eclipse.cdt.utils.spawner.Spawner.exec(Spawner.java:367)
	at org.eclipse.cdt.utils.spawner.Spawner.<init>(Spawner.java:98)
	at org.eclipse.cdt.utils.spawner.ProcessFactory.exec(ProcessFactory.java:111)
	at org.eclipse.cdt.core.CommandLauncher.execute(CommandLauncher.java:190)
	at org.eclipse.cdt.internal.core.BuildRunnerHelper.build(BuildRunnerHelper.java:274)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runProgramForLanguage(AbstractBuiltinSpecsDetector.java:799)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForLanguage(AbstractBuiltinSpecsDetector.java:727)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForEachLanguage(AbstractBuiltinSpecsDetector.java:592)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector$1.runInWorkspace(AbstractBuiltinSpecsDetector.java:518)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:42)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
```

Most certainly these are for projects which have no platform.

jantje, 2021-10-17:
>As once the upgrade is done this is not necessary anymore. Correct ?

As the comment says so that should be correct. However the code has never been tested without it so it may be needed.
Also the curProject.refreshLocal is the same as pressing F5 so if something doesn't work you can always press F5 as workaround.

>Sloeber starts without indexing all open projects, does not show platform/port information in the project explorer view.

I assume this is something you see after you did the code change you mentioned.
Which kind of proves the code does more than only trigger the migration.

The problem is as follows (as I recall it):
When a project is opened Sloeber needs to internally configure itself. This is done inside 
`public synchronized boolean configure(ICProjectDescription prjCDesc, boolean prjDescWritable)`
in SloeberProject class
This code will do the migration if needed.
This code should be called at the project open event (send by eclipse) but both the ICProjectDescription and the project need to be writeable for the migration to be successful which they aren't at that point in time. I'm not sure they need to be writeable after the migration but I don't think they should be.
The ICProjectDescription needs to be writeable to remove the environment variables and the project needs to be writeable to add the .sproject file (with the conceptual content of the environment variables)

The listener for the open project is resourceChangeListener 
In the method resourceChanged you will see it calls curSloeberProject.sloeberCfgChanged(); which calls Helpers.deleteBuildFolder(myProject, activeConfig.getName());

but curSloeberProject.sloeberCfgChanged(); should not be called for newly opened projects because above the call it states
```
        // no sloeber.cfg files have been modified
        if (changedSloeberCfgFiles.size() == 0)
            return;
```
This test should be true and the method should return (and as such stop the deletion of the release folder)
So the question is why is changedSloeberCfgFiles not empty

jantje, 2021-10-17:
I have been looking at the resourceChanged  code and .... there is room for improvement :-(

hreintke, 2021-10-31:
Thx, found all the info you were referring to.
Now focussing on sloeberProject.configure()

hreintke, 2021-11-07:
Still busy, as you said it's not easy.
See a lot of things happening but figuring out why it happens.

Also comparing 4.3 & git master. Also 4.3 does (sometimes ?) indexing at startup.

Is there somewhere a list on which listeners can be installed/used in the eclipse/cdt environment ?

jantje, 2021-11-07:
>Still busy, as you said it's not easy.

I know. It is a very steep learning curve. I've been doing this for over 10 years now but I still feel like a newbee.

>See a lot of things happening but figuring out why it happens.

Is about the hardest thing with all this triggering and asynchronous actions and lots of programmers not really knowing what they do (That is about 100% of us)

>Is there somewhere a list on which listeners can be installed/used in the eclipse/cdt environment ?

I fear not. 
But honestly I do think that /io.sloeber.core/src/io/sloeber/core/listeners/IndexerController.java is your best bet.
If the sloeber project is not yet configured; indexing should be postponed.
Maybe you can configure the sloeber project at that time if not you should postpone it and call /io.sloeber.core/src/io/sloeber/core/listeners/IndexerController.index at the end of configure

FYI I'm currently working on the java compatibility issue and doing some major refactoring. I'm trying to stay away from the stuff you are looking at so we do not have merging problems.

jantje, 2021-11-28:
Any progress?
I ask because I have a huge change in regards to java compatibility. I do not want to bother you with merging your changes ;-)

jantje, 2021-12-08:
I have fixed #1391 which causes a reindexing for sure because all the output files are deleted.
I'm not sure whether Sloeber runs an indexing now at the project open but I'm also not sure if CDT indexes a project at project open.
The root cause was that project open and configuration changes were both mapped on the same call in resourceChangeListener.
When there is a config change Sloeber deletes the build folder as that may be needed.
To fix the problem I implemented a different strategy for config changes and project open.
The other changes are related to making a cleaner cut between CDT (holding the configurations) and Sloeber (holding the arduino info related to the configurations) and old sloeber way (all is in cdt configurations)

If you have some spare time; please check whether this fixes your problem

jantje, 2021-12-19:
This bugzilla passed by today. May be related
https://bugs.eclipse.org/bugs/show_bug.cgi?id=575408

hreintke, 2022-03-19:
Hi Jantje,
Did another try to find/analyze the issue but have to admit that I have no progress at all.
No idea how to get any further. 
I will stick to the workaround of having most of the projects closed to minimize the effect,

Wil stop working on it, but will restart if you want specific options to be evaluated.

jantje, 2022-03-19:
It is a very complicated issue requiring in depth knowledge of eclipse cdt and sloeber (and maybe even other plugins installed)
Thinking about this. If CDT does not persist it's indexing info (and why should it) an indexing is necessary at startup.
However multiple indexing starts were made and still may be made.

hreintke, 2022-03-19:
CDT does persist indexing info.

It is in <workspace>\.metadata\.plugins\org.eclipse.cdt.core\<projectname>\<timestamp>.pdom

jantje, 2022-03-19:
>Have you seen that before and/or do you know how to solve ?

I have seen similar issues on my build server.

This is a typical eclipse dependency problem probably related to new eclipse release where @wimjongman  normally comes to rescue.

closed 2024-02-20 by jantje