Sloeber/arduino-eclipse-plugin #1401: degree symbol bug

neorob85, 2021-10-31:
I'm using sloeber 4.4 on windows 10 to program a esp8266 (nodemcu) with latest arduino libraries (pubsubclient 2.8.0 and esp8266 platform 3.0.2).

Take this instruction:

```
  mqttclient.publish("topic", "symbol °C");

```

If it's compiled with arduinoIDE or platformIO, everything works on the right way.
On the contrary if it's compiled with sloeber, is trasmitted the weird "ef bf bd" character instead of the right degree symbol (c2 b0)

Any idea how to resolve it?

jantje, 2021-11-01:
This is probably related to the file format setup in eclipse.
If you select a file and right click it select properties->resource you can see the type of file that is used
[image]
Most likely that will be CP1215 inherited from the workspace. You can change the workspace settings in windows->preferences->general->workspace
[image]

neorob85, 2021-11-02:
I set up "Text file encoding" from "Cp1252" to "UTF-8" and now seems working as expected.
Thankyou @jantje ! :)

jantje, 2021-11-02:
I labelled this as a known limitation but basically the problem is that eclipse has plenty of file encoding options to store files so you have to set the file encoding to a toolchain "supported" value.
So basically it is a known limitation because eclipse is over qualified 😮

closed 2021-11-02 by jantje