Sloeber/arduino-eclipse-plugin #1402: indexing

nicoverduin, 2021-11-03:
Once upon a long time ago.... Jantje explained how to avoid unresolved functions. Mea culpa but I forgot how (must be age). I know it was reindexing  but how do I set those options?
Regards
Nico

jantje, 2021-11-03:
Hi Nico
Are you using the new ESP32 version? I ask because there is a discovery (add that word to your vocabulary ;-) ) issue
If you have a multiconfig environment you need to go to windows->preferences->C/C+->indexer. Though I heard the default changed it may be ...
[image]
If all this fails I need more specific info

nicoverduin, 2021-11-04:
Seems to work now. So we can leave it as it is.
thx

closed 2021-11-04 by nicoverduin