Sloeber/arduino-eclipse-plugin #1403: board jsons not parsing

MangyDogUK, 2021-11-03:
[image]

Running latest sloeber...
been having a nightmare getting it to install on a fresh windows install, with JDK installed.

MangyDogUK, 2021-11-03:
Sorry for not the great bug report... im really tired from loads of system rebuilding over the last few days... This is the last thing to get working.

jantje, 2021-11-03:
Do you have the log file? Seems very similar to #1384

MangyDogUK, 2021-11-03:
[error.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7471270/error.log)

MangyDogUK, 2021-11-03:
Adding --illegal-access=permit to eclipse ini didnt work.

jantje, 2021-11-04:
According to the log you are on java 17 so parsing json files is not going to work.Which is confirmed by the loggings
```
!ENTRY io.sloeber.core 4 0 2021-10-31 23:17:51.186
!MESSAGE Failed to parse json file "G:\embedcpp-2021-09\eclipse\arduinoPlugin\package_index.json"
!STACK 0
java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.lang.String java.io.File.path accessible: module java.base does not "opens java.io" to unnamed module @13140c17
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:354)
	at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
```
It can be the workaround no longer works in java17.

I also see `java.io.IOException: Unable to resolve plug-in "io.sloeber.application".` which basically says your install is not ok.

MangyDogUK, 2021-11-05:
OK, its not ok :D so what do i need to do to get it working... as the usual methods dont seem to be working.

jantje, 2021-11-05:
use java 11

MangyDogUK, 2021-11-05:
[image]

still getting this after removing JDK17 and installing 11...

MangyDogUK, 2021-11-05:
[error.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7487916/error.log)
fresh install fresh workspace

jantje, 2021-11-05:
How do you install?

MangyDogUK, 2021-11-05:
Downloaded latest eclipse from eclipse.... Selected the embedded option as I use some of the tools it come with.
Then i go to the market place and install sloeber from there.

I have also tried installing via install new software and the link off your site.

jantje, 2021-11-05:
The marketplace install on latest eclipse will hit #1361
We do not support this path as it is very error prone.
FYI Part of CDT embedded is (and has been for a long time) in Sloeber

>I have also tried installing via install new software and the link off your site.

I do not understand what you mean with that statement
Here is where you should be http://eclipse.baeyens.it/stable.php?OS=Windows
or you can go directly to the releases here on github https://github.com/Sloeber/arduino-eclipse-plugin/releases

MangyDogUK, 2021-11-05:
[image]
installed via the install new software.... And getting that again.

jantje, 2021-11-05:
You need java 11

MangyDogUK, 2021-11-05:
JAVA11 is installed  Sent from Mail for Windows From: jantjeSent: 05 November 2021 22:46To: Sloeber/arduino-eclipse-pluginCc: MangyDogUK; AuthorSubject: Re: [Sloeber/arduino-eclipse-plugin] board jsons not parsing (Issue #1403) You need java 11

jantje, 2021-11-05:
please provide error log

MangyDogUK, 2021-11-05:
[error.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7489359/error.log)

MangyDogUK, 2021-11-05:
looking at the log it says java 16, but i did install jdk 11

jantje, 2021-11-06:
As I said it is using java 16. 
You can modify the sloeber.ini to point to the java you want it to use
```
-vm
C:/Program Files/Java/jdk-11.0.4/bin/javaw.exe
```
By the way ... you are not using the product as that comes with a compatible java

MangyDogUK, 2021-11-06:
i think its already set
[image]

MangyDogUK, 2021-11-06:
oh wait i see it now

MangyDogUK, 2021-11-06:
seems to be working now

uzi18, 2021-11-06:
@MangyDogUK what is your solution?

closed 2021-12-05 by jantje