Sloeber/arduino-eclipse-plugin #1404: Error in Variant.h with STM32

nicoverduin, 2021-11-04:
In de STM32duino library an error occurs when I create an stm32 project. There is an environment variable called VARIANT.H which is opened in a file called variant.h. The problem that occurs is that VARIANT_H is used in a statemetn #include VARIANT_H.
This environment variable comes from build.variant_h. And in the source code it is translated without the extra double quotes. Below is a fix for it.

The cause is in a file called sloeber.txt in the folder : D:\Sloeber_4.4\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.1.0.

I modified the line :

build.info.flags=-D${build.series} -DARDUINO=${runtime.ide.version} -DARDUINO_${build.board} -DARDUINO_ARCH_${build.arch} "-DBOARD_NAME="${build.board}"" -DVARIANT_H="${build.variant_h}"

into

build.info.flags=-D${build.series} -DARDUINO=${runtime.ide.version} -DARDUINO_${build.board} -DARDUINO_ARCH_${build.arch} "-DBOARD_NAME="${build.board}"" -DVARIANT_H=""${build.variant_h}""

Now it works like a charm
Regards
Nico

jantje, 2021-11-04:
IMHO This has been reported in #1378 and confirmed as fixed in the nightly.

nicoverduin, 2021-11-04:
I just downloaded the stable 4.4 a couple of days ago. So I need to download the nightly build as well?

jantje, 2021-11-04:
No I should release 4.4.1 but there are some problems :-(
I think the problem is fixed though.

nicoverduin, 2021-11-04:
I downloaded the nightly build and overwritten 4,4 with it. Deleted the whole STM32 package and re-installed it. Works like a charm. Thanks

closed 2021-11-04 by nicoverduin