Sloeber/arduino-eclipse-plugin #1406: Sloeber not linking files correctly. Causing missing globals or functions.

MangyDogUK, 2021-11-11:
Sorry hard one to explain, but since rebuilding sloeber with eclipse. Im now getting really strange errors with missing functions and variables....
For a long while it didnt even understand what a uint8_t was. 
I have re included libraries using the "arduino/add library to selected project" but its still refusing to correctly look up local header files within the project folder.

eg
[image]

which is inside this header include

[image]

These are both within the same main .ino file.

I should point out before the migration to the new computer (which is why i was reintalling sloeber) my project was building find. Nothing has changed in that regard.
The linker is just borked.

jantje, 2021-11-11:
First of. 
1 Is it a linking problem or a indexer/discovery problem?
2 as you are on a new computer. Are you using the same platforms/libraries?

MangyDogUK, 2021-11-11:
yeah ive set the same libraries and hardware links in the properties, but like i say it wont even link correctly to header includes that are in the same project folder.

As for linker index or discovery problem...... uhh all the above?

MangyDogUK, 2021-11-11:
Just gone through a bunch of stuff and tripple checked my code... And still having problems with linking or resolving not working correctly....

Also it builds in Arduino IDE just fine.

jantje, 2021-11-11:
You provide way to little info to help you out.
What I would do is compare the link command in the 2 environments you have

MangyDogUK, 2021-11-11:
Well tbh i really have no idea how sloeber or even eclipse works behind the scenes. 
But im running out of ideas of how this is meant to work. I have added the libraries to the libraries list the hardware to the hardware list. I also have to both add the direct path to each library im using in the properties list or sloeber doesnt see the libraries in the library folder. 
[image]

I then use the attach library to project tool...

Honestly I dont know what else im meant to do. 
Like I say it does compile in Arduino IDE. So my code is fine. just something in eclipse isnt...

jantje, 2021-11-11:
> sloeber doesnt see the libraries in the library folder.

That is probably because the library folder contains code (*.cpp *.c *.h file)

What is the console output after a clean build?

MangyDogUK, 2021-11-12:
00:41:15 **** Build of configuration Release for project Voyager_Tricorder_V2_Body ****
"C:\\eclipse\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\eclipse\arduinoPlugin\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10812 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ "-IF:\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple" "-IF:\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/include" "-IF:\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/stm32f1/include" "-IF:\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/usb/stm32f1" "-IF:\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/usb/usb_lib"   -I"F:\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple" -I"F:\Arduino\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103c" -I"F:\Arduino\hardware\Arduino_STM32\STM32F1\libraries\Wire\utility" -I"F:\Arduino\hardware\Arduino_STM32\STM32F1\libraries\Wire" -I"F:\Arduino\hardware\Arduino_STM32\STM32F1\libraries\SPI\src" -I"F:\Arduino\libraries\Adafruit_MPR121_STM32" -I"F:\Arduino\hardware\Arduino_STM32\STM32F1\libraries\EEPROM" -I"F:\Arduino\libraries\Mangy_Dog-framedSerial" -I"F:\Arduino\hardware\Arduino_STM32\STM32F1\libraries\FreeRTOS900" -I"F:\Arduino\libraries\Gameduino2" -I"F:\Arduino\libraries\Mangy_Dog_Adafruit_TLC5947" -I"F:\Arduino\hardware\Arduino_STM32\STM32F1\libraries\FreeRTOS900\utility" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
In file included from ..\page_handling.h:929:0,
                 from ..\sloeber.ino.cpp:39:
..\page_molescan.h:71:23: error: macro "min" passed 3 arguments, but takes just 2
     { low, mid, high });
                       ^
In file included from ..\page_handling.h:929:0,
                 from ..\sloeber.ino.cpp:39:
..\page_molescan.h: In function 'void setMoleSpin()':
..\page_molescan.h:70:21: error: 'min' was not declared in this scope
   uint8_t minimum = min(
                     ^
make: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 1
"C:/eclipse/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

00:41:16 Build Failed. 3 errors, 0 warnings. (took 480ms)

MangyDogUK, 2021-11-12:
I should point out I have about 14 thousand lines of code!

Like I say something just isnt right in the project setup or linker.. its turning into a nightmare.

jantje, 2021-11-12:
Now I know you are
1) probably suffering from #1395
2) Using Sloeber without using the boards manager to install boards
3) have plain coding issues in file page_molescan.h on line 71
4)Not even close to linking your code
5) not reading instructions and as such causing suffering on yourself and others
:-(

MangyDogUK, 2021-11-12:
Well I can't use the boards manager to install Roger Clark stm32duino core. But adding its fir to the hardware list worked before. No coding issue in molscan. That's part of the linking issue not finding the min function.

What instructions?

jantje, 2021-11-12:
[image]

MangyDogUK, 2021-11-12:
Can you explain to me how one is meant to reproduce my bug where linking isnt working as it should?
Its not an action on my part thats caused this problem...

Im currently in limbo on the edge of finishing my project. And sloeber simply isnt working as it should be...

I've installed jdk11 i have set the ini file to link to that. All my libraries are added in the library properties, as well as box ticked in the add library to project menu. 

I have tried this both on the sloeber standalone version and the eclipse plugin version. both have the same problem.

The problem is not my code nor the libraries, as I have said, it builds perfectly fine in Arduino IDE, as it did on my old machine that I had sloeber running on. Only difference I can think is I was using eclipse from last year on the old machine. 

So if you say this is not a bug... Then what is it?

MangyDogUK, 2021-11-12:
Ok It think I have fixed the problems I was having. It may have been some redefines of functions within the SPI.h i had as part of my core library for rogerclarks stm32duino...
The thing is. Nothing has changed since it was previously working. its just only now for some reason its started throwing errors about these possible redefine conflicts. 

It was also showing errors like Serial1 is undefined.. Or random and delay arguments are ambiguous. Simple basic functions like those. 

Maybe something has changed about how the indexer and compiler collects and reorganises the code since the version I had before. And highlighted this problem I had already in the code backend. 

But its honestly been a really annoying time spending nearly a whole week trying to get this to work. All the time Arduino IDE was compiling it just fine not showing any errors. So I was sure it wasnt my code thats the issue. 

For now the fix has been to comment out those redefined functioned from SPI.h in rogers core as I have done with the Min function. So its now reverting to the original calls of those functions in both arduino.h / wireish and stdlib.

jantje, 2021-12-05:
I guess this can be closed now. If not please provide proper info to what the remaining issue is

closed 2021-12-05 by jantje