Sloeber/arduino-eclipse-plugin #1407: Sloebber does not honnor programmers.txt like Arduino

MiBiMiFlo, 2021-11-16:
I'm using sloebber as Eclipse plugin on Windows 10 to program new ATTiny and ATMega MCUs with DxCore and MegaTinyCore from SpenceKonde by using a SerialUPDI programmer (with pyUPDI).

This setup is working fine in Ardiono IDE but upload is not working in Sloebber. The error message is as follows:

`avrdude: Can't find programmer id "uart"`

In my setup avrdude should not be used as programmer, but Sloebber always uses avrdude regardless of the selected programmer.

It seems that Sloebber only honors the programmer defined in boards.txt (boards.sloebber.txt) but not the one in programmers.txt file. After changing the programmer in boards.txt file, the upload works correctly.

The code for the cores can be found at:
https://github.com/SpenceKonde/megaTinyCore
https://github.com/SpenceKonde/DxCore

jantje, 2021-11-16:
did you try
[image]

jantje, 2021-11-16:
Also note that -for reasons I do not recall- I excluded the package form sloeber testing

https://github.com/Sloeber/arduino-eclipse-plugin/blob/533582d3fcd92d4eaade891ef4129e82426c6a89/io.sloeber.tests/src/io/sloeber/core/CreateAndCompileDefaultInoOnAllBoardsTest.java#L62

MiBiMiFlo, 2021-11-17:
I don't have this action in the Arduino Menu:

[image]

I have downloaded standalone Sloeber IDE to ensure I have the most recent version, but still the item is not available.
This menu item (best would be to place it in the toolbar as well) would be great and completely solve this issue.

Thanks!

jantje, 2021-11-17:
This is something that has changed recently and might only be in the nightly.
Basically before (and as you point out still n the latest release) the upload does upload when you select default in project properties->arduino. If you selected a programmer Sloeber did upload using programmer.
This is not compliant to arduino IDE so I changed that (together with adding the programmer burn bootloader functionality)
[image]
Now there is no longer a default but you can leave the programmers blank
[image]

As I expect this change will be confusing I thought you were confused by this. Seems you are not.
I propose you try to get confused by using the nightly ;-) (available here on git via the actions menu)

>This menu item (best would be to place it in the toolbar as well) would be great and completely solve this issue.

The great thing of eclipse is that you can do this yourself. As most people do not use the programmers I think this is the way to go.
Here is a explanation how.
https://stackoverflow.com/questions/8648215/how-to-customize-a-toolbar-in-eclipse#8648370
I tried, and it didn't work. Can you create a issue?

MiBiMiFlo, 2021-11-17:
I just downloaded nightly and the "Upload the sketch using the selected programmer" works great!

I have created an issue for the missing feature to add this action to toolbar as well.

closed 2021-11-17 by MiBiMiFlo

flexphperia, 2022-08-21:
Any news with this issue? Unfortunately night build is not running for me.

jantje, 2022-08-21:
>Any news with this issue? Unfortunately night build is not running for me.

I should release the nightly.
If you can't run the nightly you should create a issue on that problem.