Sloeber/arduino-eclipse-plugin #1408: Sloabber does not resolve {serial.port} in programmers.txt

MiBiMiFlo, 2021-11-16:
It seems that sloebber in the recent version does not resolve {serial.port} correctly when used in programmers.txt.

I'm using DxCore from SpenceKonde to program new megaAVR MCUs. This is working fine in Arduino IDE, but seems to make issues in Sloebber. When doing an upload with python based SerialUPDI programmer, sloebber does not resolve the placeholder {serial.port} as defined in programmers.txt. The error mesasge is as follows:

`Launching: C:\Temp\Sloeber\arduinoPlugin\packages\megaTinyCore\tools\python3\3.7.2-post1/python3 -u C:\Temp\Sloeber\arduinoPlugin\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py -t uart -u {serial.port} -b 230400 -d avr128db28 --fuses 5:0b11001001 7:0x00 8:0x00 -fC:\Temp\Sloeber\weiss_temp\sloeber-workspace\TestTiny\Release/TestTiny.hex -a write 
Output:
Arduino <---> pymcuprog bridge by Quentin Bolsee and Spence Konde
Version 1.1.0 - May 2021
Using serial port {serial.port} at 230400 baud.
Target: avr128db28
Set fuses: ['5:0b11001001', '7:0x00', '8:0x00']
Action: write
File: C:\Temp\Sloeber\weiss_temp\sloeber-workspace\TestTiny\Release/TestTiny.hex
pymcuprog.serialupdi.physical - ERROR - Unable to open serial port '{serial.port}'
Traceback (most recent call last):`

When changing {serial.port} to ${serial.port} it is working correctly. As even in original Arduino cores the {serial.port} is used without $ I expect this is an issue with Sloebber.

The DxCore core can be found at https://github.com/SpenceKonde/DxCore

jantje, 2021-11-16:
I guess this is a duplicate of #1362

MiBiMiFlo, 2021-11-17:
you are right, sorry I searched but did not find this one.

closed 2021-11-17 by MiBiMiFlo

jantje, 2021-11-17:
No problem