Sloeber/arduino-eclipse-plugin #1409: FeatureRequest - Allow to use "Upload the sketch using the selected programmer" in Tool Bar

MiBiMiFlo, 2021-11-17:
The Action "Upload the sketch using the selected programmer" from Arduino Menu should be available in the Tool Bar as well. This would be great!

jantje, 2021-11-17:
For one reason or another the actions are not visible when in windows->perspective->customize perspective->action set availability
One available there it should be possible to create toolbars/menus to execute the actions.

Note for "upload sketch using the selected programmer"
This action is actually a make target and can as sutch also be started in the project->config folder(normally Release)->targets

jantje, 2021-12-30:
Seems the tooolbar manipulation in eclipse is not working the way I expected.
Therefore I added a extra toolbar that allows you to access nearly all Sloeber menu items. you can hide them via menu:window->perspective->customize perspective

The idea is that this extra toolbar is hidden by default and can be enabled by advanced users.
However I don't get the hidden by default to work.
It seems there is a long lasting bug in eclipse (reported 2013) https://bugs.eclipse.org/bugs/show_bug.cgi?id=400217
This article explains how to work around it
https://jonathanjwright.wordpress.com/2014/07/15/eclipse-4-4-0-perspective-based-show-hide-of-tool-bar-items/
it states
`The first problem I encountered was the fact that expressions are not evaluated for items added directly to the main tool bar (see Bug 400217). By ‘added directly’ I mean added as children of the tool bar element in the e4xmi application model. Luckily, when added as ToolBar Contributions visible-when expressions are evaluated as expected.`
I fail to understand what this means.
As CDT toolbars get hidden/shown by switching perspective; I checked the CDT code but that leaded to nothing as CDT uses the deprecated action set.

@wimjongman @rlogiacco 
Maybe one of you know how to get the toolbar visibility to work. Don't spend to much time one it as one may very rightfully say it is a bad idea to hide this toolbar by default.

PS I also added a show project properties->sloeber button to the Sloeber toolbar in https://github.com/Sloeber/arduino-eclipse-plugin/commit/4311ae4b5cfbb0377c7407609bf0492eb4ad97ab

wimjongman, 2021-12-30:
> Seems the tooolbar manipulation in eclipse is not working the way I expected.
Therefore I added a extra toolbar that allows you to access nearly all Sloeber menu items. you can hide them via menu:window->perspective->customize perspective

I don't think an extra toolbar is a good idea. Certainly not for the reasons stated.

I have downloaded the latest code but I don't see the extra toolbar. 

What is it exactly that you want to achieve? It goes beyond the initial request of this issue.

> The Action "Upload the sketch using the selected programmer" from Arduino Menu should be available in the Tool Bar as well.

jantje, 2021-12-30:
This is what I have in my Sloeber
[image]
As you can see I plenty more buttons
[image]

jantje, 2021-12-30:
What is it exactly that you want to achieve? It goes beyond the initial request of this issue.

    The Action "Upload the sketch using the selected programmer" from Arduino Menu should be available in the Tool Bar as well.

Well I was thinking .... If I add 1 menu item I can just as well add more ...

wimjongman, 2022-01-01:
Ok, this makes sense. I would make it easy and always show the toolbar. Let the user hide it in perspective customization if they don't want it.

wimjongman, 2022-01-01:
May I suggest using branches and pull requests instead of inserting commits directly into the master branch?

jantje, 2022-01-01:
>Ok, this makes sense. I would make it easy and always show the toolbar. Let the user hide it in perspective customization if they don't want it.

Ok. I think that is how it is now.

jantje, 2022-01-02:
@MiBiMiFlo 
I'm closing this as fixed in the nightly. Please check and reopen if this doesn't suit your needs.

closed 2022-01-02 by jantje