Sloeber/arduino-eclipse-plugin #1410: Allow Burn bootloader with selected programmer

MiBiMiFlo, 2021-11-17:
I noticed that Burn Bootloader works possibly with the same upload mechanism as Upload, but not with the new "selected programmer" feature.

I'm not sure if this makes sense at all, but it would be great if either Burn Bootloader uses the selected programmer or if an additional action "Burn Bootloader with selected programmer" could be added.

jantje, 2021-12-05:
Seems I missed this issue. Sorry for the delay.
Can you give an example of the "actual behaviour" versus the "wanted behaviour"?

There are still 2 good reasons to keep arduino IDE around
1) installation/update of drivers
2)see whether it is a sloeber issue or a platform issue

jantje, 2022-01-02:
After waiting for 25 days I close the issue. 
Feel free to reopen when answering the questions in the previus post

MiBiMiFlo, 2022-01-05:
Sorry I did not find the time to respond more early.

In my setup I have a different tool selected for programming (not the default), additionally I always only use Upload with Programmer, not the normal upload. The current behavior does not honor the selected programmer when using "Burn the Bootloader", but always refers to the board's default bootloader tool. (as far as I understand).

I did some more deep dives in the internals of sloeber and found a workaround that explains my request quite well.

When I create a makefile.defs file in my project root with the following definition:
`
bootloader.tool=${program.tool}
`
This fixes my issue and is more aligned with the way how Arduino IDE honors the selected programmer for burning bootloader.

I'm still not sure if the actual behavior is really wanted, as you always need a programmer to burn the bootloader. As far as I understand the normal upload process requires a bootloader to be installed already.

Hope this clarifies my request.

jantje, 2022-01-09:
Thanks for the update.
It looks like abug like you describe it.
I havn't use the burn bootloader much but I did do dome burning to test, so there probably are bugs in the code but we need to be very carefully making changes as testing is pretty hard (due to difficulties to test for instance due to lack of hardware).

Anyway I tried here using
[image]

When I tried 'burn bootloader" the command obvious failed (as I do not have the bootloader connected)
[image]

Then I created the makefile.defs as you explained and ran again
[image]

As far as I can see the command that was executed is exactly the same.
So I do not see a issue.
Can you provide the same information from your system?

MiBiMiFlo, 2022-01-10:
Hi,

to reproduce this, you need to select a programmer that either does not use avrdude or at least requires different parameter set.
I'm using megaTinyCore from SpenceKonde and selected the SerialUPDI programmer:

[image]

When I run burn Bootloader the following is executed:
[image]

After I created the makefile.defs and added `bootloader.tool=${program.tool}` I get the following:
[image]

I guess the bootloader tool is taken from `atxy4.bootloader.tool=avrdude` in boards.txt, but there is no way to change this. Arduino IDE honors the selected programmer and the `serialupdi.bootloader.tool=serialupdi` from programmers.txt in this case.

jantje, 2022-01-10:
Can you provide the json url you used to install this platform?

It looks to me that the problem is as follows:
Sloeber first gets the environment variables from platform .txt then from programmers.txt then from boards.txt
https://github.com/Sloeber/arduino-eclipse-plugin/blob/15767ce4ee5755bcf331db7ece1a0d297ca2b933/io.sloeber.core/src/io/sloeber/core/api/BoardDescription.java#L807
https://github.com/Sloeber/arduino-eclipse-plugin/blob/15767ce4ee5755bcf331db7ece1a0d297ca2b933/io.sloeber.core/src/io/sloeber/core/api/BoardDescription.java#L823
https://github.com/Sloeber/arduino-eclipse-plugin/blob/15767ce4ee5755bcf331db7ece1a0d297ca2b933/io.sloeber.core/src/io/sloeber/core/api/BoardDescription.java#L828

This means that boards.txt environment variables will overwrite the platform.txt variables
This complies with your perception
>I guess the bootloader tool is taken from atxy4.bootloader.tool=avrdude in boards.txt

From the point of view of programming and bootloading; processing programmers.txt last seems a good idea. 
From the ppoint of view of uploading and building; processing the boards.txt last seems most natural.
As in Sloeber the build, programming and burning bootloader all start from the same makefile; changing the order of txt processing is a all or nothing situation.

Though I do not really see risk in processing programmers.txt after boards.txt it does mean: "lots of testing"

FYI If you can build sloeber you can change the order and test your scenario.

MiBiMiFlo, 2022-01-10:
I used the following json to install megaTinyCore in version 2.5.4:

http://drazzy.com/package_drazzy.com_index.json

Yes I understand your point and it explains the observed behavior. If I will find some time I will try to change the order and do a test.

As I said I have a working workaround and it's not urgent for me, still it differs from the way how Arduino handles the files and might confuse others as well, when the feature is released. 
I see two possible options here:

1. Change the order of config file processing to match Arduino (possibly with a Global Option to select one way or the other?)
2. Add an Option in Project Settings next to programmer Selection where bootloader programmer can be selected per Project

Anyway I'm totally happy that I can use Sloeber, so thanks a lot for this great tool!

MiBiMiFlo, 2022-01-10:
I just noticed this option:
[image]

Not sure what it does, but it sounds like it should align to Arduino IDE and therefore would possibly make sense to also align this more with the Arduino style when enabled?

jantje, 2022-02-16:
This one should be fixed now.

closed 2022-02-16 by jantje