Sloeber/arduino-eclipse-plugin #1414: failed to compile Arduino_core_STM32 V2.1 based projects

graham-r, 2021-11-22:
Using Sloeber v4.4 under Fedora 34  I am getting the #include "srcwrapper.h" not found again

I have tried using earlier STM version 2.0 but the problem persists.

The code is an existing project that compiles for Uno/Nano with no problems

wimjongman, 2021-11-22:
Are you referring to a previous issue?

graham-r, 2021-11-22:
Hi Wim

Yes, this was apparently resolved before (for Windows users ?), IIRC that issue was closed.

wimjongman, 2021-11-22:
You have to provide more information if you want us to say something meaningful. Did you not see the issue template when you filed a new issue?

http://www.catb.org/~esr/faqs/smart-questions.html

jantje, 2021-11-22:
The main reasonns for `#include "srcwrapper.h" not found ` are
1) starting the build before the indexer finsihed so the library is not yet added. Work aroiund : build again
2) the automatic adding of llibraries is disabled in windows->preferences->arduino. work around add srcwrapper lib manually

graham-r, 2021-11-22:
Jante

I have started a new simple (example:blink) project - and the srcwrapper problem still happened until I set auto install/install libraries on #include - this seems to have sorted the srcwrapper problem (so far) - thanks for that !

The new problem (poss an STM problem) is several 
"error: #include expects "FILENAME" or <FILENAME> or #include VARIANT_H" (see attached file)

This may have something to do with the properties/settings - I shall experiment when I can.
[stm_g071rb_variant_h.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/7582560/stm_g071rb_variant_h.txt)

jantje, 2021-11-22:
That is issue #1369 
Workaround described in #1369 
or use a version smaller than V2
or use nightly

graham-r, 2021-11-22:
Thanks for the reply - I was going to try exactly that (adding a separate compile option).

I will see tomorrow.

graham-r, 2021-11-23:
Jante

I used your fix (-DVARIANT_H=\"${build.variant_h}\") and it works (usually !) 

But now I am getting this new problem - the same blink example works fine with STM32CubeProg.

Failed to run the Upload recipe

Cannot run program "" (in directory "/home/graham/work/code/etc/etc"): error=2, No such file or directory
failed to upload "L/testSTM/${workspace_loc:/testSTM}/Release/testSTM.hex"

jantje, 2021-11-23:
probably #1364

jantje, 2021-12-05:
I guess this is fixed.
If not please reopen with remaining issue

closed 2021-12-05 by jantje