Sloeber/arduino-eclipse-plugin #1416: Not able to upload code on (some) ESP32 boards

aiotech-pub, 2021-11-28:
I have a ESP32 Wemos D1 R32 board that is compatible with EspDuino. I am able to build an example in Sloeber (the elf file, not the hex and the bin also because it is just a copy of the elf file) but I am not ble to upload the code. This is the returned error:
upload failed for "L/BlinkESP32/BlinkESP32}/Release/BlinkESP32.hex"
esptool
tools.esptool.upload.pattern : not found in the platform.txt file

I tryed to change the board to see if something changes and I discovered that produce the sam or other errors mainly related to the esptool (wrong parameters or not found). Instead, generally, there are no probolems to build the application.

jantje, 2021-11-28:
Please provide the info requested for a bug report

**Before creating a bug report check for existing issues**  
Make sure you are on the latest and greatest Sloeber version. We do not support older version.  
Here is a list of [known limitations](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue++label%3A%22status%3A+known+limitation%22+)   
Here is a list of [common issues ](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue+label%3A%22Hot+issue%22+)   
Here is the list of all [open issues](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue+is%3Aopen)  
Here is a list of issues that have been [fixed but are not yet available in a release](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=label%3A%22status%3A+fixed+in+nightly%22+)  

---

**Describe your environment**   
Do you use the Sloeber product or plugin?   
Sloeber version number  
OS: [e.g. windows 10]   
The url of the json that you used to download the platform  
A image of the project properties arduino showing the board settings   

**Describe the bug**   
A clear and concise description of what the bug is.

**To Reproduce**   
Steps to reproduce the behavior:   
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

provide the name of the board and the platform version you are using.  
Provide the json url and a screenshot of project properties->arduino  

**Expected behavior**   
A clear and concise description of what you expected to happen.  

**Data to back up your claim**   
If applicable, add screenshots, console output and/or logs to help explain your problem.   

**Additional context**   
Add any other context about the problem here.

aiotech-pub, 2021-11-30:
1. I am using Sloeber product on Win7 and Win10 (2 installations).
2. 2. The board is a Wemos D1 R 32 that is espduino32 compatible. In the list of supported boards it is not present but the most adequate are ESP32 Dev Module and  DOIT EspDuino32
3. The json file for the ESP32 boards: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
4. The 1st problem is that building the project does not make the .hex files. At the moment I solved that adding this line as project post build step:
xtensa-esp32-elf-objcopy -O ihex ${BuildArtifactFileName}.elf ${BuildArtifactFileBaseName}.hex
5. when I press the upload button this is the output (so I suppose there is a configuration error somewhere):
Inizio upload
Caricamento in corso per il progetto "BlinkESP32" mediante "esptool_py"

In avvio: C:\Prog\Sloeber\arduinoPlugin\packages\esp32\tools\esptool_py\3.1.0/esptool.exe 
Output:
esptool.py v3.1
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
               ...

esptool.py v3.1 - ESP8266 ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    merge_bin           Merge multiple raw binary files into a single file for
                        later flashing
    version             Print esptool version
    get_security_info   Get some security-related data

optional arguments:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}, -c {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,usb_reset,no_reset,no_reset_no_sync}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset,no_reset_stub}, -a {hard_reset,soft_reset,no_reset,no_reset_stub}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM
                        bootloader. Some features will not be available.
  --trace, -t           Enable trace-level output of esptool.py interactions.
  --override-vddsdio [{1.8V,1.9V,OFF}]
                        Override ESP32 VDDSDIO internal voltage regulator (use
                        with care)
  --connect-attempts CONNECT_ATTEMPTS
                        Number of attempts to connect, negative or 0 for
                        infinite. Default: 7.
Esecuzione del comando "3.1.0/esptool.exe" completata

jantje, 2021-11-30:
Can you provide
A image of the project properties arduino showing the board settings

aiotech-pub, 2021-12-02:
1. I am using Sloeber product on Win7 and Win10 (2 installations).
2. 2. The board is a Wemos D1 R 32 that is espduino32 compatible. In the list of supported boards it is not present but the most adequate are ESP32 Dev Module and  DOIT EspDuino32
3. The json file for the ESP32 boards: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
4. The 1st problem is that building the project does not make the .hex files. At the moment I solved that adding this line as project post build step:
xtensa-esp32-elf-objcopy -O ihex ${BuildArtifactFileName}.elf ${BuildArtifactFileBaseName}.hex
5. when I press the upload button this is the output (so I suppose there is a configuration error somewhere):
Inizio upload
Caricamento in corso per il progetto "BlinkESP32" mediante "esptool_py"

In avvio: C:\Prog\Sloeber\arduinoPlugin\packages\esp32\tools\esptool_py\3.1.0/esptool.exe 
Output:
esptool.py v3.1
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
               ...

esptool.py v3.1 - ESP8266 ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    merge_bin           Merge multiple raw binary files into a single file for
                        later flashing
    version             Print esptool version
    get_security_info   Get some security-related data

optional arguments:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}, -c {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,usb_reset,no_reset,no_reset_no_sync}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset,no_reset_stub}, -a {hard_reset,soft_reset,no_reset,no_reset_stub}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM
                        bootloader. Some features will not be available.
  --trace, -t           Enable trace-level output of esptool.py interactions.
  --override-vddsdio [{1.8V,1.9V,OFF}]
                        Override ESP32 VDDSDIO internal voltage regulator (use
                        with care)
  --connect-attempts CONNECT_ATTEMPTS
                        Number of attempts to connect, negative or 0 for
                        infinite. Default: 7.
Esecuzione del comando "3.1.0/esptool.exe" completata

> 
[image]

aiotech-pub, 2021-12-02:
I think the problem is somewhere in the espressif files, something relted to the boards and tools setup.

jantje, 2021-12-02:
The json file https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json does not contain version 2.0.1
[image]
And that reminded me of #1364

aiotech-pub, 2021-12-02:
Nothing changed, this is the output of the upload:
Inizio upload
Caricamento in corso per il progetto "BlinkESP32NoDelay" mediante "esptool_py"

In avvio: C:\Prog\Sloeber\arduinoPlugin\packages\esp32\tools\esptool_py\3.1.0/esptool.exe 
Output:
esptool.py v3.1
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
               ...

esptool.py v3.1 - ESP8266 ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    merge_bin           Merge multiple raw binary files into a single file for
                        later flashing
    version             Print esptool version
    get_security_info   Get some security-related data

optional arguments:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}, -c {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,usb_reset,no_reset,no_reset_no_sync}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset,no_reset_stub}, -a {hard_reset,soft_reset,no_reset,no_reset_stub}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM
                        bootloader. Some features will not be available.
  --trace, -t           Enable trace-level output of esptool.py interactions.
  --override-vddsdio [{1.8V,1.9V,OFF}]
                        Override ESP32 VDDSDIO internal voltage regulator (use
                        with care)
  --connect-attempts CONNECT_ATTEMPTS
                        Number of attempts to connect, negative or 0 for
                        infinite. Default: 7.
Esecuzione del comando "3.1.0/esptool.exe" completata

aiotech-pub, 2021-12-02:
And in any case there is 2.0.1 in the json file:
{
  "packages": [
    {
      "name": "esp32",
      "maintainer": "Espressif Systems",
      "websiteURL": "https://github.com/espressif/arduino-esp32",
      "email": "hristo@espressif.com",
      "help": {
        "online": "http://esp32.com"
      },
      "platforms": [
        {
          "name": "esp32",
          "architecture": "esp32",
          "version": "2.0.1",
          "category": "ESP32",
          "url": "https://github.com/espressif/arduino-esp32/releases/download/2.0.1/esp32-2.0.1.zip",
          "archiveFileName": "esp32-2.0.1.zip",
          "checksum": "SHA-256:3a7cd46ba47990dd37fbe02b7f0a910dd5cc7af1d190350b69d320ed36cd6b41",
          "size": "148976301",
          "help": {
            "online": ""
          },
          "boards": [
            {
              "name": "ESP32 Dev Board"
            },
            {
              "name": "ESP32-S2 Dev Board"
            },
            {
              "name": "ESP32-C3 Dev Board"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "esp32",
              "name": "riscv32-esp-elf-gcc",
              "version": "gcc8_4_0-esp-2021r2"
            },
            {
              "packager": "esp32",
              "name": "xtensa-esp32-elf-gcc",
              "version": "gcc8_4_0-esp-2021r2"
            },
            {
              "packager": "esp32",
              "name": "xtensa-esp32s2-elf-gcc",
              "version": "gcc8_4_0-esp-2021r2"
            },
            {
              "packager": "esp32",
              "name": "esptool_py",
              "version": "3.1.0"
            },
            {
              "packager": "esp32",
              "name": "mkspiffs",
              "version": "0.2.3"
            },
            {
              "packager": "esp32",
              "name": "mklittlefs",
              "version": "3.0.0-gnu12-dc7f933"
            }
          ]
        },

aiotech-pub, 2021-12-02:
In my opinion the button to upload the ihex file to ESP32 calls esptool.exe without or missing or wrong arguments.

jantje, 2021-12-02:
Yes like in #1364
I'm not sure why sloeber didn't show the 2.0.1 version :-(
Need to check that.
But I'm pretty sure this is a duplicate of #1364

jantje, 2021-12-02:
Now version 2.0.1 is there.
I have no clue what went wrong.
Anyway if that is an issue it should be discussed  with another issue number
I'm pretty sure this is a duplicate of  #1364

aiotech-pub, 2021-12-03:
I mostly agree with you but the solution of #1364 as is does not work always for me. This is what I did:
using esp32 lib for Arduino 2.0.0
1. delete platform.sloeber.txt
2. modify platform.txt (at the end of the file, modification evidenced by ###) in this way:
###tools.esptool_py.upload.pattern="{path}/{cmd}" {upload.pattern_args}
tools.esptool_py.upload.pattern="{path}/{cmd}" {tools.esptool_py.upload.pattern_args}
###
3. start sloeber
4. build and load

in my case it works if I select ESP32 Dev Module, instead if I select the DOIT EspDuino32 it does not work, the uploaf fails with this error:
      tools.esptool.upload.pattern : not found in the platform.txt file
but as you may check in the attached file the entry is present so I think there are much more troubles around.
I am checking if switching to 2.0.1 release solves or not these problems.
[platform.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/7650109/platform.txt)
.

jantje, 2021-12-07:
I do not find tools.esptool.upload.pattern in the platform.txt link you provided
[image]

The fix for #1364 is very generic and applies for more cases then mentioned in the issue

kagouraki, 2021-12-15:
i think there is a bug when sloeber makes the configuration files for esp32
I am running linux. 
in file /home/labcnc/eclipse/cpp-2021-09/eclipse/arduinoPlugin/packages/esp32/hardware/esp32/2.0.1/platform.sloeber.txt there is a typo.
line tools.esptool_py.upload.pattern=python "${tools.esptool_py.path}/${tools.esptool_py.cmd}" ${upload.pattern_args}
should be :
tools.esptool_py.upload.pattern=python "${tools.esptool_py.path}/${tools.esptool_py.cmd}" ${tools.epstool_py.upload.pattern_args}

aiotech-pub, 2021-12-16:
Yes, you are rigth, in Windows there is the same problem also if the syntax is a little bit different (the commented row was wrong):
#tools.esptool_py.upload.pattern="{path}/{cmd}" {upload.pattern_args}
tools.esptool_py.upload.pattern="{path}/{cmd}" {tools.esptool_py.upload.pattern_args}

aiotech-pub, 2021-12-16:
@jantje because iti is called tools.esptool_py:
#tools.esptool_py.upload.pattern="{path}/{cmd}" {upload.pattern_args}
tools.esptool_py.upload.pattern="{path}/{cmd}" {tools.esptool_py.upload.pattern_args}

aiotech-pub, 2021-12-16:
I opened another ticket because I found that Sloeber also does not compile some examples.

jantje, 2021-12-16:
As I stated before. This is a duplicate of #1364
If you think it is not prove it.

closed 2021-12-16 by jantje

abecikxp, 2022-07-21:
> In my opinion the button to upload the ihex file to ESP32 calls esptool.exe without or missing or wrong arguments.

yes, you have right. Upload button only runs esptool.exe  without params. 
I have same problem :(