Sloeber/arduino-eclipse-plugin #1419: After upgrade from V4.3 to v4.4.1, Eclipse crashes when attempting to access serial port

mduchain, 2021-12-09:
Hello team,
I just updated the Sloeber plugin from v4.3 to v4.4.1 using Eclipse's built-in update procedure.
Now, whenever Eclipse attempts to access the serial port, whether from the serial monitor, or trying to upload a sketch, Eclipse crashes.
I'd rather not reinstall eclispe from scratch. Is it possible to manually revert to 4.3?
TIA
Mike

**Environment:**
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2020-09 (4.17.0)
Build id: 20200910-1200
OS: Windows 10, v.10.0, x86_64 / win32, Build 18362 (10.0.18362.1916)
Java version: 14.0.2

This the stack trace produced by Eclipse (full log file attached):

...
EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=25692, tid=15308

JRE version: OpenJDK Runtime Environment (14.0.2+12) (build 14.0.2+12-46)
Java VM: OpenJDK 64-Bit Server VM (14.0.2+12-46, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)

---------------  S U M M A R Y ------------

Command Line: -Dosgi.requiredJavaVersion=11 -Duser.home=C:\dev -Dosgi.instance.area.default=C:\dev\projects -Dosgi.configuration.area=C:\dev\.eclipse -XX:+UseG1GC -XX:+UseStringDeduplication --add-modules=ALL-SYSTEM -Dosgi.requiredJavaVersion=11 -Dosgi.dataAreaRequiresExplicitInit=true -Xms256m -Xmx2048m --add-modules=ALL-SYSTEM -Declipse.p2.max.threads=10 -Doomph.update.url=http://download.eclipse.org/oomph/updates/milestone/latest -Doomph.redirection.index.redirection=index:/->http://git.eclipse.org/c/oomph/org.eclipse.oomph.git/plain/setups/ 

Host: Intel(R) Core(TM) i7-9850H CPU @ 2.60GHz, 12 cores, 31G,  Windows 10 , 64 bit Build 18362 (10.0.18362.1916)
Time: Thu Dec  9 20:22:29 2021 W. Europe Standard Time elapsed time: 320 seconds (0d 0h 5m 20s)

---------------  T H R E A D  ---------------

Current thread (0x00000284e25ae000):  JavaThread "main" [_thread_in_native, id=15308, stack(0x000000fe90a00000,0x000000fe90b00000)]

Stack: [0x000000fe90a00000,0x000000fe90b00000],  sp=0x000000fe90afd490,  free space=1013k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  0x000000007110b5db

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  io.sloeber.core.api.Serial.connect(I)V+28
j  io.sloeber.core.api.Serial.connect()V+2
j  io.sloeber.core.api.Serial.<init>(Ljava/lang/String;ICIFZ)V+99
j  io.sloeber.core.api.Serial.<init>(Ljava/lang/String;I)V+9
j  io.sloeber.ui.monitor.views.SerialMonitor.connectSerial(Ljava/lang/String;I)V+30
j  io.sloeber.ui.monitor.views.SerialMonitor$7.run()V+37
j  org.eclipse.jface.action.Action.runWithEvent(Lorg/eclipse/swt/widgets/Event;)V+1
j  org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(Lorg/eclipse/swt/widgets/Event;Z)V+361
j  org.eclipse.jface.action.ActionContributionItem.lambda$5(Lorg/eclipse/swt/widgets/Event;)V+54
j  org.eclipse.jface.action.ActionContributionItem$$Lambda$381.handleEvent(Lorg/eclipse/swt/widgets/Event;)V+5
J 17425 c2 org.eclipse.swt.widgets.EventTable.sendEvent(Lorg/eclipse/swt/widgets/Event;)V (592 bytes) @ 0x00000284ee7501d4 [0x00000284ee750140+0x0000000000000094]
J 18414 c2 org.eclipse.swt.widgets.Display.runDeferredEvents()Z (109 bytes) @ 0x00000284ee7af3b0 [0x00000284ee7af360+0x0000000000000050]
J 17894 c1 org.eclipse.swt.widgets.Display.readAndDispatch()Z (94 bytes) @ 0x00000284e8628454 [0x00000284e8627e60+0x00000000000005f4]
j  org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run()V+564
j  org.eclipse.core.databinding.observable.Realm.runWithDefault(Lorg/eclipse/core/databinding/observable/Realm;Ljava/lang/Runnable;)V+12
j  org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(Lorg/eclipse/e4/ui/model/application/MApplicationElement;Lorg/eclipse/e4/core/contexts/IEclipseContext;)Ljava/lang/Object;+57
j  org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(Lorg/eclipse/e4/ui/model/application/MApplicationElement;)V+20
j  org.eclipse.ui.internal.Workbench.lambda$3(Lorg/eclipse/swt/widgets/Display;Lorg/eclipse/ui/application/WorkbenchAdvisor;[I)V+393
j  org.eclipse.ui.internal.Workbench$$Lambda$141.run()V+12
j  org.eclipse.core.databinding.observable.Realm.runWithDefault(Lorg/eclipse/core/databinding/observable/Realm;Ljava/lang/Runnable;)V+12
j  org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Lorg/eclipse/swt/widgets/Display;Lorg/eclipse/ui/application/WorkbenchAdvisor;)I+16
j  org.eclipse.ui.PlatformUI.createAndRunWorkbench(Lorg/eclipse/swt/widgets/Display;Lorg/eclipse/ui/application/WorkbenchAdvisor;)I+2
j  org.eclipse.ui.internal.ide.application.IDEApplication.start(Lorg/eclipse/equinox/app/IApplicationContext;)Ljava/lang/Object;+113
j  org.eclipse.equinox.internal.app.EclipseAppHandle.run(Ljava/lang/Object;)Ljava/lang/Object;+138
j  org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(Ljava/lang/Object;)Ljava/lang/Object;+85
j  org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(Ljava/lang/Object;)Ljava/lang/Object;+79
j  org.eclipse.core.runtime.adaptor.EclipseStarter.run(Ljava/lang/Object;)Ljava/lang/Object;+99
j  org.eclipse.core.runtime.adaptor.EclipseStarter.run([Ljava/lang/String;Ljava/lang/Runnable;)Ljava/lang/Object;+132
v  ~StubRoutines::call_stub
j  jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Ljava/lang/reflect/Method;Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+0 java.base@14.0.2
j  jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+100 java.base@14.0.2
j  jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+6 java.base@14.0.2
j  java.lang.reflect.Method.invoke(Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+59 java.base@14.0.2
j  org.eclipse.equinox.launcher.Main.invokeFramework([Ljava/lang/String;[Ljava/net/URL;)V+201
j  org.eclipse.equinox.launcher.Main.basicRun([Ljava/lang/String;)V+159
j  org.eclipse.equinox.launcher.Main.run([Ljava/lang/String;)I+4
v  ~StubRoutines::call_stub
...

[hs_err_pid25692.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/7688025/hs_err_pid25692.log)

jantje, 2021-12-09:
We always advice to take a backup when upgrading for this kind of issues.
I'm no expert on eclipse downgrades but as it seems you are not using the Sloeber product you should be able to de-install sloeber and install the version you want
keep in mind 4.4.0 no longer stores it's data in CDT so the workspace has been upgraded which will make downgrading more difficult. Check the .sProject file in your project folder to see the old settings to reconfigure.

You say you updated to 4.4.1 but there is no 4.4.1 official release. Did you upgrade to the nightly?
This is important as 4.4.0 uses the same jssc as 4.3 whereas the nightly uses a newer jssc

mduchain, 2021-12-10:
I checked the installed version by going to
Help -> Eclipse Market Place, tab "Installed", mentions v4.4.1.
nightly? that was no choice of mine - I believe.
i used the upgrade procedure from inside eclipse: 
Help -> Check for Updates 
and 4.4.1 was there to grab.

I will try the v4.4.0 and let you know.
thx

[image]

jantje, 2021-12-10:
That is confusing, indeed.
If you installed from the marketplace you should have 4.4.0
If you go to help->about->installation detail->plugins you get the version number which contains the version and the build date
[image]

mduchain, 2021-12-14:
Sorry didn't get back to you earlier.
The mistery thickens. Looking at plugin i get all 4.3.3 ...
[image]

Yet, in the installed application of the same form, It's 4.4.0 ...
[image]

This is messy  :)

Despite all this, Sloeber is working perfectly: compiling ok, new projects get created as they should. Only exception the serial.
I'm using OTA for downloading the software, and putty for the occasional serial output. 

Since I only use Eclipse for ESP32 projects, I'm debating internally to start afresh using the product this time.
When i read the installation instructions on your site last year, It wasn't clear - at least to me - that the product is actually an Eclipse + sloeber deployment. 👍

jantje, 2021-12-14:
The feature group version number is not important. (But it is messy)
So I would say you are on 4.3.3
That may be caused by the fact Sloeber 4.4.0 is dependent on CDT 10 so eclipse can not upgrade sloeber to 4.4.0

jantje, 2021-12-14:
And the product is download unzip and run (on windows with 4.4.0 and earlier; other oses need a compatible java installed)

mduchain, 2021-12-14:
CDT is at 10 ... assuming this output is correct
[image]
.

jantje, 2021-12-15:
Then the updater should update. But I have seen many wierd things with the updater.
So you may have upgraded from 4.3.X (x<3) to 4.3.3
Checking the fixed issue I see no reason for a change in behaviour to the serial port.
JRE version: OpenJDK Runtime Environment (14.0.2+12) (build 14.0.2+12-46)  (which the log above states) should be fine
But this looks like #1277 which is fixed in 4.4

mduchain, 2021-12-25:
closing this issue without a proper resolution. 
From the mail exchange, I understands Eclispe updater messed-up the sloeber installation.
After spending a few hours looking for the problem, I downloaded the sloeber product v4.4.0 and was back up and running in 10 minutes.
Thx for the support during this time.

closed 2021-12-25 by mduchain