Sloeber/arduino-eclipse-plugin #1420: Upload to Wemos Lolin32 silently fails

jskubick, 2021-12-11:
**Describe your environment**   
Do you use the Sloeber product or plugin?   product
Sloeber version number  4.4.0.202107040421
OS: Windows 10
The url of the json that you used to download the platform :
There isn't one. I followed the official instructions at https://www.youtube.com/watch?v=szRrvJPvQz4 ("Sloeber install platform Arduino-esp32 from github").
* `git clone https://github.com/espressif/arduino-esp32.git` into c:\src\github\arduino-esp32
* Arduino->Preferences, added c:\src\github\arduino-esp32 as private hardware path
 
A image of the project properties arduino showing the board settings   
[image]

**Describe the bug**   
The project builds without errors, and no obvious error message gets generated when uploading... but it doesn't actually GET uploaded. The original Arduino sketch that was on the ESP32 is untouched.

**To Reproduce**   
Steps to reproduce the behavior:   
1. Create new Arduino sketch. Board = Wemos Lolin32, Partition scheme=default, flash frequency=80mhz, cpu frequency=80mhz, upload speed = 921600
2. Add the following two lines to the newly-created .cpp file's setup():
~~~
        Serial.begin(115200);
	Serial.println("test");
~~~
3. Run Arduino -> Upload Sketch
4. See Console output:
~~~
Starting upload
Uploading project "esp32First" with "esptool_py"

Launching: C:\src\Github\arduino-esp32/tools/esptool/esptool.exe 
Output:
esptool.py v3.1
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
               ...

esptool.py v3.1 - ESP8266 ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    merge_bin           Merge multiple raw binary files into a single file for
                        later flashing
    version             Print esptool version
    get_security_info   Get some security-related data

optional arguments:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}, -c {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3,esp32c6beta}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,usb_reset,no_reset,no_reset_no_sync}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset,no_reset_stub}, -a {hard_reset,soft_reset,no_reset,no_reset_stub}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM
                        bootloader. Some features will not be available.
  --trace, -t           Enable trace-level output of esptool.py interactions.
  --override-vddsdio [{1.8V,1.9V,OFF}]
                        Override ESP32 VDDSDIO internal voltage regulator (use
                        with care)
  --connect-attempts CONNECT_ATTEMPTS
                        Number of attempts to connect, negative or 0 for
                        infinite. Default: 7.
The execution of command "arduino-esp32/tools/esptool/esptool.exe" is done.
~~~

provide the name of the board and the platform version you are using.  
first-gen HiLetGo Wemos Lolin32 board with SSD1306 OLED (two pushbuttons on underside)

Provide the json url and a screenshot of project properties->arduino  
(see above)

**Expected behavior**   
The new sketch to replace the old one on the ESP32 board sitting at com8

The board has an existing sketch, compiled and uploaded using the Arduino IDE... so it absolutely, positively works... just not (presently) with Sloeber. If it helps, here's the output from the Arduino IDE itself when it performs the upload:

~~~
Serial port COM8
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 30:ae:a4:23:4b:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5461.3 kbit/s)...
Hash of data verified.
Compressed 17120 bytes to 11164...
Writing at 0x00001000... (100 %)
Wrote 17120 bytes (11164 compressed) at 0x00001000 in 0.1 seconds (effective 985.3 kbit/s)...
Hash of data verified.
Compressed 229504 bytes to 119462...
Writing at 0x00010000... (12 %)
Writing at 0x00014000... (25 %)
Writing at 0x00018000... (37 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (62 %)
Writing at 0x00024000... (75 %)
Writing at 0x00028000... (87 %)
Writing at 0x0002c000... (100 %)
Wrote 229504 bytes (119462 compressed) at 0x00010000 in 1.9 seconds (effective 970.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1890.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
~~~

The problem in Sloeber is limited to the ESP32. I can create a new Mega2560 project and flash it to a Mega2560 just fine. It's just the attempt to flash the ESP32 that's silently failing & leaving the original sketch untouched.

jantje, 2021-12-11:
There are quite a lot of issues fixed in this area in the nightly. I guess this one is a duplicate of #1371
https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=label%3A%22status%3A+fixed+in+nightly%22+

jskubick, 2021-12-12:
OK, that fixed it, thanks!!!

closed 2021-12-12 by jskubick