Sloeber/arduino-eclipse-plugin #1423: "Default" (USB bootloader) upload protocol missing from nightly #1418

jskubick, 2021-12-16:
The default (USB bootloader) upload protocol is missing from the "Upload Protocol" dropdown when creating a new Arduino project in nightly #1418:

[image]

In contrast, here's the same dialog from the latest official release version (4.4.0):

[image]

I'm not sure whether it's a related or unrelated bug, but when I tried creating the new Mega2560 project in 1418 with the board connected, it didn't detect the USB com port, either. A few minutes earlier, I had an open project with an ESP32 on com8. When I went to create the new project for the Mega2560 (with the ESP32 disconnected, and the Mega2560 connected on com5), the dropdown showed only com1 and com8.

For what it's worth, I encountered a similar bug a year and a half ago when I tried using CLion with Vladimir Schneider's Arduino plugin. Hopefully, the fact that it works fine in Sloeber 4.4.0, and the breakage occurred somewhere after that point, will make it easier to track down the problem. It's kind of a guess, but maybe the Arduino people changed something 1.5-3 years ago that broke everyone's older code, you fixed it in the code that ended up as 4.4.0, but somehow an older version that didn't incorporate the fix ended up in 1418, causing it to re-break?

jskubick, 2021-12-16:
Hmmm. I'd argue that the way Sloeber did it originally was actually a lot *nicer* than the way Arduino IDE implemented it. :-)

 I'd definitely recommend changing the label in the dialog from (blank) to something like "Default USB Bootloader". Regardless of what's going on behind the scenes, it'll make the correct choice (the one likely to result in a working new project) a lot more obvious for new users (and people who haven't used it for a while), without impairing experienced users.

jantje, 2021-12-17:
>Hmmm. I'd argue that the way Sloeber did it originally was actually a lot nicer than the way Arduino IDE implemented it. :-)

I agree but Arduino IDE is the standard.

>I'd definitely recommend changing the label in the dialog from (blank) to something like "Default USB Bootloader"

That is not necessary as when you use the upload button this field is not accounted for
I'm working on a "experanced" toolbar that will allow to do burn bootloader.
Maybe I should rename the field to something "the uploader you use". I gues most people only have one :-s.

jantje, 2021-12-30:
I tried to get the programmer to the "other options" tab but that turned out to be hard as the selection of programmers is platform dependent.
I checked and plenty of platforms provide a programmers.txt. Some referencing the same thing (to me) with different names/ID's; so making it a workspace parameter would not work.
So I decided to rename the field to mention "programmer" hoping this will make things more clear.
The other languages have been checked in as part of #1409
Still waiting for the Italian description https://twitter.com/Arduino_jantje/status/1476562163993042948

jantje, 2022-01-16:
I'm closing this as I think it is fixed as good as I can.
Please reopen if you think there is a better way.

closed 2022-01-16 by jantje