Sloeber/arduino-eclipse-plugin #1424: Feature Request: Add custom Makefile steps like strip

marcosx86, 2021-12-21:
One thing that I missed on Sloeber 4.4 was the possibility to use xtensa-esp32-elf-strip (esp32 project in the example) as final step to produce a much smaller binary. Would leave some space for more code.

jantje, 2021-12-21:
No need to miss that ;-)
CDT knows the post build step that allows you to add a command to be executed at the end of a build
You can use the platform provided environment variables (platform.txt and boards.txt) as you can see below.
[image]

If you want to trigger the command at will (not part of the build) you can create a build target yourself and launch that when you want.
The nightly started using this feature to support "upload using programmer" and "burn bootloader". As it is all pretty new to me I'm not really feeling comfortable with these features but they seem to have been in CDT for a long time so it should all work fine. 
What I'm trying to say is that there are probably better ways to do this but this is how I got it to work.
Create a file called makefile.init in your project root folder.
Using make formatting create a target and add the commands you want to execute.
[image]

Go into your target folder (typically release) and open the makefile. Search for a target in the outline window (for instance all). Select it and right click; select "add build target". Replace the name with your target name.
[image]

Now you have in your project explorer [Release]/Build Targets/[my target] which you can double click to run
[image]

You can use all make features you want so you can force a build before the command is run.

marcosx86, 2021-12-22:
Thank you! Already did the first image, but this Makefile custom step were what I was looking for, I'll try and reply.

marcosx86, 2021-12-22:
It worked as separated step. Tried to modify the main Makefile but it was overwritten when I ran the build project. Trying now to search for an "Makefile.am"-like so I can attach a strip step on every project automagicaly.

marcosx86, 2021-12-22:
Makefile.am is like a template for automake/autoconf projects, it uses for creating final Makefile after some ./configure step on opensource projects. Dunno if CDT generates using it or from scratch, but would be nice to have it as optional step here:

[image]

jantje, 2021-12-22:
Thanks for the info
Sloeber makes makefiles from scratch
Sloeber forked the CDT makefile generation from scratch and adapted it to Arduino requirements not supported by the CDT makefile generated (most important is the library and executable generation in 1 step)
I do not think the toolchain editor is the place to be as this is the input for the Sloeber makefile generator.
Note that the whole makefile generation implementation in CDT is basically considered bad and dead. So everything you see under C/C++ build is bad and dead.
For now CDT does not have a decent alternative so CDT keeps with it. And due to legacy reasons probably will for a long time.

I have a really hard time to understand what you are trying to do and what problems you encountered. 
So taking the risk to completely miss the point, I'd say: In my book I see 2 approaches
1) if this is a single project problem just add 
1a) the post build command
1b) a makefile.init target
1c) use external commands (with the nightly which allows you to use Arduino variables in the commands)
2) if it is a platform problem modify the platform.txt and or boards.txt
2a) have the call part of the build
2b) have a menu option to switch between it being part of the build
2c) Have the command in an environment variable so the options described in 1 can become simple commands like ${myCommand}

jantje, 2021-12-24:
I guess this one can be closed now.

closed 2021-12-24 by jantje