Sloeber/arduino-eclipse-plugin #1427: use "Sloeber" instead of "Arduino" when we are talking about sloeber stuf

jantje, 2021-12-30:
There are lot of places Sloeber should use the name "Sloeber" instead of the name "Arduino".
For instance project and preference configuration and perspective should be named Sloeber and not Arduino.
On the other hand the board name and references to the Arduino framework should remain "Arduino"

This is strictly text naming changes (no class names will be changed) but it is important to have a issue as this may become confusing for users to find what they are looking for.

MiBiMiFlo, 2022-01-10:
Is it ok to comment here if I find some places where it still is names Arduino?

First is the tool tip of The Perspective:
[image]

And in Project Properties it still says "**Arduino** Board Selection"
[image]

And third is in Shortcuts of "Customize Perspective - Sloeber" Category is **Arduino**:
[image]

jantje, 2022-01-10:
Thanks for the feedback. If you find more please report.

I tested the perspective hover. On my system it states Sloeber. Did you select the sloeber perspective in the list again?  If noit it may be remains of the arduino perspective you see.

I think the  "Arduino Board Selection" is the right name as you select a arduino board and not a sloeber board.

The rest I fixed.

https://user-images.githubusercontent.com/78593933/148801819-8aa41730-e9ad-4cbb-bfe8-cd008108a1ea.png
I'm not sure about "Arduino sketch" and "convert to sloeber project" 
I think it is inconsistent but on the other hand all the things I can think of to make it consistent seem less clear 😕

MiBiMiFlo, 2022-01-11:
You are right, seems I used an old workspace. With a new workspace the tool tip of the perspective is correct now.

jantje, 2022-02-16:
I think I changed all needed places.

closed 2022-02-16 by jantje