Sloeber/arduino-eclipse-plugin #1441: Nightly - Project settings are not persistent / written to .sproject

ArminJo, 2022-01-27:
This is my current version: 
 Sloeber, the Eclipse Arduino IDE (plugin version)	4.4.1.202201260122	io.sloeber.feature.feature.group	Sloeber.io

.sproject is not changed, not even on closing project, so after re opening all changes (naturally) are lost.

ArminJo, 2022-01-27:
Restaring Sloeber 2 times helped. Until next time.. (I had the error 3 times in the last 2 days)

jantje, 2022-01-27:
How do you change .sproject? I mean you should not change .sproject manually. That should not have effect.
You could change .sproject when the project is closed and then on open the changes should be taken into account but this is not a supported/adviced way of working.

ArminJo, 2022-01-27:
> You could change .sproject when the project is closed

That the way I do it always 😀

jantje, 2022-01-27:
>That the way I do it always 😀

But you should not modify .sproject files by hand. This is not a supported scenario.

If you want to change the config in text format goto project properties->sloeber->other and check "share in version control" for all configs you want to edit.
You should now have a sloeber.cfg file (which is a subset of the .sproject file).
As to spec (there will be bugs) you can change this file whenever you want and the changes should be taken into account.

closed 2022-01-27 by ArminJo