Sloeber/arduino-eclipse-plugin #1444: Serial monitor is not reconnected if a simlink is used instead of the device

mr-miky, 2022-01-28:
**Describe your environment**   
Sloeber, the Eclipse Arduino IDE (plugin version)	4.4.1.202201260122
OS: Fedora 34   
Eclipse Platform : Version: 2021-12 (4.22) Build id: I20211124-1800

**Describe the bug**   

For some months now I have noticed that if I have the serial monitor active and I write the flash of the ESP32 at the end the serial monitor is not reconnected.
I ignored this and reconnected by hand the serial monitor for some time until I used a laptop instead of the desktop.
Weird thing is that the laptop, same OS and same plungin version, reconnects without problems.
I started looking for the differences between the two setups. Since on the desktop I have several usb serial devices, through udev / serialnumber, I create symlinks in /dev /tty... specific for each device such as ttyESP0 or ttyFTD0 etc etc.
What I discovered is that on the laptop, without udev rules that create symlinks, I directly access the device like ttyUSB0 and everything works perfectly.
Otherwise if I create a symlink by hand and use it in sloeber it happens that everything works perfectly except returning to the serial monitor after writing the flash writing the flash

**To Reproduce**   
Steps to reproduce the behavior:   
1. create a symlink to /dev/ttyUSB0 eg. /dev/ttyESP0
2. Insert /dev/ttyESP0 as target port
3. Connect serial monitor at /dev/ttyESP0
4. Update esp32 flash
5. Wait for serial monitor reconnection

**Expected behavior**   
Serial monitor reconnected. 
I'm not ultra sure but in the past it worked

jantje, 2022-01-28:
Thanks for the extensive report. On first sight I have no clue what could cause this.

>I'm not ultra sure but in the past it worked

We upgraded the jssc library about 4 months ago https://github.com/Sloeber/arduino-eclipse-plugin/tree/master/io.sloeber.core/lib
If it used to work this is a potential candidate.

mr-miky, 2022-01-29:
I do not know if things are related but in the eclipse log there is reference to an error related to JSSC. The strange thing is that he says he can't find ttyUSB0. I state that all this happens without using symlinked the sloeber project expressly plans to use /dev/ttyUSB0.

`
!ENTRY io.sloeber.arduino.core 2 0 2022-01-28 18:24:19.398
!MESSAGE Error inside Serial. Break detected
!STACK 0
java.lang.Exception
	at io.sloeber.core.api.Serial.serialEvent(Serial.java:271)
	at jssc.SerialPort$LinuxEventThread.run(SerialPort.java:1381)`

and more

`
!ENTRY io.sloeber.arduino.core 4 0 2022-01-28 18:28:49.603
!MESSAGE Serial port /dev/ttyUSB0 not found. Did you select the right one from the project properties -> Sloeber?
!STACK 0
jssc.SerialPortException: Port name - /dev/ttyUSB0; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:219)
	at io.sloeber.core.api.Serial.connect(Serial.java:159)
	at io.sloeber.core.api.Serial.connect(Serial.java:150)
	at io.sloeber.core.api.Serial.<init>(Serial.java:103)
	at io.sloeber.core.api.Serial.<init>(Serial.java:82)
	at io.sloeber.ui.monitor.views.SerialMonitor.connectSerial(SerialMonitor.java:610)
	at io.sloeber.ui.monitor.views.SerialMonitor$7.run(SerialMonitor.java:428)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:474)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:580)
	at org.eclipse.jface.action.ActionContributionItem.lambda$5(ActionContributionItem.java:453)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5895)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1515)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:5126)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4581)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1154)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1045)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:644)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1440)

!ENTRY io.sloeber.arduino.core 2 0 2022-01-28 18:29:27.898
!MESSAGE Error inside Serial. Break detected
!STACK 0
java.lang.Exception
	at io.sloeber.core.api.Serial.serialEvent(Serial.java:271)
	at jssc.SerialPort$LinuxEventThread.run(SerialPort.java:1381)
`

jantje, 2022-01-29:
As you can see from the log the first sloeber mention is
at io.sloeber.ui.monitor.views.SerialMonitor$7.run(SerialMonitor.java:428)
here is the code
https://github.com/Sloeber/arduino-eclipse-plugin/blob/72a6e48b70dc6b98cb688d9d219abb6f4aabf2b9/io.sloeber.ui/src/io/sloeber/ui/monitor/views/SerialMonitor.java#L428
This is actually the code responding to the GUI action "open serial connection" in the serial monitor and should not be related to reopening the serial monitor.

in regards to 
>!ENTRY io.sloeber.arduino.core 2 0 2022-01-28 18:24:19.398 !MESSAGE Error inside Serial. Break detected !STACK 0 java.lang.Exception at io.sloeber.core.api.Serial.serialEvent(Serial.java:271) at jssc.SerialPort$LinuxEventThread.run(SerialPort.java:1381)

This happens at the sloeber code here
https://github.com/Sloeber/arduino-eclipse-plugin/blob/72a6e48b70dc6b98cb688d9d219abb6f4aabf2b9/io.sloeber.core/src/io/sloeber/core/api/Serial.java#L271
As you can see from the code this is logged based on a message coming from the serial port telling a "break" has happened. 
I can't say it is unrelated to the reconnection of the serial port but if it is it is more like a knock on event and not the real root cause (or even pointing towards one)
I have no clue whether this break event happens regularly or not. I mean, ..... it is logged but who looks into log files? ;-)

Here you see the reconnection code 
https://github.com/Sloeber/arduino-eclipse-plugin/blob/72a6e48b70dc6b98cb688d9d219abb6f4aabf2b9/io.sloeber.core/src/io/sloeber/core/tools/uploaders/UploadSketchWrapper.java#L192
One possible cause may be timing or the port popping up and disappearing immediately.

Easiest way out would be a debugging session or you learning me to setup these symlinks so I can get my upload tests working properly in linux :-)

mr-miky, 2022-01-29:
well if you have a real or virtual machine with linux just create a symbolic link. 

`sudo ln -s /dev/ttyUSB0  /dev/ttyESP0`

then use /dev/ttyESP0 instead of /dev/ttyUSB0. Sloeber does not list it among the available coms but just enter it by hand.

The symlink is simply an alias, a name that points to another file/device.

Why it should be different for

`while (!portFound & counter++ < 100) { `

?

At the moment I'm desperately trying to figure out what version of jssc I'm using...

mr-miky, 2022-01-29:
I think the problem is here

```
while (!portFound & counter++ < 100) {
     List<String> currentPorts = Serial.list();
     portFound = currentPorts.contains(myProvidedUploadPort);

```
I suppose serial.list() returns a list with only physical devices ttyUSB0 etc. Probably for the same reason ttyESP0 does not appear in the available serials of the serial selection widget and I have to put by hand /dev/ttyESP0.

mr-miky, 2022-01-29:
from jssc 2.9.4 source  

`public class SerialPortList {`

....

```
 static {
        serialInterface = new SerialNativeInterface();
        switch (SerialNativeInterface.getOsType()) {
            case SerialNativeInterface.OS_LINUX: {
                PORTNAMES_REGEXP = Pattern.compile("(ttyS|ttyUSB|ttyACM|ttyAMA|rfcomm|ttyO)[0-9]{1,3}");
                PORTNAMES_PATH = "/dev/";
                break;
            }
```

The regular expression prevents other 'fake' serials from being detected.
I don't know if it is possible to set a custom regexp and consequently the regexp should be set in the sloeber preferences.
Maybe just add an 'OR' to the regexp with the device name entered by hand in the serial selection widget.

jantje, 2022-01-29:
I'm not sure this regex is still needed at all
It is an old contribution from someone using linux (I'm mostly on windows) to work around an issue.
but lots of things have changed since then.

jantje, 2022-01-29:
Sorry Previous post is wrong
I didn't see you referred jssc code
However sloeber has some linux filtering code as well and that is the code I don't know how/why it works and wether it is still needed.
https://github.com/Sloeber/arduino-eclipse-plugin/blob/636e89af14f49c7c34067ecac2076285c921764c/io.sloeber.core/src/io/sloeber/core/Activator.java#L93

mr-miky, 2022-01-29:
I modified the jssc  regexp to include | ESP | FTD and then recompiled it.
Now sloeber sees the FTDx and ESPx serials also on the selection widgets and no longer has the problem of reconnecting the serial monitor
Obviously it is not the cleanest solution but it shows where the problem arises.

jantje, 2022-01-29:
just to be sure
You recompiled jssc and made sloeber use your newly compiled jssc

mr-miky, 2022-01-29:
> just to be sure You recompiled jssc and made sloeber use your newly compiled jssc

Yes

jantje, 2022-01-29:
So basically on linux if your symlink does not match the regular expression ttyS|ttyUSB|ttyACM|ttyAMA|rfcomm|ttyO)[0-9]{1,3}
the reconnection does not work because jssc does not list it in the available serial ports
Which makes that Sloeber doesn't see it and doesn't try to reconnect to it.
One could also say that if your serial port is not shown in the serial the ports combo box (in project properties->sloeber or the connect to serial port) reconnection will not work

mr-miky, 2022-01-29:
> One could also say that if your serial port is not shown in the serial the ports combo box (in project properties->sloeber or the connect to serial port) reconnection will not work

Yes exactly.

Con jssc modificata  vede piu com ports.

[image]

jantje, 2022-01-29:
Any idea why jssc filters for these ports?

mr-miky, 2022-01-29:
Because that expression matches commonly used usb-serial devices.

On linux all the devices are in /devs and there are not only the usb <> serial adapters so they have to somehow filter the right ones.

The need to create symlinks only serves to overcome the problem that device names are assigned in order of connection. Whoever is connected first will call ttyUSB0 and then the one connected after ttyUSB1.
If for any reason the connection order is exchanged the names change therefore you risk thinking of programming an arduino while in reality you are working on another.
By using  symlinks, through appropriate rules, the name is bound to the serial number of the usb<>serial adapter and therefore you work calmly because that arduino will always be called the same way

mr-miky, 2022-01-29:
wanting to work on it jssc also has

```
   /**
     * Get sorted array of serial ports in the system matched pattern
     *
     * @param pattern RegExp pattern for matching port names <b>(not null)</b>
     * 
     * @return String array. If there is no ports in the system String[]
     *
     * @since 2.3.0
     */
    public static String[] getPortNames(Pattern pattern) {
        return getPortNames(PORTNAMES_PATH, pattern, PORTNAMES_COMPARATOR);
    }
```
it would be enough to pass it the pattern and you get the correct list

jantje, 2022-01-29:
>The need to create symlinks only serves to overcome the problem that device names are assigned in order of connection. Whoever is connected first will call ttyUSB0 and then the one connected after ttyUSB1.
If for any reason the connection order is exchanged the names change therefore you risk thinking of programming an arduino while in reality you are working on another.

I thought you solved this with udev rules. Though I never got all my upload test arduinos connected and recognized properly with udev

>By using symlinks, through appropriate rules, the name is bound to the serial number of the usb<>serial adapter and therefore you work calmly because that arduino will always be called the same way

Given your simlink example above I assume this is a manual process.

>it is like saying COM3 will always be the one connected to the arduino with specific serial number regardless of the USB port on which it is connected and regardless of the temporal order of connection.

In windows you can do this in the gui and windows will assign the same com port based on vedorID and product ID. You should be able to do the same with udev rules but my "Linux" is not good enough to do so
That was what I thought you were doing which I hoped to get to work for my upload tests as these are fully automated so need to have the correct port or they will fail.

jantje, 2022-01-29:
As to
`public static String[] getPortNames(Pattern pattern) `
This definitely looks like an option.
As mentioned above Sloeber uses some kind of filtering which may be redundant now.
We could switch to this way in linux/mac and include some extra options

mr-miky, 2022-01-30:
> I thought you solved this with udev rules. Though I never got all my upload test arduinos connected and recognized properly with udev

Well there is a half trick because by default the various usb serial chips come out of the factory without serial number so you can not distinguish  between them as they have the same VID and PID. The chips of FTDI (arduino) and those of Silicon Labs (ESP32) have a tool for windows that allow you to assign the serial number and/or change the USB description of the device. Once you have assigned a serial number it is easy to distinguish them. I don't know if CH34x chips have a serial number or allow you to set it. anyway the udev rule is simple...
 
**lsusb -v**

```
Bus 010 Device 002: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6001 FT232 Serial (UART) IC
  bcdDevice            6.00
  iManufacturer           1 FTDI
  iProduct                2 FT232R USB UART
  iSerial                 3 A9CQDYPQ
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               90mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 FT232R USB UART
      Endpoint Descriptor:
```
**iSerial A9CQDYPQ** I entered it and it is a unique number

udev rules of my FTDI      /etc/udev/rules.d/99-ft232.rules

```
KERNEL=="ttyUSB*", ATTRS{serial}=="A63PFU8Y", SYMLINK+="ttyFTD0"
KERNEL=="ttyUSB*", ATTRS{serial}=="A9G6N0AA", SYMLINK+="ttyFTD1"
KERNEL=="ttyUSB*", ATTRS{serial}=="A9CQDYPQ", SYMLINK+="ttyFTD2"
KERNEL=="ttyUSB*", ATTRS{serial}=="A947LNQ6", SYMLINK+="ttyPRT0"
```

jantje, 2022-01-31:
Trying to get these udev rules working in Linux I found that boards that do not appear in the port list and require a reset with a reappear probably won't work as well.

mr-miky, 2022-01-31:
When you connect the cards, a ttyUSB* device is generated?
Some arduinos generate a ttyACM* device.
Udev rules i sent you filter only ttyUSB*
What vid and pid do they have?

jantje, 2022-01-31:
The rules I ended up with are here https://github.com/Sloeber/arduino-eclipse-plugin/blob/59f17b0cc905623c00bc26f71615095b566deeb0/io.sloeber.tests/src/io/sloeber/core/MySystem.java#L50

The boards that fail seem to have udev rules and the rules in my udev file seem to be ignored.

The most important thing for me is that creating symlinks causes issues
1) upload with wait for reappear assumes only the port reset is disappearing (with symlinking it may be more)
2) ports are filtered by jssc (and should be) as sutch the ports wil not be available for selection
3) Reopen the serial monitor will not work when used portname is not returned by jssc

mr-miky, 2022-01-31:
There are a couple of things that puzzle me about your rules. To be sure I would put an ACTION=="add" so that you create the link only to the insertion then try to use ATTR instead of ATTRS. After you change the rules you need to reload them.

`sudo udevadm control --reload-rules && sudo udevadm trigger`

Even if nothing works, does the link create it? 
The rule file has the correct name ##-sloeber.rules where ## is usually above 90, 99 in my case and is on /etc/udev/rules.d/ ?

There may be some other rule with higher priority (## less than yours) that creates problems or even finishes scanning the rules before yours.

jantje, 2022-01-31:
>There are a couple of things that puzzle me about your rules. 

Honestly I don't know what I'm doing. I took your line and modded it so it works. But I didn't get it to work for all boards.
For the once that didn't work I tried to "debug" them with
```
udevadm control --log-priority=debug
journalctl -f
```
As described here https://unix.stackexchange.com/questions/470960/how-can-i-debug-a-udev-rule#470963
And as far as I can tell (and honestly I do not feel confident at all) my rules file 99-jantje.rules was not actioned upon for these boards but others were

> To be sure I would put an ACTION=="add" so that you create the link only to the insertion then try to use ATTR instead of ATTRS

I need simpler more clear instructions 🙏 

>The rule file has the correct name ##-sloeber.rules where ## is usually above 90, 99 in my case and is on /etc/udev/rules.d/ ?

As you can see from the screendump some rules are applied but only lilipad and arduino 101 won't work.
There is even a rules file for arduino 101 in the same folder. 

[image]

>There may be some other rule with higher priority (## less than yours) that creates problems or even finishes scanning the rules before yours.

That is what I think I see. I see adafruit lilipad mentioned (in the output of journalctl -f) when connecting my lilipad (which is a fried board) and I remember other rules for 101.
I however don't feel confident modding udev rules; for sure not if the header states "don't modify" 😉

mr-miky, 2022-01-31:
I did the test myself, I put your rules on my pc

/etc/udev/rules.d/99-sloeber.rules 

```
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="003d", SYMLINK+="ttyDueProg"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0041", SYMLINK+="ttyYun"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="8041", SYMLINK+="ttyYun"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0043", SYMLINK+="ttyUno"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0042", SYMLINK+="ttyMega2560"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0010", SYMLINK+="ttyMega2560"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="8036", SYMLINK+="ttyLeonardo"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0036", SYMLINK+="ttyLeonardo"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="1B4F", ATTRS{idProduct}=="9207", SYMLINK+="ttyFried"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="1B4F", ATTRS{idProduct}=="9208", SYMLINK+="ttyFried"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2157", SYMLINK+="ttyZeroProg"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="804d", SYMLINK+="ttyZeroNa"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="004d", SYMLINK+="ttyZeroNa"

KERNEL=="ttyUSB*", ACTION=="add", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="7523", SYMLINK+="ttyNanoFakeChina"
```

The line with the "fake" is for chinese arduino with CH340

ls -la /dev/tty*

Result:

```
[michele@fx-8350 ~]$ ls -la /dev/tty*
....
crw-rw----+ 1 root dialout 166,  0 31 gen 21.39 /dev/ttyACM0
crw-rw----  1 root dialout 166,  1 31 gen 21.39 /dev/ttyACM1
crw-rw----  1 root dialout 166,  2 31 gen 21.39 /dev/ttyACM2
crw-rw----  1 root dialout 166,  3 31 gen 21.42 /dev/ttyACM3
lrwxrwxrwx  1 root root          7 31 gen 21.42 /dev/ttyDueProg -> ttyACM3
lrwxrwxrwx  1 root root          7 31 gen 21.39 /dev/ttyESP7 -> ttyUSB0
lrwxrwxrwx  1 root root          7 31 gen 21.41 /dev/ttyFTD2 -> ttyUSB2
lrwxrwxrwx  1 root root          7 31 gen 21.39 /dev/ttyLeonardo -> ttyACM1
lrwxrwxrwx  1 root root          7 31 gen 21.39 /dev/ttyMega2560 -> ttyACM0
lrwxrwxrwx  1 root root          7 31 gen 21.39 /dev/ttyNanoFakeChina -> ttyUSB1
lrwxrwxrwx  1 root root          7 31 gen 21.39 /dev/ttyUno -> ttyACM2
crw-rw----  1 root dialout 188,  0 31 gen 21.39 /dev/ttyUSB0
crw-rw----  1 root dialout 188,  1 31 gen 21.39 /dev/ttyUSB1
crw-rw----+ 1 root dialout 188,  2 31 gen 21.41 /dev/ttyUSB2

```

I have no other types of arduino here at home to try but I think it works

Later I rewrite the JSSC regexp and recompile it so that it doesn't filter your symlinks and then let you know what happens.
What is written in 99-lylipad.rules?

mr-miky, 2022-01-31:
JSSC modified ruegexp:

```PORTNAMES_REGEXP = Pattern.compile("(ttyS|ttyUSB|ttyACM|ttyAMA|ttyESP|ttyFTD|rfcomm|ttyO)[0-9]{1,3}|tty[Zero,Uno,Due,Mega,Leo,Yun]*");```

result:
[image]

For the moment I would leave out the lylipad problem and I would try to understand why it shows you the list of devices but not the symlinks. 

`Public static String[] getPortNames(Pattern pattern)` looking at the source does not seem to consider the RegExp mentioned above. 

Can you put that RegExp as a pattern for getPortNames and see what it returns?

jantje, 2022-01-31:
seems the udev rules are not giving the expected result
[image]

I'm not a fan of changing jssc
I believe it is better to change sloeber to provide it's own regex that can be changed by the user

mr-miky, 2022-01-31:
Well it seems obvious to me that patching JSSC is not the solution.
I did it because it is faster for me rather patch sloeber thus introducing other possibilities of malfunction.
Did uoy give sudo udevadm control --reload-rules & sudo udevadm trigger after adding/editing the rules?
If you want instead to give the command sudo..... you can do as with windows, exit & restart :-) 
Joking aside if you see that the rules are not applied restart ubuntu too. 

P.S. ubuntu is on a real non-virtual machine?

jantje, 2022-01-31:
tried the extra command to no avail
[image]
It is a actual machine
[image]

mr-miky, 2022-01-31:
apart from leonardo all the others are programmed and return connected to the serial monitor also using symlinks.

One ok.
Two ok.
Mega ok. 

Leonardo, does not reconnect serial mon but is programmed and reconnecting by hand the serial mon works.

**The final question is therefore why the udev rules do not work for you Tomorrow I do some tests with a live ubuntu ?**

jantje, 2022-02-01:
I'm making a sloeber version that will allow you to provide the filter pattern.

If I remove ` ACTION=="add", ` from the rules it works
The fried lillypad seems to be recognized as a ttyNanoFakeChina
So only arduino 101 left

mr-miky, 2022-02-01:
But you did not detach and reattach the arduino after changing the rules?
The add is used to prevent a link from being created only when you attack a device. 
Strictly speaking, without the add, it also creates a link to the disconnection...

 Anyway installing ubuntu without other things 20.04.3

with "add" rule

[image]

jantje, 2022-02-01:
I told you I'm a dummy with udev rules :-(

mr-miky, 2022-02-01:
Check the vid and pid of your 101 with lsusb 
Check that the nanofake is not from some other Chinese product.

Ok I'm going to bed, wake up in 5 hours...

jantje, 2022-02-01:
I've created a branch that allows you to modify the filter used to detect serial ports.
You can create a environment variable of the name jssc_serial_filter_pattern containing the filter
you can also use the window ->preference->C/C++->build->environment to create a variable jssc_serial_filter_pattern (same name) and add the filter.
Strangely enough when you create the OS environment variable the reading of the CDT environment variable fails so if you create a OS level environment variable you can not modify it.

I changed the udev rules to
```
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="003d", SYMLINK+="s_DueProg"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0041", SYMLINK+="s_Yun"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="8041", SYMLINK+="s_Yun"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0043", SYMLINK+="s_Uno"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0042", SYMLINK+="s_Mega2560"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0010", SYMLINK+="s_Mega2560"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="8036", SYMLINK+="s_Leonardo"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0036", SYMLINK+="s_Leonardo"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="1B4F", ATTRS{idProduct}=="9207", SYMLINK+="s_Fried"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="1B4F", ATTRS{idProduct}=="9208", SYMLINK+="s_Fried"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2157", SYMLINK+="s_ZeroProg"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="804d", SYMLINK+="s_ZeroNa"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="004d", SYMLINK+="s_ZeroNa"
KERNEL=="ttyACM*", ACTION=="add", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="0487", SYMLINK+="s_teensy"

KERNEL=="ttyUSB*", ACTION=="add", ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="7523", SYMLINK+="s_NanoFakeChina"
```

I currently use following regex (ttyS|ttyUSB|ttyACM|ttyAMA|ttyESP|ttyFTD|rfcomm|ttyO)[0-9]{1,3}|s_.*

Still the same boards failing
[image]

I also made a fix so the leonardo and yun upload should now work. I'm surprised it worked on your system as symlinking found by sloeber should make the reset to fail.

I would expect a action to pick up the branch and make a product but I do not see it yet

jantje, 2022-05-30:
you can now modify the filter used to detect serial ports.
You can create a environment variable of the name jssc_serial_filter_pattern containing the filter
you can also use the window ->preference->C/C++->build->environment to create a variable jssc_serial_filter_pattern (same name) and add the filter.
Strangely enough when you create the OS environment variable the reading of the CDT environment variable fails so if you create a OS level environment variable you can not modify it.
I think this closes this issue. Please reopen if not.

closed 2022-05-30 by jantje

mr-miky, 2022-06-09:
@jantje
I tried tonight and I have to say that the serial inclusion pattern works really well.
Nice result