Sloeber/arduino-eclipse-plugin #1446: `runtime.tool` properties are not evaluated for customized board definitions referencing installed platform

jandrassy, 2022-02-08:
Sloeber, plugin version 4.4.1.202202010122
Eclipse  2020-12 (4.18.0)
Linux Mint 

It worked in Sloeber 4.3 and older

custom hardware definitions in `hardware` folder stopped working. 
**the variables for tools locations (runtime.tool) are not evaluated.**
I get 
```
Starting C++ compile
"/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR     -I"/home/duro/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.8.4/cores/arduino" -I"/home/duro/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.8.4/variants/mega" -I"/home/duro/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.8.4/libraries/SPI" -I/home/duro/eclipse/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.2.0/system/Drivers/STM32F1xx_HAL_Driver/Inc -I"/home/duro/Arduino/libraries/ArduinoOTA/src" -I"/home/duro/Arduino/libraries/Ethernet/src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
/bin/sh: 1: /bin/avr-g++: not found
```
this is just an example. I get similar error if referencing the SAMD package. 

If I make a workaround by entering a full path in platform(.sloeber).txt in the referenced platform (arduino/avr), then I get an error at upload
```
Launching: /bin/avrdude -C/etc/avrdude.conf -patmega2560 -carduino -P/dev/ttyACM0 -b115200 -D -Uflash:w:/home/duro/workspace/OTEthernetTest/Release/OTEthernetTest.hex:i 
Output:
Cannot run program "/bin/avrdude" (in directory "/home/duro/workspace"): error=2, No such file or directory
```

Arduino doc on referencing platforms: https://arduino.github.io/arduino-cli/0.21/platform-specification/#referencing-another-core-variant-or-tool

my customized boards definitions as example: https://github.com/jandrassy/my_boards

jantje, 2022-02-14:
I tried it here and I confirm the issue and it is regression.

jantje, 2022-02-14:
This is the root cause
https://github.com/Sloeber/arduino-eclipse-plugin/blob/9f86244fa9d4eb776bf76a7399d4022bd0358672/io.sloeber.core/src/io/sloeber/core/api/BoardDescription.java#L857

The root cause is that sloeber changed from simply using boards.txt all the time to starting from the json files.
As private hardware doesn't have a json the "platform is not found" and currently the code assumes nothing needs to be done in regards to tool path extraction from the referenced platforms.
Seems your boards need it; so the assumption is wrong.

I'm not seeing a quick way around this right now. I will need a think before I can fix it.

closed 2022-02-16 by jantje

jandrassy, 2022-02-17:
thank you. it works