Sloeber/arduino-eclipse-plugin #1448: runtime.hardware.path should evaluate to "board platform"

jandrassy, 2022-02-17:
Sloeber, plugin version 4.4.1.202202160121
Eclipse 2020-12 (4.18.0)
Linux Mint

When using customized board definitions in sketchbook's hardware folder, Sloeber evaluates `runtime.hardware.path`  to referenced "core platform", but it should evaluate to "board platform".

The property `runtime.platform.path` is evaluated OK in Sloeber.

Arduino doc on global properties: https://arduino.github.io/arduino-cli/0.21/platform-specification/#global-predefined-properties

my customized boards definitions as example: https://github.com/jandrassy/my_boards

jantje, 2022-02-17:
>When using customized board definitions in sketchbook's hardware folder, Sloeber evaluates runtime.hardware.path to referenced "core platform", 

Yes it does
https://github.com/Sloeber/arduino-eclipse-plugin/blob/2f4e9822e55b1ca2ea5635b2894dcda3608e9aaa/io.sloeber.core/src/io/sloeber/core/api/BoardDescription.java#L786

>Arduino doc on global properties: https://arduino.github.io/arduino-cli/0.21/platform-specification/#global-predefined-properties

Seems the documentation has become a bit more clear since I last read it. Seems I should do some refactoring to get the code compliant with https://arduino.github.io/arduino-cli/0.21/platform-specification/#platform-terminology

>my customized boards definitions as example: https://github.com/jandrassy/my_boards

Can you describe a case where the build fails due to this issue? I'm considering to add it as a test case. in the regression tests

jandrassy, 2022-02-17:
in my_boards/avr I use my build of arduinoOTA tool (with a timeout parameter). the path is defined in platform.txt with `runtime.hardware.path`

jantje, 2022-02-17:
> in my_boards/avr I use my build of arduinoOTA tool (with a timeout parameter). the path is defined in platform.txt with `runtime.hardware.path`

If this is your description of a case where the build fails due to this issue; I fear I will need a more detailed description.

jantje, 2023-01-16:
Based on https://arduino.github.io/arduino-cli/0.21/platform-specification/#global-predefined-properties
I would say runtime.hardware.path is the parent of runtime.platform.path

jantje, 2023-01-16:
@JAndrassy can you validate the change?

JAndrassy, 2023-01-18:
it is OK
(sorry, it took me hours to build and run Sloeber from sources)

jantje, 2023-01-18:
>sorry, it took me hours to build and run Sloeber from sources

That is not needed as Sloeber supports github actions and these have the artifacts that are the product and the update site