Sloeber/arduino-eclipse-plugin #1450: Ordering of boards.txt is not respected in "project"->"properties"->"sloeber"->"board part number"

MasteringTheMess, 2022-02-25:
When selecting a board from the sloeber - properties menu, all the boards are listed randomly, which is not a big problem in most cases, but when playing around with STM32 generic variants, there are often dozens of boards which makes it quite difficult to find a specific one

[image]

I am aware that this is rather a "cosmetic" issue, but I have a small hope that it is merely a small change that is necessary...
I really am only a novice programmer, but if someone points me to the specific location in the code, I could have a look and try to make up a fix?

MasteringTheMess, 2022-02-25:
yes very similar, if not a duplicate.
I just checked the ESP8266 RAM size option and it is similarily random-ordered, but in a different way :)

I run a somewhat recent nightly 4.4.1
[image]

I could upgrade to the most recent for testing purpose if that helps?

jantje, 2024-08-19:
I think this is fixed in V5
[image]

closed 2024-08-19 by jantje

vvs42, 2025-12-02:
I've been using Slobber since version 4.4.0 and haven't seen a solution to this problem in any version. Which version 5 are you talking about?

[image]
[image]