Sloeber/arduino-eclipse-plugin #1451: release version of sloeber unuseable under linux.  cant update,  missing natures, broken packages invalid repository address.

eadthem, 2022-02-27:
Eclipse 2021-12
when i tried to install sloeber from the eclipse marketplace 4.4.1 it fails
An error occurred while collecting items to be installed
session context was:(profile=_home_user_eclipse_embedcpp-latest-released_eclipse, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,org.eclipse.nebula.cwt,1.1.0.202011020719
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.cdatetime,1.5.0.202011020719
No repository found containing: osgi.bundle,org.hamcrest.core,1.3.0.v20180420-1519
No repository found containing: osgi.bundle,org.junit,4.13.2.v20211018-1956

java version 15.0.2 from oracle  built in package from OpenSUSE 15.2

I tried this again with 2021-03    and again with java 11   instead of 1.8.0
Same or similar results.

I then downloaded the official beryllium release of sloeber from http://eclipse.baeyens.it/index.shtml  and it would not even launch or provide an error at first with java 1.8.0   i installed the openJDK 11 packages from OpenSUSE15.2 on a guess (nothing said 1 version or another was required) and it would launch after that.

But its still broken.
The nature sloeber is missing in any old projects / work spaces.   i cant make any new Arduino projects.   there are no settings related to sloeber in window > preferences 

I notice anything with sloeber trys to add the repo http://https://download.eclipse.org/technology/babel/update-site/R0.17.1/2019-12/
Which is an invalid address, but it re adds it if i disable or fix the http:// every time i update.  removing the http:// helps that but still fails to resolve the above issues with the missing sloeber nature and arduino project types.

What exactly is the working stable release version / where do i get it?   What version of eclipse and java is required to run this?

I have used a dozen combinations of java versions and eclipse versions all with similar failures to when i tried 2021-12 and of course beryllium has its own separate issues disabling it.

I was using eclipse 2018 but needed to upgrade.

let me know what info you need to help resolve these issues.

Thanks

jantje, 2022-02-27:
try this version https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/1846854129
I should make this the latest release as many of the problems you mentioned are fixed.

eadthem, 2022-02-27:
Thank you for your help.
That version resulted in this message.

[image]

I think i was able to figure out what i needed in eclipse 2018 which has sloeber 4.3  so i will use that for now.

jantje, 2022-02-27:
searching for the error it states you have a 64 32 bit conflict.
Strange as this version should come with java.
Also strange as the build runs on a linux and creates and compiles projects
Something is off on your system

jantje, 2022-05-07:
I assume the issue can be closed. please reopen with comment why if you disagree

closed 2022-05-07 by jantje