Sloeber/arduino-eclipse-plugin #1459: "fake" unresolved includes on STM32

MasteringTheMess, 2022-03-25:
sloeber nightly 4.4.1.202201252228 on windows 10
stm32 package is 2.2.0
board is Nucleo-L4R5ZI

seems like I ran into an issue where sloeber/eclipse does not correctly resolve includes of stm32 arduino core

It compiles without error or warning and the board does what it should - although I am barely at the stage of blinking some LEDs...

Behaviour ist not quite consistent when hitting F3 (open declaration) on **Serial** and **uint32_t** nothing is found, but on **PinName** and **pinMode** it works

[image]

Don't know what further details could be helpful, just ask and I'll be happy to provide all information required

jantje, 2022-03-25:
This is clearly a indexer problem. 
Did you install the STM32 with the boardsmanager?
Can you check the log file/view for errors?

jantje, 2022-03-25:
I see nothing that could explain the problem you are seeing.
please close the project delete the log and reopen the project. If the problem persists please provide the log in text format.

MasteringTheMess, 2022-03-28:
Hi jantje.

sorry for the long wait...

here is the log after opening the project:

```
!ENTRY org.eclipse.core.expressions 4 201 2022-03-28 06:34:57.115
!MESSAGE No property tester contributes a property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu to type class org.eclipse.core.internal.resources.Project
!STACK 1
org.eclipse.core.runtime.CoreException: No property tester contributes a property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu to type class org.eclipse.core.internal.resources.Project
	at org.eclipse.core.internal.expressions.TypeExtensionManager.getProperty(TypeExtensionManager.java:131)
	at org.eclipse.core.expressions.TestExpression.evaluate(TestExpression.java:104)
	at org.eclipse.core.expressions.CompositeExpression.evaluateAnd(CompositeExpression.java:54)
	at org.eclipse.core.expressions.AndExpression.evaluate(AndExpression.java:36)
	at org.eclipse.ui.internal.dialogs.RegistryPageContributor.failsEnablement(RegistryPageContributor.java:284)
	at org.eclipse.ui.internal.dialogs.RegistryPageContributor.isApplicableTo(RegistryPageContributor.java:227)
	at org.eclipse.ui.internal.dialogs.PropertyPageContributorManager.getApplicableContributors(PropertyPageContributorManager.java:219)
	at org.eclipse.ui.internal.dialogs.PropertyPageContributorManager.getApplicableContributors(PropertyPageContributorManager.java:237)
	at org.eclipse.ui.dialogs.PropertyDialogAction.hasPropertyPagesFor(PropertyDialogAction.java:104)
	at org.eclipse.ui.dialogs.PropertyDialogAction.isApplicableForSelection(PropertyDialogAction.java:145)
	at org.eclipse.ui.dialogs.PropertyDialogAction.isApplicableForSelection(PropertyDialogAction.java:126)
	at org.eclipse.ui.internal.navigator.resources.actions.PropertiesActionProvider.fillContextMenu(PropertiesActionProvider.java:56)
	at org.eclipse.ui.navigator.NavigatorActionService$2.run(NavigatorActionService.java:228)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.ui.navigator.NavigatorActionService.addCommonActionProviderMenu(NavigatorActionService.java:222)
	at org.eclipse.ui.navigator.NavigatorActionService.fillContextMenu(NavigatorActionService.java:177)
	at org.eclipse.ui.navigator.CommonNavigatorManager.fillContextMenu(CommonNavigatorManager.java:258)
	at org.eclipse.jface.action.MenuManager.fireAboutToShow(MenuManager.java:338)
	at org.eclipse.jface.action.MenuManager.handleAboutToShow(MenuManager.java:468)
	at org.eclipse.jface.action.MenuManager$2.menuShown(MenuManager.java:495)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:259)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4243)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1060)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1084)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1065)
	at org.eclipse.swt.widgets.Control.WM_INITMENUPOPUP(Control.java:5150)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4782)
	at org.eclipse.swt.widgets.Canvas.windowProc(Canvas.java:340)
	at org.eclipse.swt.widgets.Decorations.windowProc(Decorations.java:1480)
	at org.eclipse.swt.widgets.Shell.windowProc(Shell.java:2286)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5002)
	at org.eclipse.swt.internal.win32.OS.TrackPopupMenu(Native Method)
	at org.eclipse.swt.widgets.Menu._setVisible(Menu.java:237)
	at org.eclipse.swt.widgets.Display.runPopups(Display.java:4101)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3626)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1154)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1045)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:644)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)
!SUBENTRY 1 org.eclipse.core.expressions 4 201 2022-03-28 06:34:57.115
[... cut ...]
!ENTRY io.sloeber.core 8 0 2022-03-28 06:35:01.797
!MESSAGE index Netzteil_Netzwerker

!ENTRY io.sloeber.core 8 0 2022-03-28 06:35:03.940
!MESSAGE do not pospone index Netzteil_Netzwerker

!ENTRY io.sloeber.core 8 0 2022-03-28 06:35:03.960
!MESSAGE Index of project changed :Netzteil_Netzwerker

!ENTRY io.sloeber.core 8 0 2022-03-28 06:35:03.962
!MESSAGE Looking for libraries for project :Netzteil_Netzwerker

!ENTRY io.sloeber.core 8 0 2022-03-28 06:35:04.065
!MESSAGE libraries added for project Netzteil_Netzwerker
```

jantje, 2022-03-28:
Assuming the problem is in project Netzteil_Netzwerker I see no issues in the log that relate to indexing problems

MasteringTheMess, 2022-03-29:
yes, that is one of the projects that has unresolved includes and missing syntax coloring...
I found out that it is not necessarily related to STM32.
I created a small example sketch for Nucleo-144 (SerialPassthrough) and there all is well...
I think I'll just reinstall sloeber and re-import all projects

MasteringTheMess, 2022-03-30:
I did not reinstall but instead I installed a fresh copy of sloeber 4.4.0 alongside the existing version.
There I found out that updating to the latest nightly solved this issue.
I then updated my initial sloeber and now it seems to work without problems.

No idea what the actual problem is/was but it seems to be fixed now...

jantje, 2022-03-30:
Guess we can close this issue then.

MasteringTheMess, 2022-03-31:
Yes

closed 2022-03-31 by MasteringTheMess