Sloeber/arduino-eclipse-plugin #1460: Failure during Eclipse startup, "Unable to make field private final..."

cabintech, 2022-03-31:
Environment:
- Newly installed Eclipse (2022-03) for Java EE (required for other projects)
- Newly installed CDT plugin for C++ tooling
- Newly installed Sloeber plugin via the Marketplace
- Windows 10

During Eclipse startup it fails to read 3 JSON files with the message:
```
Failed to parse json file "C:\Users\mamcmi\eclipse\jee-2022-03\eclipse\arduinoPlugin\package_index.json"
Unable to make field private final java.lang.String java.io.File.path accessible: module java.base does not "opens java.io" to unnamed module @e1fba9a
```

I did both of the recommendations in issue 705 (run eclipse.exe from a short path via a BAT file, and set windows policy to allow long path names). This does not seem like a path length issue and I could find no other reports of a problem like this.

cabintech, 2022-03-31:
Ha, OK. Maybe that is a ridiculously obscure message for "file not found". There is no such file at that location. I guess this does not work 'out of the box', 

I am having a hard time finding out exactly how to get boards defined... some instructions say "do it just like the Arduino IDE"... but my Arduino IDE directory does not have those missing files either. This page (http://eclipse.baeyens.it/how_to.shtml#/h) seems out of date because that is not what the Preferences->Arduino settings look like. It shows how to pick a board from the list, but not how to get the list in the first place.

Any help or pointers to how to get this configured and working would be appreciated...

cabintech, 2022-03-31:
Downloaded package_index.json from https://downloads.arduino.cc/packages/package_index.json and put it into the arduinoPlugin directory.

Same error on startup, and it then deletes the file. Ugh. I don't get how this is supposed to work...

cabintech, 2022-03-31:
Gave up on the plugin. Downloaded the full integrated Sloeber package. At least it starts properly configured, except...

I am using a Teensy 3.2 board and I cannot get Sloeber configured for it. I added a private hardware path to my existing Arduino IDE "hardware" directory but it does not show up in the board list. 

[image]

[image]

[image]

Also tried specifying the teensy subdir, and the teensy/avr subdir but still no change in the board list.

jantje, 2022-03-31:
Stuff added to the private hardware is not managed by the boardsmanager so it should not show up anywhere in the preferences.
It should only be visible in the project properties->sloeber (used to be arduino)

Your original issues are related to java disabling reflection making reading json files (on which the arduino framework relies heavily) failing. Bad json files are deleted by Sloeber as they are unusable. However in this case that makes the behaviour hard to understand.

jantje, 2022-05-07:
I think this issue can be close. pease reopen if you think this issue is not "fixed" ad explain why you think so

closed 2022-05-07 by jantje