Sloeber/arduino-eclipse-plugin #1461: Cannot find #include library

cabintech, 2022-04-04:
Running Sloeber full install, current version, on Windows 10 with Teensy 3.2 board.

I am trying to get code that compiles with the Arduino IDE moved over to Sloeber. I have worked through a number of issues but am stuck on this one...

My code has a number of #includes, specifically:

```
#include "Arduino.h"

#include "TideSensor.h"
#include "Sample.h"
#include "FullSample.h"
#include "ShortSample.h"
#include "RadioSender.h"

#include <EEPROM.h>      // Read/write EEPROM
#include <Snooze.h> //https://github.com/duff2013/Snooze
```

I cannot get the compiler to find EEPROM.h even though it is in the same set of library directories as Snooze.h which is found by the compiler just fine.

I notice that an Include directive is generated for Snooze.h, but not for EEPROM.h in the compiler command:

`"C:\Users\Mark\Dropbox\CabinTech\TeensyDev\arduino-1.8.15\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -Os --specs=nano.specs -g -Wall -ffunction-sections -fdata-sections -nostdlib -mno-unaligned-access -MMD -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=154 -DARDUINO=10812 -DARDUINO_TEENSY32 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Users\Mark\Documents\sloeber-workspace\TideSensor\Release/pch"   -I"C:\Users\Mark\Dropbox\CabinTech\TeensyDev\arduino-1.8.15\hardware\teensy\avr\cores\teensy3" -I"C:\Users\Mark\Dropbox\CabinTech\TeensyDev\arduino-1.8.15\hardware\teensy\avr\libraries\Snooze\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"`

I have tinkered with the Arduino->Preferences->Private Library Path to no avail, no matter what directory level I specify it still does not find EEPROM.h

A side issue: If I comment-out the #include of EEPROM.h and rebuild, it still fails trying to find EEPROM.h because sloeber.ino.cpp does not get updated. I have to clean the project and do a full rebuild to get sloeber.ino.cpp in sync with my code.

jantje, 2022-04-04:
I tried this here and I can reproduce this. However it does not look like a Sloeber issue.
When you ctrl click the code
`#include <EEPROM.h>  `
You will see sloeber takes you to [teensy install]\teensy\avr\cores\teensy3\avr\eeprom.h
This tells you sloeber thinks EEPROM.h is resolved by eeprom.h so Sloeber will not try to add a library. As windows os is case insensitive to filenames this is perfectly fine.
However the toolchain used to build the teensy seems to be case sensitive and does not find the file.
This leads to the confusing situation.
To fix this problem you can manually add the library.

Remarks:
1) one could say that  [teensy install]\teensy\avr\cores\teensy3\avr is not in the include path so the headers there should not be found during include searches. However he idea of the indexer is to be optimisti!c so the developer does not need to search for things.

2) Wether the the soeber.ino.cpp is properly updates is dependent on you indexer settings. These are mine
[image]

jantje, 2022-04-05:
>Sorry to be obtuse, but can you tell the details on how to do this?

http://eclipse.baeyens.it/how_to.shtml#/lib

>Also, the Arduino IDE resolves the file case problems properly on Windows. 

Arduino IDE uses a completely different way to Sloeber on finding the missing libraries.

cabintech, 2022-04-06:
In the Arduino->Add a library to the selected project

I checked the "EEPROM" project and the #include is resolved, but it is resolved to the wrong header file - it resolves to "eeprom.h" in C:\Users\Mark\Dropbox\CabinTech\TeensyDev\arduino-1.8.15\hardware\teensy\avr\cores\teensy3\avr instead of "EEPROM.h" in C:\Users\Mark\Dropbox\CabinTech\TeensyDev\arduino-1.8.15\hardware\teensy\avr\libraries which is the correct header file.

I think it finds it in .../cores/.. first (although the case does not match) and stops looking. Is there a way to fix that?

cabintech, 2022-04-06:
Well, I found a crude fix, Arduino->Add a source folder to the selected project

and I pointed this directly to the .../libraries/EEPROM directory. It imported EEPROM.h and EEPROM.cpp.

The project now resolves everything in the editor, compiles, and links. Yea!

jantje, 2022-04-06:
>I think it finds it in .../cores/.. first (although the case does not match) and stops looking. Is there a way to fix that?

can you provide a screenshot of the private libraries. I guess there is stuff in there that should not be there because it worked fine for me.

closed 2022-05-07 by jantje