Sloeber/arduino-eclipse-plugin #1462: Sloeber - STM32Bluepill - fail to call STM32Cube to download target

SigP226, 2022-04-08:
Hello,

ENVIRONMENT: 
- Windows 10 / Eclipse 3-22 / CDT / sloeber 4.4 / STM32 v2.2.0 /dark theme / JDK18
- STM32 Bluepill connected via STLINK V2 
Preliminary test to disciminate the issue: 
- JDK16 or 18 same problem, Was in 16, I upgrade to 18 to see if it was java version the root cause
- Code compilation OK, no errors nor warnings despite this bizarre error in the LOG : 
[image]
- To check the STLINK link, I have used STM32cube (graphical version, not the command line) and I can download the HEX file without any issue and code works well. So the issue is not coming from the connection between the board, the STLINK and the PC.
- test on full new empty project (LED blink). Same result

ISSUE DESCRIPTION 

After code compilation, when I push the download button the system fail to launch the downloder (STM32cube)
I get this error:
[image]
and this error
[image]

I have check the "stm32CubeProg.sh" file and eveything seems ok. Seems the parameter sent to the SH file are wrong somwhere.

any idea?

jantje, 2022-04-08:
Can you provide the output from the upload console?

SigP226, 2022-04-09:
Here is

[image]

jantje, 2022-04-09:
If I understand correctly you want to use a programmer STM32Cube to upload
If so this may be a duplicate of #1362

SigP226, 2022-04-09:
Yes, I use STM32cube. What cause me trouble is that it was perfectly OK before, that's not the first time I use STLINK system and sloeber/eclipse and it was working fine. I am not sure it is a duplicate of #1362 because I have no issue with arduino. It works fine. The main difference with before is the switch to 2.2.0 of STM32duino.
I dont know very well how it work, I understand the upload command line is created via a pattern in Platform_sloeber.txt : tools.stm32CubeProg.upload.pattern="${busybox}" sh "${tools.stm32CubeProg.path}/${tools.stm32CubeProg.cmd}" ${upload.protocol} "${build.path}/${build.project_name}.bin" ${upload.options}
the command line written in the upload console is correct, the file are at the good location.
I continue to search how it works, but I think the issue is between the creation of command line and the call to the shell macro stm32CubeProg.sh.

SigP226, 2022-04-09:
root cause found! I think the issue is in plateform_sloeber.txt 
[image]

the pattern does not contain the right path for busybox : 
[image]

The upload pattern ask to "busybox" path, but this variable is never declared as we can see in the environment of the project.

[image]

[image]
So I have manually corrected the environment like this : 

[image]
 and... magic !!!

[image]

I don't know how works Sloeber, I think if I modify manually the platform_sloeber.txt, it will take effect for each new project creation.
I thank you for this very useful tool, I like sloeber, with eclipse it is a dream compare to the arduino IDE.

jantje, 2022-05-30:
closing as duplicate of #1362

closed 2022-05-30 by jantje