Sloeber/arduino-eclipse-plugin #1463: definition of return type is missing in sloeber.ino.cpp

pischky, 2022-04-09:
Environment Sloeber 4.4.0 
Arduino AVR board 1.8.2
Arduino Nano / ATmega328P (old bootloader)

````
#include "Arduino.h"
typedef struct tuple{ char a; char b; };
//using tuple2 = tuple;
tuple makeTuple(uint8_t a, uint8_t b) {
	tuple x; x.a = a; x.b = b; return x;
}
void setup() { tuple x = makeTuple(1, 2); }
void loop() {}
````

gives:
````
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"G:\PGMFILES\sloeber-4.4.0\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR     -I"G:\PGMFILES\sloeber-4.4.0\arduinoPlugin\packages\arduino\hardware\avr\1.8.2\variants\eightanaloginputs" -I"G:\PGMFILES\sloeber-4.4.0\arduinoPlugin\packages\arduino\hardware\avr\1.8.2\cores\arduino" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
..\sloeber.ino.cpp:10:1: error: 'tuple' does not name a type
 tuple makeTuple(uint8_t a, uint8_t b) ;
 ^~~~~
````

Same problem when using ``using``.

pischky, 2022-04-09:
Same Problem with latested Platform AVR board 1.8.5.
Arduino IDE compiles this without error.

jantje, 2022-04-09:
This is a duplicate of known limitation #775

pischky, 2022-04-09:
Thanks, I understand.

closed 2022-04-09 by pischky