Sloeber/arduino-eclipse-plugin #1466: Set binary type for RPi Pico (RP2040)

YuryMalyshev, 2022-04-18:
Is it possible to set the binary type using flags?
https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf#page=359 section "Control of binary type produced (advanced)"

I would like to compile my program so that it can run from RAM (using copy_to_ram flag), so that I could modify the flash without affecting the running program.

I tried adding the lines to the makefile and environmental variables but no luck.

jantje, 2022-04-18:
It is pretty unclear to me what you want to do.
If you want to modify the build command; this is the place to be
[image]

YuryMalyshev, 2022-04-18:
It seems it's not the place.
At the moment, I just want to re-create this example - https://github.com/raspberrypi/pico-examples/blob/master/flash/nuke/nuke.c but using Sloeber and Arduino libraries.

So, like this:
```
#include "Arduino.h"

#if !PICO_NO_FLASH
#error "This example must be built to run from SRAM!"
#endif

#include "pico/stdlib.h"
#include "hardware/flash.h"
#include "pico/bootrom.h"

void setup() {
    uint flash_size_bytes;
    flash_size_bytes = PICO_FLASH_SIZE_BYTES;
    flash_range_erase(0, flash_size_bytes);
    // Leave an eyecatcher pattern in the first page of flash so picotool can
    // more easily check the size:
    static const uint8_t eyecatcher[FLASH_PAGE_SIZE] = "NUKE";
    flash_range_program(0, eyecatcher, FLASH_PAGE_SIZE);

    // Flash LED for success
    pinMode(LED_BUILTIN, OUTPUT);
    for (int i = 0; i < 3; ++i) {
        digitalWrite(LED_BUILTIN, HIGH);
        delay(100);
        gpio_put(LED_BUILTIN, LOW);
        delay(100);
    }

    // Pop back up as an MSD drive
    reset_usb_boot(0, 0);
}

void loop(){}
```
But they use CMake and CMakeLists, so I'm not sure if it's even possible

jantje, 2022-04-18:
I still don't have a clue what you are trying to do and what "does not work"

jantje, 2022-05-07:
as I do not understand the issue there is little I can do to help

closed 2022-05-07 by jantje